zigpy/zigpy #601: Delayed switching (Tuya zigbee TS004x)

pjkersten, 2020-12-29:
Hi,

I'm experiencing delay issues with Tuya TS004x remote control switches. After pressing a key, it takes 4-5 seconds before HA reacts. The ZigBee message itself is 'instantaneous' (using debug level logging), but HA's reaction isn't.

I suspect that discriminating between 'press' and 'long press' or 'double press' is the culprit some time, but this delay makes things next to unusable. I was not able to find anything useful in the code.

Any hints, suggestions on where to find the handling code / what I can do to enhance the UX (aka remove delays)?

Any help is appreciated.

TIA!

Adminiuga, 2020-12-29:
debug log including zigbee message and homeassistant.core event fired.
I can confirm the delay, but apparently the delay is from the device itself, specially when you are doing double clicks

pjkersten, 2020-12-30:
I've contacted the manufacturer. Hope they can shed some light on this.

closed 2021-09-07 by github-actions[bot]