zigpy/zigpy #606: handle_message() got an unexpected keyword argument 'dst_addressing'

zoic21, 2021-01-05:
Hello,

With last zigpy version I got this type of error. I search but I can't find the real issue.

 ````
[2021-01-04 12:03:51][DEBUG] : Frame received: 0x0e24000700aa00
[2021-01-04 12:03:51][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[2021-01-04 12:03:51][DEBUG] : Command Command.aps_data_indication (1, 1)
[2021-01-04 12:03:51][DEBUG] : Send: 0x1724000800010001
[2021-01-04 12:03:51][DEBUG] : Frame received: 0x17240022001b00220200000102d705040401060004000112fd0000afff47a40700c9
[2021-01-04 12:03:51][DEBUG] : APS data indication response: [27, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x05d7>, 4, 260, 6, b'\x01\x12\xfd\x00', 0, 175, 255, 71, 164, 7, 0, -55]
[2021-01-04 12:03:51][ERROR] : Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.7/dist-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/dist-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.7/dist-packages/zigpy_deconz/api.py", line 360, in data_received
    getattr(self, "_handle_%s" % (command.name,))(data)
  File "/usr/local/lib/python3.7/dist-packages/zigpy_deconz/api.py", line 493, in _handle_aps_data_indication
    data[16],
  File "/usr/local/lib/python3.7/dist-packages/zigpy_deconz/zigbee/application.py", line 324, in handle_rx
    self.handle_message(device, profile_id, cluster_id, src_ep, dst_ep, data)
  File "/usr/local/lib/python3.7/dist-packages/zigpy/application.py", line 229, in handle_message
    dst_addressing=dst_addressing,
TypeError: handle_message() got an unexpected keyword argument 'dst_addressing'
````

I think is due to recently add of dst_addressing on zigpy but maybe not on zigpy-deconz.

Thank in advance

Adminiuga, 2021-01-05:
update your version of zha-device-handlers???

zoic21, 2021-01-06:
Is that... I use my own github repo for zha-device-handlers and it's not up to date.

Sorry for the completely stupid worries

closed 2021-01-06 by zoic21

vali959, 2025-07-20:
else:
    bot.forward_message(ADMIN_ID, message.chat.id, message.message_id)  # Sizga yuboradi
    bot.send_message(message.chat.id, "✅ Buyurtmangiz qabul qilindi. Tez orada aloqaga chiqamiz.")  # Avtojavob