zigpy/zigpy #608: Danfoss Ally cannot set heating mode to OFF

wormiedk, 2021-01-06:
Whenever I try to set my Danfoss Ally to off via homeassistant I get the following entries in the log:
`2021-01-06 14:23:17 DEBUG (MainThread) [zigpy.device] [0x252b] Extending timeout for 0x93 request
2021-01-06 14:23:19 DEBUG (MainThread) [zigpy.zcl] [0x252b:1:0x0201] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=147 command_id=Command.Write_Attributes_rsp>
2021-01-06 14:23:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x252B:1:0x0201]: wrote {'system_mode': <SystemMode.OFF: 0>} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.INVALID_VALUE: 135>, attrid=28)]]
2021-01-06 14:23:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x252B:1:0x0201]: couldn't set 'SystemMode.OFF' operation mode
2021-01-06 14:23:22 DEBUG (MainThread) [zigpy.zcl] [0x14f3:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=60 command_id=Command.Report_Attributes>
2021-01-06 14:23:22 DEBUG (MainThread) [zigpy.zcl] [0x14f3:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=28>)]]

2021-01-06 14:25:36 DEBUG (MainThread) [zigpy.device] [0x252b] Extending timeout for 0xd2 request
2021-01-06 14:25:37 DEBUG (MainThread) [zigpy.zcl] [0x252b:1:0x0201] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=210 command_id=Command.Write_Attributes_rsp>
2021-01-06 14:25:37 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x252B:1:0x0201]: wrote {'system_mode': <SystemMode.OFF: 0>} attrs, Status: [[WriteAttributesStatusRecord(status=<Status.INVALID_VALUE: 135>, attrid=28)]]
2021-01-06 14:25:37 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x252B:1:0x0201]: couldn't set 'SystemMode.OFF' operation mode`

wormiedk, 2021-01-06:
Created under zha device handlers instead where I guess it is more correctly placed:
https://github.com/zigpy/zha-device-handlers/issues/702

Hedda, 2021-01-14:
@wormiedk maybe you can just close this duplicate issue in favour of only leaving the https://github.com/zigpy/zha-device-handlers/issues/702 issue open? 

As both issues are under the zigpy organization its developers can actually move an issue from one repo to another repo. 

FYI, nice is that if an issue is moved then the old URL will still work as it will then just redirect to the new URL of the moved issue.

closed 2021-01-14 by wormiedk