zigpy/zigpy #626: Does Time cluster implements time responses?

javicalle, 2021-01-17:
I have run into some problems implementing decoupled mode on a Xiaomi wall switch without neutral [(zha-device-handlers #715)](https://github.com/zigpy/zha-device-handlers/pull/715).
Looking for some information, I have found several reports like mine but for people who use Z2M. In most cases it seems that the problem was located in the implementation of the Time cluster responses:

> Users of the zigbee2mqtt HA solution were experiencing problems with no-neutral Aqara wired switches dropping their connection which was related to the switches querying the time cluster from the coordinator (hub) and receiving no response. The issue was resolved by the author of zigbee2mqtt implementing the coordinators time cluster response 

(from [device-handler-for-aqara-wired-wall-switch #163](https://community.smartthings.com/t/device-handler-for-aqara-wired-wall-switch/101062/163))

Running into the same kind of problems and not being able to figure out the answer myself, let's see if someone can tell me (or tell me where to look) if it is being implemented in the case of zigpy.

TL; DR;
It seems that some Xiaomi devices consult the Time cluster every few minutes. Some reports indicate that the device is stuck when not receiving a response and others only indicate that they remain offline when not receiving it.
This seems to be related only with non-neutral wall switches.

javicalle, 2021-01-17:
Similar reports for others libraries/integrations:
* https://github.com/dresden-elektronik/deconz-rest-plugin/issues/757
* https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2048#issuecomment-617695113

Aditional info about time response implementation:
* https://github.com/Koenkk/zigbee2mqtt/issues/943#issuecomment-487172996
* https://github.com/Koenkk/zigbee2mqtt/issues/1214#issuecomment-471914038
* https://github.com/Koenkk/zigbee2mqtt/issues/441

I don't have any zigbee sniffer to be able to provide specific data. I hope that in the above links you can get useful information if necessary.

Adminiuga, 2021-01-17:
[Time cluster](https://github.com/zigpy/zigpy/blob/69e019287b61d9b8941b9121bd10e267458491f9/zigpy/zcl/clusters/general.py#L507) does implement responses and I have a few devices which successfully update their time.
AFAIK Tuya does not use the time cluster but wants to receive the time response on their manufacturer specific cluster. Check open/closed issues and PRs regarding Tuya in zha-device-handlers repo, as someone else (was it you?) ask the similar question and IIRC implemented the response for Tuya thermostat

javicalle, 2021-01-17:
Thank you again, sensei.

I had gotten to that implementation, but I am still not able to understand the whole implementation. Thanks for clarifying.
I'm going to investigate what you have indicated to see where it takes me.

This issue can be closed..

closed 2021-01-17 by javicalle