zigpy/zigpy #628: King Of Fans MR101Z missing MAX setting

flyize, 2021-01-18:
_I'm completely guessing that I need to report this here, so apologies..._

I’m in the process of migrating all my Zigbee devices off SmartThings into HA directly. The fan controllers that I use are missing a fan setting. In HA, I only have low, medium, and high (plus a couple other irrelevant settings).

The actual remote for the fan and SmartThings had low, medium, high, and max. How might I work to get this corrected?

If it matters, this is the fan controller in question:
https://www.homedepot.com/p/Hampton-Bay-Universal-Wink-Enabled-White-Ceiling-Fan-Premier-Remote-Control-99432/206591100

And this is the DH I was using in ST, if it helps at all.
https://github.com/dcoffing/KOF-CeilingFan/blob/65c7ce03177fb955c149d10717399b442b480f9a/devicetypes/dcoffing/kof-zigbee-fan-controller.src/kof-zigbee-fan-controller.groovy

walthowd, 2021-01-18:
I believe "on" is the max setting. 
[image]

flyize, 2021-01-18:
I'll be damned, you're right! Is there some way to change this so that it's more clear then?

walthowd, 2021-01-19:
We'd have to override the Fan cluster from zigpy...
https://github.com/zigpy/zigpy/blob/7e05720e57d371177c4543cee8838b8e056fbf99/zigpy/zcl/clusters/hvac.py#L345

....From the zha-device-handlers quirk for this device:

https://github.com/zigpy/zha-device-handlers/blob/b12ebc37790af34253fce4a0c549d05a8ffb995c/zhaquirks/kof/kof_mr101z.py

Adminiuga, 2021-01-19:
theses are the speeds from ZCL
[image]

For KoF the "On" setting is the max speed, but for other fans the "On" setting could have a different meaning, eg for FANs clusters on Zigbee Thermostats 🤷

walthowd, 2021-01-19:
I believe "Auto" or "Smart" is what KoF calls "Wave" mode where the fan randomly changes speed too.

Adminiuga, 2021-01-19:
Yes.  I'm not quite positive how to make it "general" across all models

flyize, 2021-01-20:
> theses are the speeds from ZCL
> [image]
> 
> For KoF the "On" setting is the max speed, but for other fans the "On" setting could have a different meaning, eg for FANs clusters on Zigbee Thermostats 🤷

So it's not possible to fix it for just this fan controller?

walthowd, 2021-01-20:
I looked through the code base, and no it would require quite a bit of work through ZHA and zigpy to abstract the speed settings. 

You could create a template fan that would let you represent everything correctly:
https://www.home-assistant.io/integrations/fan.template/

If you do so, maybe share it on the HA forums so it can help others. Thanks.

Adminiuga, 2021-01-20:
>So it's not possible to fix it for just this fan controller?

It is not, as ZHA tries to provide a generalized interface for different Zigbee Fan devices. in other words ZHA follows the ZCL specs in order to be compatible with all devices which also follow the specs. On the downside, there won't be device specific customizations for devices like KoF

But Walthowd's suggestion to use the fan.template is a good approach

flyize, 2021-01-20:
> I looked through the code base, and no it would require quite a bit of work through ZHA and zigpy to abstract the speed settings.
>
> You could create a template fan that would let you represent [...]

In light of all of your answers, this will have to do. Thanks for the suggestion.

Adminiuga, 2021-01-29:
HA is changing FAN entity model to support speed %, then we could treat "on" state as 100%, unless for some ZB Fans the on state is whatever last speed setting was...

Hedda, 2021-02-01:
Should this discussion be moved to "[zha-device-handlers](https://github.com/zigpy/zha-device-handlers)" issues instead? -> https://github.com/zigpy/zha-device-handlers/issues'

...or alternatively, discussed in zigpy's ZHA dev forum? - https://github.com/zigpy/zigpy/discussions/categories/zha-development

Hedda, 2021-02-01:
> In light of all of your answers, this will have to do. Thanks for the suggestion.

@bdraco is as it seems apparently looking into a ZHA solution for this for in Home Assistant -> https://github.com/home-assistant/core/pull/45758

Hedda, 2021-02-18:
FYI, the ZHA pull request https://github.com/home-assistant/core/pull/45758 for the new fan entity model has now been merged into Home Assistant core.

So I believe that you could now start to look at fan speed quirk(s) for zha-device-handlers of specific models of different fans(?).

> That's something that would need to be added to https://github.com/zigpy/zha-device-handlers/blob/b12ebc37790af34253fce4a0c549d05a8ffb995c/zhaquirks/kof/kof_mr101z.py before we could add it in HA.

Note however that some fan speed interpretation inconsistencies in Home Assistant so bdraco is now also working on additional new pull requests for Home Assistant core ( https://github.com/home-assistant/core/pull/46512 and https://github.com/home-assistant/core/pull/46714 ) which will add fan speed procentage interpretation support which in turn will allow for the addition of "fan.increase_speed" and "fan.decrease_speed" commands in Home Assistant. This is really needed for intuitive voice assistant control of fan speeds via Amazon Alexa / Amazon Echo and Google Assistant / Google Home speakers.

Hedda, 2021-02-19:
By the way, I also read that the "smart" speed should probably be renamed in the quirk to either "breeze" or "variable" as either of those would be a better descrition of what that speed actially does, which is to vary the speed up and down to create a breeze.

> [image]

flyize, 2021-02-19:
@Adminiuga @walthowd Now that it seems some of the foundation is in place to get these fan options to be more descriptive, would it be possible to move this issue to https://github.com/zigpy/zha-device-handlers so that the quirk can be updated?

Thanks guys!

flyize, 2021-03-16:
Hey guys, now that the new fan model has been implemented, I was wondering if we could get this quirk fixed? It seems that setting the device to 100% does not actually set the device to MAX (or what HA says is the ON setting).

Thanks!

Hedda, 2021-03-22:
FYI, since part of the same organization on GitHub now members should be able to move issues between all the org repositories.

As such, developers that are members of the origination should be able to move issues from https://github.com/zigpy/zigpy/issues to https://github.com/zigpy/zha-device-handlers/issues and vice versa as needed, like when a uses post issues to the wrong repo.

That way if an issue can now be moved instead then there should no longer be any need for users to create a copy of an issue.

flyize, 2021-06-16:
Should anyone come back across this. I hacked together a template for this device as a bandaid.  Posted here:

https://community.home-assistant.io/t/king-of-fans-mr101z-missing-max-setting/267131/15?u=flyize

Hedda, 2021-11-18:
For reference, the discussion about quirk update reques for King Fans MR101Z fan will about https://github.com/zigpy/zha-device-handlers/issues/1092

closed 2022-05-24 by github-actions[bot]