zigpy/zigpy #641: Zigpy giving the error : Error doing job: Task exception was never retrieved

nickbarau, 2021-01-24:
Looking to understand why all Zigbee devices drop off the network when this error occurs :

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:99
First occurred: January 23, 2021, 12:58:00 PM (2 occurrences)
Last logged: January 23, 2021, 11:24:19 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 223, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

```

When the above happens, **all** Zigbee devices become unreachable.

puddly, 2021-01-24:
Can you enable [ZHA debug logging](https://www.home-assistant.io/integrations/zha/#debug-logging) and include the `home-assistant.log` file that includes this error (redacting any personal information, if you care)? Your log contains no context whatsoever.

nickbarau, 2021-01-29:
Attaching the full log. Didn't want to accidentally truncate important bits.
Errors can be seen at 2021-01-29 06:41:00 and 2021-01-29 06:53:09
[home-assistant.log](https://github.com/zigpy/zigpy/files/5891026/home-assistant.log)

Adminiuga, 2021-01-29:
What version?
What bridge?
What system?
per logs, the network is well alive, having traffic in both directions.

Adminiuga, 2021-01-29:
towards the end of the log
```
2021-01-29 08:17:35 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.lumi_lumi_sensor_motion_aq2_95cd6104_ias_zone, old_state=<state binary_sensor.lumi_lumi_sensor_motion_aq2_95cd6104_ias_zone=off; friendly_name=Living Room Motion Sensor, device_class=motion @ 2021-01-29T08:14:22.872331+08:00>, new_state=<state binary_sensor.lumi_lumi_sensor_motion_aq2_95cd6104_ias_zone=on; friendly_name=Living Room Motion Sensor, device_class=motion @ 2021-01-29T08:17:35.352106+08:00>>
```

so both network is alive and device is updated in HA 🤷

nickbarau, 2021-01-30:
HI all,

@Adminiuga 
## System Health

version | 2021.1.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | Asia/Kuala_Lumpur

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4992
Installed Version | 1.10.1
Stage | running
Available Repositories | 709
Installed Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 3, 2021, 8:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.10
-- | --
update_channel | stable
supervisor_version | 2021.01.7
docker_version | 19.03.13
disk_total | 28.6 GB
disk_used | 12.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Duck DNS (1.12.4), NGINX Home Assistant SSL proxy (3.0.1), Grafana (6.0.0), InfluxDB (4.0.1), Glances (0.10.0), Samba share (9.3.0), Home Assistant Google Drive Backup (0.103.1), Terminal & SSH (8.10.0), Grocy (0.11.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 10
resources | 0

</details>

Again I see this error pop up which usually precedes some zigbee devices dropping off :

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:99
First occurred: January 29, 2021, 12:33:10 PM (4 occurrences)
Last logged: 1:33:44 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 223, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

Also sharing logs for 30th Jan and hopefully will show the occurrence and correlation. 
Apologies, I don't know what to specifically look for considering the file is 65mb large 
https://1drv.ms/u/s!AgJKaOf8zPn1jz6qnFdxEgaCAHBV

Adminiuga, 2021-01-30:
Some devices go offline or all devices go offline?
What exact devices go offline? Model and manufacturer and don't tell me it is xiami.
I can say that I don't see a correlation between this error and devices going offline, as you have same error in your previous logs and the network seems alive.

nickbarau, 2021-01-31:
This sometimes changes with some other devices in the network such as FNB56-SKT1JXN1.0 by FeiBit (Not in DB but works to add devices and control) or Ewelink Plugs https://zigbee.blakadder.com/eWeLink_Zigbee-Plug-UK.html

Only some devices go offline and in this instance the following. Obviously, with repeaters going down, other devices downstream will suffer.
The examples of device and manufacturer as below :

14:b4:57:ff:fe:36:43:11 Last Seen: 2021-01-30T12:50:32 
14:b4:57:ff:fe:36:41:6d Last Seen: 2021-01-30T12:48:06
TS0207 by _TZ3000_m0vaazab
https://zigbee.blakadder.com/Tuya_TS0207.html

e.g. for 14:b4:57:ff:fe:36:41:6d
```
2021-01-30 10:39:19 DEBUG (MainThread) [zigpy.neighbor] [0xefa2] request status: Status.SUCCESS. response: Neighbors(entries=10, start_index=6, neighbor_table_list=[Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=00:15:8d:00:04:4a:77:01, nwk=0xED42, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=6), Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=14:b4:57:ff:fe:36:41:6d, nwk=0x1FC2, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=1), Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=ec:1b:bd:ff:fe:ac:52:81, nwk=0xA169, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=31)])
2021-01-30 10:39:20 DEBUG (MainThread) [zigpy.neighbor] [0xefa2] Querying next starting at 9
2021-01-30 10:39:20 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-01-30 10:39:20 DEBUG (MainThread) [zigpy.neighbor] [0xefa2] request status: Status.SUCCESS. response: Neighbors(entries=10, start_index=9, neighbor_table_list=[Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=ec:1b:bd:ff:fe:ac:53:8e, nwk=0x4F5E, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=1)])
2021-01-30 10:39:20 DEBUG (MainThread) [zigpy.neighbor] [0xefa2] Done scanning. Total 10 neighbours
2021-01-30 14:41:41 DEBUG (MainThread) [zigpy.neighbor] [0xbe2e] request status: Status.SUCCESS. response: Neighbors(entries=11, start_index=6, neighbor_table_list=[Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=14:b4:57:ff:fe:36:43:0f, nwk=0x7A84, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=1), Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=00:12:4b:00:1c:d5:92:cf, nwk=0xEFA2, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=0), Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=14:b4:57:ff:fe:36:41:6d, nwk=0x1FC2, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=0)])
2021-01-30 14:41:42 DEBUG (MainThread) [zigpy.neighbor] [0xbe2e] Querying next starting at 9
2021-01-30 14:41:42 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-01-30 14:41:43 DEBUG (MainThread) [zigpy.neighbor] [0xbe2e] request status: Status.SUCCESS. response: Neighbors(entries=11, start_index=9, neighbor_table_list=[Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=ec:1b:bd:ff:fe:ac:52:81, nwk=0xA169, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=1), Neighbor(extended_pan_id=ca:35:f7:5b:ff:8e:6e:7d, ieee=ec:1b:bd:ff:fe:ac:53:8e, nwk=0x4F5E, packed=57, permit_joining=<PermitJoins.Unknown: 2>, depth=255, lqi=1)])
2021-01-30 14:41:43 DEBUG (MainThread) [zigpy.neighbor] [0xbe2e] Done scanning. Total 11 neighbours
2021-01-30 14:43:06 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of 14:b4:57:ff:fe:36:41:6d/0x1fc2 device
2021-01-30 14:43:06 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2021-01-30 14:43:11 DEBUG (MainThread) [zigpy.device] [0x1fc2] Delivery error for seq # 0x24, on endpoint id 0 cluster 0x0031: message send failure
2021-01-30 14:43:12 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2021-01-30 14:43:17 DEBUG (MainThread) [zigpy.device] [0x1fc2] Delivery error for seq # 0x27, on endpoint id 0 cluster 0x0031: message send failure
2021-01-30 14:43:18 DEBUG (MainThread) [zigpy.util] Tries remaining: 1
2021-01-30 14:51:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=14:b4:57:ff:fe:36:41:6d, unique_id=14:b4:57:ff:fe:36:41:6d, device_id=14cd1b6e10eb11eb8006b988b7a97d03, device_event_type=device_offline>
2021-01-30 14:51:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Repeaters offline, entity_id=automation.repeater_1_offline, source=event 'zha_event'>
2021-01-30 14:51:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.repeater_1_offline, old_state=<state automation.repeater_1_offline=on; last_triggered=2021-01-24T17:21:40.138908+08:00, mode=single, current=0, id=1602067711460, friendly_name=Repeaters offline @ 2021-01-29T06:23:37.562272+08:00>, new_state=<state automation.repeater_1_offline=on; last_triggered=2021-01-30T14:51:20.226311+08:00, mode=single, current=1, id=1602067711460, friendly_name=Repeaters offline @ 2021-01-29T06:23:37.562272+08:00>>
```

What I understand from the above bits is the Tuya repeater 14:b4:57:ff:fe:36:41:6d successfully reported back its neighbors, and then eventually failed triggering a automation to let me know it "died". 

I'll dig a little further and will close out this issue for now as it looks like the general consensus is that there is no correlation. 

Thanks everyone.

closed 2021-01-31 by nickbarau

Adminiuga, 2021-01-31:
It's 0xbe2e and 0xefa2 devices successfully nreported their neighbors with 14:b4:57:ff:fe:36:41:6d being one of the neighbors. However there wasn't any traffic from 14:b4:57:ff:fe:36:41:6d for 2 hours nor did it answer to attribute read request so zha marked it offline.

MattWestb, 2021-01-31:
Is the antenna to week in the repeater is not being used by devices or do it not having any clusters that can being red from ZHA ??