zigpy/zigpy #661: Tuya 3 gang switch support is needed

loglux, 2021-02-12:
https://www.aliexpress.com/item/1005001532044556.html
[image]

loglux, 2021-02-20:
Any hope?

Adminiuga, 2021-02-20:
It got the quirk, so must be doing zha_event then.

loglux, 2021-02-20:
How can I help?

Adminiuga, 2021-02-20:
keywords for google: home assistant, zha_event
This is a remote, does not have entities it does `zha_event` Check home assistant on how to listen to event to see what's being generated. zigbee channel on Home assistant discord channel is also a good place to start

Device got a quirk, so must be supported by `zha-device-handlers`. Zigpy is really wrong place for this kind of the issue.

Hedda, 2021-03-22:
FYI, since part of the same organization on GitHub now members should be able to move issues between all the org repositories.

As such, developers that are members of the origination should be able to move issues from https://github.com/zigpy/zigpy/issues to https://github.com/zigpy/zha-device-handlers/issues and vice versa as needed, like when a uses post issues to the wrong repo.

That way if an issue can now be moved instead then there should no longer be any need for users to create a copy of an issue.

closed 2021-03-22 by Adminiuga