zigpy/zigpy #662: Binding between two devices

zoic21, 2021-02-12:
Hello,

I need to create direct binding two zigbee devices. I wonder how to do that with zigpy I think I need to use that https://github.com/zigpy/zigpy/blob/3bd6a4fa1e8523e36f4c1780ef8d99f319f84189/zigpy/zdo/__init__.py#L97 but I need a confirmation.

I also wonder how can I get current binding between two devices ?

Thank in advance

Adminiuga, 2021-02-13:
Yes. There's a zdo command to read the current bindings. Find and download zigbee device base device behavior revision 21

MattWestb, 2021-02-13:
I think only version 1.0 is publicized [Base Device Behavior Specification
Version 1.0](https://zigbeealliance.org/wp-content/uploads/2019/12/docs-13-0402-13-00zi-Base-Device-Behavior-Specification-2-1.pdf).

ZCL ver 7 is linked from here https://github.com/zigpy/zigpy/discussions/595#discussioncomment-287276.

Aso good if working with lights in ZB3 if [ZigBee Lighting & Occupancy Device Specification Version 1.0](https://zigbeealliance.org/wp-content/uploads/2019/11/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf).

I putting the first and last on the reference collection in the discussion and if Adminiuga like in one new wiki page.

zoic21, 2021-04-14:
Hello,

I make some test on binding but I can't make it work. I think I make something wrong. For test I take ikea  store and it's remote, both device are include on deconz key. I try to bind input cluster covering of store with remote (I also test to bind output cluster of remote with store). Here the log : 

```
0000|[2021-04-14 09:58:19][DEBUG] : [DeviceHandler.prepare] Json arg : {'src': {'ieee': '00:0d:6f:ff:fe:9c:d2:81', 'endpoint': 1, 'cluster': 258, 'cluster_type': 'in'}, 'dest': {'ieee': '00:0d:6f:ff:fe:7b:7d:4b'}}
0001|[2021-04-14 09:58:19][DEBUG] : [0x2c37] Extending timeout for 0x56 request
0002|[2021-04-14 09:58:19][DEBUG] : Sending Zigbee request with tsn 86 under 87 request id, data: b'564b7d7bfeff6f0d0001020103a4b403ffff2e210001'
0003|[2021-04-14 09:58:19][DEBUG] : Command Command.aps_data_request (37, 87, 0, <DeconzAddressEndpoint address_mode=2 address=0x2C37 endpoint=0>, 0, <ZDOCmd.Bind_req: 0x0021>, 0, b'VK}{\xfe\xffo
\x00\x01\x02\x01\x03\xa4\xb4\x03\xff\xff.!\x00\x01', 2, 0)
0004|[2021-04-14 09:58:19][DEBUG] : Send: 0x127a002c002500570002372c0000002100001600564b7d7bfeff6f0d0001020103a4b403ffff2e2100010200
0005|[2021-04-14 09:58:19][DEBUG] : Frame received: 0x127a00090002002257
0006|[2021-04-14 09:58:19][DEBUG] : APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 87]
0007|[2021-04-14 09:58:19][DEBUG] : Frame received: 0x0e7b000700a600
0008|[2021-04-14 09:58:19][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
0009|[2021-04-14 09:58:19][DEBUG] : Command Command.aps_data_confirm (0,)
0010|[2021-04-14 09:58:19][DEBUG] : Send: 0x047b0007000000
0011|[2021-04-14 09:58:19][DEBUG] : Frame received: 0x047b0013000c00225702372c00000000000000
0012|[2021-04-14 09:58:19][DEBUG] : APS data confirm response for request with id 87: 00
0013|[2021-04-14 09:58:19][DEBUG] : Request id: 0x57 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x2c37 endpoint=0>, status: 0x00
0014|[2021-04-14 09:58:22][DEBUG] : Frame received: 0x0e7c000700aa00
0015|[2021-04-14 09:58:22][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
0016|[2021-04-14 09:58:22][DEBUG] : Command Command.aps_data_indication (1, 1)
0017|[2021-04-14 09:58:22][DEBUG] : Send: 0x177c000800010001
0018|[2021-04-14 09:58:22][DEBUG] : Frame received: 0x177c0021001a00220100000002372c0104010201030001150000afb745618000b9
0019|[2021-04-14 09:58:22][DEBUG] : APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0000>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2c37>, 1, 260, 258, b'\x01\x15\x00', 0, 175, 183, 69, 97, 128, 0, -71]
0020|[2021-04-14 09:58:22][DEBUG] : [0x2c37:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=21 command_id=0>
0021|[2021-04-14 09:58:22][DEBUG] : [0x2c37:1:0x0102] ZCL request 0x0000: []
0022|[2021-04-14 09:58:22][INFO] : [00:0d:6f:ff:fe:7b:7d:4b][listener.cluster_command] Cluster: <zhaquirks.ikea.opencloseremote.IkeaWindowCovering object at 0x7faccdb0a898> ClusterId: 0x0102 tsn: 21 args: (0, [])
0030|[2021-04-14 09:58:23][DEBUG] : Frame received: 0x0e7d000700aa00
0031|[2021-04-14 09:58:23][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
0032|[2021-04-14 09:58:23][DEBUG] : Command Command.aps_data_indication (1, 1)
0033|[2021-04-14 09:58:23][DEBUG] : Send: 0x177d000800010001
0034|[2021-04-14 09:58:23][DEBUG] : Frame received: 0x177d0026001f00220100000002373501040106000800010142000807000000afa735618000b7
0035|[2021-04-14 09:58:23][DEBUG] : APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0000>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3537>, 1, 260, 6, b'\x01\x01B\x00\x08\x07\x00\x00', 0, 175, 167, 53, 97, 128, 0, -73]
0036|[2021-04-14 09:58:23][DEBUG] : [0x3537:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=66>
0037|[2021-04-14 09:58:23][DEBUG] : [0x3537:1:0x0006] ZCL request 0x0042: [0, 1800, 0]
0038|[2021-04-14 09:58:23][DEBUG] : [0x3537:1:0x0006] No handler for cluster command 66
0053|[2021-04-14 09:58:23][DEBUG] : Frame received: 0x0e7e000700aa00
0054|[2021-04-14 09:58:23][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
0055|[2021-04-14 09:58:23][DEBUG] : Command Command.aps_data_indication (1, 1)
0056|[2021-04-14 09:58:23][DEBUG] : Send: 0x177e000800010001
0057|[2021-04-14 09:58:23][DEBUG] : Frame received: 0x177e0020001900220200000002372c00000021800200560000afff35618000dd
0058|[2021-04-14 09:58:23][DEBUG] : APS data indication response: [25, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2c37>, 0, 0, 32801, b'V\x00', 0, 175, 255, 53, 97, 128, 0, -35]
0059|[2021-04-14 09:58:23][DEBUG] : 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2c37>, ep: 0, profile: 0x0000, cluster_id: 0x8021, data: b'5600'
0060|[2021-04-14 09:58:23][INFO] : 200 PUT /device/bind (127.0.0.1) 4336.45ms
```

It's seem ok (I need to wake up remote to do that) but after that no move of store on remote clic.

Thank in adance

MattWestb, 2021-04-14:
I think (= not knowing) you is trying binding one IKEA Open/Close remote with one IKEA blinds with device <-> device binding.

IKEA is in the ground ZLL and is exclusive using groups for all thing and all controlling devices only accepting group bindings and no devices bindings.

Making on group and bind the open/close remote cluster to the group and adding the blind to the group so it receiving the command the remote is sending.

If working with IKEA blinds its working little different then they is end devices and not routers so normal group commands broadcast is not forwarded to them and IKEA is using one special and is sending group commands with broadcast address that i also forwarded to end devices and the signal repeater is doing some "IKEA magic" but its not documented but i think its also need being added to the group your blinds and remote is using for helping the special to working OK also then  the coordinator is off line.

MattWestb, 2021-04-14:
PS: I dont have IKEA blinds or the signal repeater but have looking for the problematic but have not feeding some sniff of paring / adding blinds and signal repeater with IKEA GW that shall showing the "magic" they is doing.

zoic21, 2021-04-14:
Perfect it's work with group !!!! I don't use signal repeater but I think it work also with group. 

Thank a lot.

MattWestb, 2021-04-14:
## Great that jeedom is getting the blinds working !!!

IKEA is releasing on new FW for the blinds and the signal repeater its in the test feed and is Zigbee certified but i have not getting any feed back if its working well or not. https://github.com/zigpy/zigpy/discussions/660

From R21 spec:
Table 3.59 Broadcast Addresses
Broadcast Address Destination Group
0xffff All devices in PAN
0xfffe Reserved
0xfffd macRxOnWhenIdle = TRUE
0xfffc All routers and coordinator

Normal group broadcasts is made to 0xfffc and the group as payload so all routers is getting it and is using it if its member of the group.

But then the blinds is on end device it dont getting the group command so IKEA is sending it with destination 0xfffd (i have not sen it in on sniffing) for getting the group broadcast to the blinds working.

My theory is that the signal repeater is doing some "magic" then its no need making so much fixes that IKEA have doing in the firmware.

I think then IKEA is joining the signal repeater they is putting the group with touch link and is then binding the right cluster to it and can doing the magic but we cant binding it then its have no handles for it and can only being done with touch linking.

They is doing the same with the 5 button remote and binding the scene cluster to one "magic group" (0xff09) that we cant changing (but using it as one "IKEA global scene group" for the remotes).

Only my findings and theories but it can being god to getting the information then (if) getting problems with the blinds.

I dont have the blinds and the signal repeater so i cant sniffing then pairing them with the IKEA GW but i like getting one god sniff of it or analyzing !

MattWestb, 2021-04-14:
By the way in ZHA is god binding the group cluster 2 then doing the other cluster you need being bonded.
If not doing that is not possible doing on new binding after the first is being done but with the group cluster bonded its possible doing new bindings that is working in ZHA.

And also if you is changing the binding the Zigbee spec is saying that the attribute reporting shall being done after binding (that ZHA is not doing => not working attribute reporting until redoing the config of the reporting after doing one binding).

zoic21, 2021-04-16:
Hello,

I reopen this issue because I have a small question about binding. This fonction https://github.com/zigpy/zigpy/blob/3bd6a4fa1e8523e36f4c1780ef8d99f319f84189/zigpy/zdo/__init__.py#L97 take 2 parameters : 
- self : device it self
- cluster : the cluster to bind 

But if device (self) has multiple endpoint with same cluster on this endpoint. Like binding double remove with double micro-switch how can do that ? For exemple I want do bind :  
- **ep 1** cluster 6 of device 1 to **ep 1** cluster 6 on device 2
- **ep 2** cluster 6 of device 1 to **ep 2** cluster 6 on device 2

Thank in advance.

Adminiuga, 2021-04-16:
Just call device.zdo.Bind_req request directly and put the correct source/destination endpoints in. 
The zdo.bind() method assumes it's being bound to coordinator and uses the default endpoint, which is not suitable for multi source endpoint binding

zoic21, 2021-04-16:
Perfect I understand I add the endpoint selection if there is multiple endpoint.

Thank a lot

closed 2021-04-16 by zoic21