zigpy/zigpy #665: permit_with_key raise NotImplementedError

JohannCR, 2021-02-17:
Trying to permit a device using source_ieee and install_code, I get this error

`2021-02-17 19:33:45 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140241586170704]
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 136, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1455, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1490, in _execute_service
await handler.job.target(service_call)
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 665, in admin_handler
await result
File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 857, in permit
await application_controller.permit_with_key(
File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 382, in permit_with_key
raise NotImplementedError
NotImplementedError`

Thing is I'm not sure about the iee address, it's my first try I don't know how to get it from the device... It's a Legrand wall outlet that I cannot connect to using ZHA discovery even on channel 11 which the constructor uses.
If the error is my fault any idea how to connect the device ? Thanks

puddly, 2021-02-17:
What coordinator hardware are you using? At the moment, only Silicon Labs coordinators through the `bellows` library and Texas Instruments coordinators through `zigpy-znp` support install codes.

That being said, your device likely doesn't *require* an install code to join your network. It should still join using the normal procedure.

JohannCR, 2021-02-17:
Conbee 2.
I've paired like 30 zigbee devices successfully from multiple constructors, I cannot for the life of me pair this one. Driving me nuts.
Thanks for your answer anyway !

JohannCR, 2021-02-17:
Got some weird chars in the log too... I wasn't sure it came from this device so I deactivated everything else in aha. Still not sure it's the outlet, but being the only one doing that there's a good chance :

`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x9911:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=180 command_id=Command.Report_Attributes>
[0x9911:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�(!�!>)]]
[0x9911:1:0x0000] Attribute report received: 65281=!�(!�!
[0x9911:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:3045 < [Max]:3100, Battery Percent: 80.5
[0xf5c7:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=201 command_id=Command.Report_Attributes>`

puddly, 2021-02-17:
Those are corrupted attribute reports sent by Xiaomi devices.

When you say "even on channel 11", did you form your network initially on channel 11, or did you leave it on the default of 15 and then add channel 11 to the configuration? If it's the latter, your network is still on channel 15 and there isn't currently a way to switch it with ZHA and the Conbee. [deCONZ has a way to do this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2036) but many of your devices will likely need to be rejoined.

JohannCR, 2021-02-17:
Wow really ? I was so wrong in so many ways... 🤣
In that case I may even change the hardware... A CC2652RB or ZZH are tempting but they're hard to find. Will see.
Thanks for your precious help !

puddly, 2021-02-17:
Yeah, It's likely that the Legrand device only joins on 11 but your network is still on 15. If you temporarily use the deCONZ GUI to change the channel (and possibly increment the NWK update ID?), some of your devices should switch over after you reboot them.

JohannCR, 2021-02-18:
Follow up : couldn't change channel using deconz gui (it did not stick don't know why) but I successfully did using the phoscon app. You were absolutely right, now it works !
Thanks again

closed 2021-02-18 by JohannCR

MattWestb, 2021-02-18:
Its some bugs in the firmware / deCONZ GUI that is making changing channel not working.
Its reported for some mounts and is not fixed wot i knowing.

I hope ZHA is getting it implanted for making it easier for user changing channel of the coordinators (i have doing it with brute force with custom commands in ZHA on my EZSP coordinator).