zigpy/zigpy #670: Invalid checksum - stops working after a couple seconds

xdenneboom, 2021-02-21:
I have a homeassistant setup on a raspberry pi 3 with a raspbee controller. I'm using controllerX to use IKEA Tradfri remotes for my media control. This results in issues I initially reported here. Where I was asked to report my issue here on the zigpy repo.

# Bug report

## Description
When everything is set up, the controller works. the wheel controls volume. but after a few seconds stops responding. None of my controllers work anymore, until I reboot the appDaemon or Homeassistant
Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

every action that is performed, results in the invalid checksum warning in the log. But it still works. When I get a warning in the AppDaemon log, things stop working. 

## Additional information 
* Devices involved:
  * IKEA E1744 as Media Player  
  * IKEA ICTCG1 as Light
* Integration: Choose from `zha`
* AppDaemon version: v4 0.4.3 
* ControllerX version: v4.6.0
* HACS version (if installed from there): v1.11.3
* Home Assistant Core version: 2021.2.3

## AppDaemon app configuration

```yaml
---
ikea_sonos_wheel:
  module: controllerx
  class: E1744MediaPlayerController
  controller: ec:1b:bd:ff:fe:8b:97:ea
  integration: zha
  media_player: media_player.living_room
  actions:
    - move_1_195 #Left turn
    - move_0_195 #Right turn
    - stop #Stop rotation
    - toggle #1 click

ikea_bathroom_wheel:
  module: controllerx
  class: ICTCG1Controller
  controller: 00:0b:57:ff:fe:90:70:83
  integration: zha
  light: light.bathroom
```

## Logs
**AppDaemon log:**
```text
🤖 Service: media_player.volume_set
  - entity_id: media_player.living_room
  - volume_level: 0.20
2021-02-20 18:16:53.336529 INFO ikea_sonos_wheel: 
🤖 Service: media_player.volume_set
  - entity_id: media_player.living_room
  - volume_level: 0.10
2021-02-20 18:16:53.734866 WARNING AppDaemon: callback idle has now completed
```

**Homeassistant log:** 
```
Invalid checksum: 0xdbfe, data: 0x98c840410441bf
6:16:23 PM – /usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py (WARNING) - message first occurred at February 19, 2021, 8:10:58 PM and shows up 292 times

Logger: zigpy_deconz.uart
Source: /usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py:83
First occurred: February 19, 2021, 8:10:58 PM (292 occurrences)
Last logged: 6:16:23 PM
Invalid checksum: 0xdbfe, data: 0x98c840410441bf
Invalid checksum: 0x3cfe, data: 0x98c440410441bf9c3cfe98c840010441bf9c
Invalid checksum: 0xf8fc, data: 0x98e840010441bf993cfe98ec40010441bf9d
Invalid checksum: 0x79fc, data: 0xd8a6700002d2f898e840010401c29d
Invalid checksum: 0xf2f8, data: 0xd8f6708004c8f8d8da700004c9f8d8fe700004c9f898e040410401c29df2f898e440010401c29c
```
```
[0x28e8:1:0x0008] Data remains after deserializing ZCL frame
6:16:20 PM – /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py (WARNING) - message first occurred at February 19, 2021, 8:24:43 PM and shows up 55 times
Logger: zigpy.zcl

Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:123
First occurred: February 19, 2021, 8:24:43 PM (55 occurrences)
Last logged: 6:16:20 PM
[0x28e8:1:0x0008] Data remains after deserializing ZCL frame
```

## Additional Context
I've initially had seperate scenes for the double and tripple press configured, but I commented that out to scope down the problem, which still persists.

Adminiuga, 2021-02-22:
This is for zigpy-deconz. But even there, if you get the checksum error then it means the serial connection is corrupting data. 
Could be bad stick, could be something else accessing the port, bad power supply etc. Checksum errors are not problems with zigpy, but with setup or hardware

Hedda, 2021-02-22:
Regardless if to helps or not, know that it is recommended to flash upgrade the deCONZ firmware (Zigbee stack application firmware) for ConBee and RaspBee:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

As per https://github.com/dresden-elektronik/deconz-rest-plugin/releases = That firmware update is recommended for ZHA, zigbee2mqtt, WebThings, and deconz users. The last release it needs to be installed manually as described in the deconz Wiki by dresden elektronik.

http://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D

PS: Off-topic but good to know; when using internal Zigbee adapters like the RaspBee it is generally recommended to disaable both WiFi and Bluetooth on the Raspberry Pi.

MattWestb, 2021-02-22:
@xdenneboom If you is having one CorBee I or II check the USB cables and trying with other ones if they is having bad connection, also trying changing the USB port.

If having one RaspBee I or II you shall using the Pis HW comport for the PaspBee and the SW for the BT or disabling BT.

The software port is not being stable at high speed and is variating with frequency with the CPU load.

Im running deCONZ on one RP3B+ with HW port RaspBee I and  SW port for BT and WiFi and its working OK.

Take one look on the https://github.com/zigpy/zigpy/discussions/642 for making the right setting in your Pi.

xdenneboom, 2021-02-22:
Thanks guys, I'm using the raspbee II. Will have a look at the tips you gave me.

xdenneboom, 2021-02-22:
I had bluetooth disabled on my pi during installation. But there seems to be no way (that I could find) to upgrade the firmware when using HassOS. The raspbee is not usb mounted but sits on the raspberry pi board. So I'll try to setup a secondary rpi with raspbian to upgrade the deconz firmware first.

MattWestb, 2021-02-22:
You can also doing on install on one "new" SD-card and doing the update and then putting the original card back and is shall working.
You can using on  clean rasbian image and adding the flacher http://deconz.dresden-elektronik.de/raspbian/gcfflasher/ or one deCONZ SD-card image.
deCONZ instruction for Raspbian: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

Make sure that your RaspBee is using the hardware comport or you is getting problem with the communication and its not helping only have BT disabled you must doing all other things for getting  the OS switching the comport (but i dont knowing how to do that on HasssOS).