zigpy/zigpy #679: [Bug report / Enhancement request] Broken configuration of IKEA controlling device

MattWestb, 2021-03-02:
## Bug report broken configuration of IKEA controlling devices.
IKEA controlled devices is not correct setted up then its being paired to ZHA with EZSP coordinators.

Oft checking is being configurated but not always. Attribute reporting not working. Last seen normally only one time in 24 hours.

If doing one reconfiguration this devices its starting doing check ins, attribute reporting and being see every hours.

If doing one new binding to one group the attribute reporting is being broken and need doing one reconfiguration of the device for getting it working.
Zigbee standard is saying that the attribute reporting shall being setting up after the binding or its failing = wot ZHA is doing.

Now we coming to the other part of the "check in problem" / pull control problematic.
If only pairing one controller device is pulling its parent every 5 minutes and (normally) no check in is made = low battery daring.

Then dong one reconfiguring the pull control is working and is setting the long pull interval to (5 minutes butt after first check in its being sett to 24/4=) 6 seconds = draining the battery fast.

IKEA GW is setting up long pull (1160/4/60=4.83 minutes) ala 5 minutes = not reconfigured devices in ZHA.

Pulling one sleeping controlling device that is not need any commands in "real time" every 6 seconds is somthing that is more then crazy and ZHA is setting up the children time out to more then 4 hours wot i have seen (in bellows V8) and IKEA controllers is requesting one 8 minutes with `End Device Timeout Request`.

Zigbee dev papers is saying that pull control shall being adjusted / device type wot is needs. One example is the Philips HUE motion sensor that is behaving like on not sleeping end device then its listing very oft (very likely wrongly configuration).

If one controller device is pulling the EZSP coordinator with very short interval the coordinator getting problems and the device is leaving and rejoining the network and can being  locked in fast pull (4 time / second).

One creative user have deleting the pull control cluster in the quirk and its making the device puling its parent every 5 minutes but is loosing the check in functionality that is not so good on one real Zigbee 3 network but the batteries is lasting longer als IKEA GW is doing.

I see 2 ways to solvering this problem after the user have updating to EZSP 6.7.8.0 (i have verified super battery draining with EZSP 6.5.1.0 with direct children):

Deleting all pull control cluster in IKEA controller devices quirks and very likely more ones (working but is crappy and making ZHA being degraded as Zigbee 3 system).
Or
Implanting long pull control interval  / Zigbee device types or then its not possible / ZHA quirk device.

And fixing setting up attribute reporting after doing one new binding.
Also fixing so attribute reporting and check in then pairing IKEA controlling device. 

Devices that need some attention for our devs:

IKEA Controller Zigbee device types in Quirks:

TThe old dimmer: DEVICE_TYPE: `zha.DeviceType.NON_COLOR_CONTROLLE`
fivebtnremote: DEVICE_TYPE: `zll.DeviceType.SCENE_CONTROLLER`, `zha.DeviceType.NON_COLOR_CONTROLLER`, `zha.DeviceType.COLOR_SCENE_CONTROLLER`
motion: DEVICE_TYPE: `zll.DeviceType.ON_OFF_SENSOR` and `zha.DeviceType.ON_OFF_SENSOR`
opencloseremote DEVICE_TYPE:` zha.DeviceType.WINDOW_COVERING_CONTROLLER`
shortcutbtn DEVICE_TYPE: `zha.DeviceType.NON_COLOR_CONTROLLER`
symfonisk DEVICE_TYPE: `zha.DeviceType.REMOTE_CONTROL`
twobtnremote DEVICE_TYPE: `zha.DeviceType.NON_COLOR_CONTROLLER`, `zll.DeviceType.CONTROLLER`

Controllers devices (ZBL)
ZDeviceID = 0x0800 (2048)>> Color controler
ZDeviceID = 0x0810 (2064)>> Color scene controler
ZDeviceID = 0x0820 (2080) >> Non color controler
ZDeviceID = 0x0830 (2096)>> Non color scene controler
ZDeviceID = 0x0840 >> Control bridge = no sleeping end device.
ZDeviceID = 0x0850 (2128)>> on/off sensor

Oprn/cloce = 0x0203 515 (HA windows blinds controller)
Synf (ZB3 HA) = 0x0006  6 (HA remote)

Philips HUE Motion sensor have 2 endpoints one with 0x0850 and the other with 0x0107

If need more information i have "collecting" around 100 hours sniffing but ii dont sharing them here then its also have my network keys (but possible in private forum).

If this is being fixed we have 90 % of the IKEA battery draining fixed (with EZSP 6.7.8.0) and the rest is bugs in  thee device firmware that we can not fixing.

References: https://github.com/zigpy/zigpy/discussions/604 and https://github.com/home-assistant/core/issues/43969

Thanks in advance.

Mvh  Mattias W

MattWestb, 2021-03-03:
24 hour sniffing of paired 5 button remote with  broken attribute reporting but with check in.
= Battery draining.
https://github.com/zigpy/zigpy/discussions/604#discussioncomment-424835

MattWestb, 2021-03-03:
I have looking thru 10 sniffs of paring IKEA remotes and the normal is:
Setting up pull control but its not working.
Setting up attribute reporting of battery % and V and normally V is not working.

The behavior is known: Online one time in 24 hours battery reporting is bad but batteries is not being drained (if not having the Sonoff bug).

If reconfigure the device:
Pull control is working.
Attribute reporting is working OK both % and V every 3 hours.
Last seen every 55 minutes = online.

The behavior is that all is working very well but is draining its batteries.

MattWestb, 2021-03-04:
Pleas see https://github.com/zigpy/zigpy/discussions/604#discussioncomment-429274

"Fast long pull is killing IKEA remotes in 24 hours !!"

If need more information say wot you need and i trying to do it for you.

MattWestb, 2021-03-06:
Direct children is going off like is one EZSP coordinator bug not one IKEA bug !!!
And have long pull interval 6 seconds instead of 6 minuter is rising its  happens 60 times often !!

Latest analyse of marathon sniff: https://github.com/zigpy/zigpy/discussions/604#discussioncomment-436329

zoic21, 2021-03-09:
Hello,
I follow the subject with interest, when I read ZCL specification I can see that : 
[image]
These are not the parameters currently used. I will make some test with that. 

I also add variable parameter by device (I'am not on HA) to test, for exemple for ikea remote I pass to 1160 the long poll

And last think I also test with ZCL recommandation commande : 
[image]

I set 3 parameters (fast timeout, long poll and short poll) then I send stop fast poll.

MattWestb, 2021-03-09:
You shall setting up all the parameters OK for getting it working with all devices. deCONZ was having problems with smartthings devices that was being locked in fast pull because the long pull interval was not OK sett in the device and cud not going down to long pull after time out.

If reading more documents for device development shall the long pull being "adjusted" for the device type = wot IKEA have doing.
its in one link to the document in the thread but i dont remember ho was posing it but  i have reading it  2. 

I think IKEA devices is working god with 1160 as long pull. I was trying with 8 minutes (if IKEA devices is requesting on end device timeout = 8 minutes) but then they was starting rejoining all the time then doing pulls so you dont need trying so long time ;-)

If you like diving in little sniffs i have doing some hundred hours and some of then is posted in https://github.com/zigpy/zigpy/discussions/604.

Also sniffs from IKEA GW  with remote so you can see how IKEA is doing it (but no raw data then its production system and i dont like putting the keys for them online).

My conclusion is that all is one Silabs problem with the EZSP stack.
IKEA lights is being blocking state if sending commands to fast to them (then the device is dimmer up or down and so on) and its the same with the coordinator its not reacting of its children or neighbors if its "doing somthing else"  that is making the device leaving and trying rejoining. Therefor its working better with routers then the coordinator then one router is not leaving so fast its trying other routes if its possible.

Our cod master Adminiuga was not liking implanting setting the long pull time in the quirk but hi promises implanting it on the channel with device filter and i hope its working well then its coming online in the future.

If you need sniff from some system and / or devices that i having only say it and i do it for you !!

MattWestb, 2021-03-09:
I think you is 110% right with the 1160 for all controlling devices !!
More "active" end device (sleeping or not) its little different like IKEA blinds that need receiving commands for working.

Pulling one Philips HUE motion sensor that is only sending updated of its sensor readings every 5 seconds as ZHA is doing (not verified its doing  it but i think they do) is . . .

I have 2 old and 2 new IKEA motion sensors with one of etch on IKEA GA and the other 2 on ZHA (was on deCONZ before) but i have not digging in them but  the old model is not Zigbee 3 its (the only left IKEA controller device) ZLL and pull control is one Zigbee 3 (or more correct Zibee PRO) thing.
And i think there parent need supporting it (pull control) for it shall working.

Than i have getting this problem working OK i can do more digging with the motion sensors.

zoic21, 2021-03-10:
I continue my improvement on that part, last of it : don't resend long/short poll if no change on it. Here the code (don't work for HA but I think it can be adapt) : 

```
def cluster_command(cluster, tsn, *args):
		"""Handle commands received to this cluster."""
		logging.debug("["+str(cluster.endpoint.device._ieee)+"][chanels.general.PollControl.cluster_command] Received %s tsn : %s", tsn, args)
		cmd_name = cluster.client_commands.get(args[0], [])[0]
		logging.debug("["+str(cluster.endpoint.device._ieee)+"][chanels.general.PollControl.cluster_command] Command %s", cmd_name)
		if cmd_name == "checkin":
			fast_poll_timeout = PollControl.CHECKIN_FAST_POLL_TIMEOUT
			long_poll = PollControl.LONG_POLL
			short_poll = PollControl.SHORT_POLL
			ieee = str(cluster.endpoint.device._ieee)
			if ieee in shared.DEVICE_SPECIFIC and 'poll_control' in shared.DEVICE_SPECIFIC[ieee]:
				if 'long_poll' in shared.DEVICE_SPECIFIC[ieee]['poll_control'] and shared.DEVICE_SPECIFIC[ieee]['poll_control']['long_poll'] != '' :
					long_poll = int(shared.DEVICE_SPECIFIC[ieee]['poll_control']['long_poll'])
				if 'fast_poll_timeout' in shared.DEVICE_SPECIFIC[ieee]['poll_control'] and shared.DEVICE_SPECIFIC[ieee]['poll_control']['fast_poll_timeout'] != '' :
					fast_poll_timeout = int(shared.DEVICE_SPECIFIC[ieee]['poll_control']['fast_poll_timeout'])
				if 'short_poll' in shared.DEVICE_SPECIFIC[ieee]['poll_control'] and shared.DEVICE_SPECIFIC[ieee]['poll_control']['short_poll'] != '' :
					short_poll = int(shared.DEVICE_SPECIFIC[ieee]['poll_control']['short_poll'])
			utils.initSharedDeviceData(cluster,'long_poll')
			utils.initSharedDeviceData(cluster,'short_poll')		
			if shared.DEVICES_DATA[cluster.endpoint.device._ieee][cluster.endpoint._endpoint_id][cluster.cluster_id]['long_poll'] == long_poll and shared.DEVICES_DATA[cluster.endpoint.device._ieee][cluster.endpoint._endpoint_id][cluster.cluster_id]['short_poll'] == short_poll:
				logging.debug("["+ieee+"][chanels.general.PollControl.cluster_command] No checkin update needed")
				asyncio.ensure_future(cluster.checkin_response(False, fast_poll_timeout, tsn=tsn))
				return True
			shared.DEVICES_DATA[cluster.endpoint.device._ieee][cluster.endpoint._endpoint_id][cluster.cluster_id]['long_poll'] = long_poll
			shared.DEVICES_DATA[cluster.endpoint.device._ieee][cluster.endpoint._endpoint_id][cluster.cluster_id]['short_poll'] = short_poll
			asyncio.ensure_future(PollControl.set_poll_parameters(cluster,tsn,fast_poll_timeout,long_poll,short_poll))
			return True
		return False
		
	async def set_poll_parameters(cluster,tsn,fast_poll_timeout,long_poll,short_poll):
		ieee = str(cluster.endpoint.device._ieee)
		logging.debug("["+ieee+"][chanels.general.PollControl.set_poll_parameters] Send checkin response. Fastpoll timeout : %s s, long poll %s s, short poll %s s",(fast_poll_timeout/4),(long_poll/4),(short_poll/4))
		await cluster.checkin_response(True, fast_poll_timeout, tsn=tsn)
		await cluster.set_long_poll_interval(long_poll)
		await cluster.set_short_poll_interval(short_poll)
		await cluster.fast_poll_stop()
		logging.debug("["+ieee+"][chanels.general.PollControl.set_poll_parameters] End of checkin response")
```

And I confirm I contact some manufacturer (Danfoss and Develco) and for each device there is specific configuration for short/long poll so it's note possible to have same configuration everywhere.

zoic21, 2021-03-12:
Hello,

Yes I have 3 ikea remote pairing on 2 differents key (deconz and elelabs) it's work without any issue with 1160 of long poll

MattWestb, 2021-03-12:
I was by mistake setting the long pull time t 8 minutes (its wot IKEA device is requesting then end device timeout  request) and all IKEA remotes was start rejoining the network like crazy. Was putting in the 4.88 min (1160) and all is being  nice  :-))

I have seen very strange behavior then the coordinator is not acking frames from one device but looks doing it from other ones https://github.com/zigpy/zigpy/discussions/604#discussioncomment-470496.

Its importing have some routers around so the end device can jumping to them if the coordinator have one lazy period than they is not leaving so easy (i preferring having  end devices connected to routers its making the mesh working better and the device can using its functionality).

MattWestb, 2021-03-15:
I can confirming long pull 1160, EZSP 6.7.8.0 for Silabs coordinators and some routers around is 110% working !!!
Remotes can changing its parent but they not leaving the network and is reporting all configured attributes OK every 3 hours.

Perhaps i must trying some french software ;-)))

zoic21, 2021-03-16:
I don't known if french software is better, i think HA can include this modification quickly it's big improvement for battery saving

MattWestb, 2021-03-16:
Braking pull control is very bad then in Zigbee 3 then its one very elegant way to sending commands to SEDs with delay (max the check in interval) but not missing the commands and using long sleeping periods in the SED.

Very god description in the link in the post:
https://github.com/home-assistant/core/pull/47567#issuecomment-800110108

On the first i have no experience and on the second i agree 100%.
Best doing it in the quirk so its being flexible (I is running with 1160 on all now).

MattWestb, 2021-05-15:
@zoic21 Do you have trying using the device check in for sending data / commands to sleeping SED ?
ZHA is not using it and some routines can being triggered then the SED is sleeping and cant receiving the commands and getting time outs.

Also ZB3 SED is normally setting end device time out in its parent (if its parent is ZB3 one) then joining / jumping to it and the long pull interval must being stetted shorter then it or the SED is doing one rejoining all the time its trying communicating with its parent = eating batteries.

zoic21, 2021-05-15:
Hello,

I'am sur to understand. For me the Long pull is the time between 2 wake-up calls of the SED to know if there are orders for him.

So if you set this witch too high value for me there is two problem : 
- for some active device like valve the command can take the long poll time to be execute. Example if you set long poll at 5min on valve when you send a new setpoint it can take 5min to be execute (if zigpy don't throw a timeout before. On official zigbee documentation we can see it's recommande to set it at 5s but some manufacturer recommand another value (like Danfoss where its 8s)
- some device can leave the network if the value is too high 

On Jeedom I use this code : 

```
async def set_poll_parameters(cluster,tsn,fast_poll_timeout,long_poll):
		ieee = str(cluster.endpoint.device._ieee)
		logging.debug("["+ieee+"][chanels.general.PollControl.set_poll_parameters] Send checkin response. Fastpoll timeout : %s s, long poll %s s",(fast_poll_timeout/4),(long_poll/4))
		await cluster.checkin_response(True, fast_poll_timeout, tsn=tsn)
		if long_poll != -1 :
			await cluster.set_long_poll_interval(long_poll)
		await cluster.fast_poll_stop()
		logging.debug("["+ieee+"][chanels.general.PollControl.set_poll_parameters] End of checkin response")
```

I store value on separe json to send correct value by device. And stop fast poll it no update needed, for me it's exactly what the diagram describe : 
[image]

But some user have battery draining on ikea device so I recently add long poll -1 value. It prevent to send a value for long poll and leave the device to use its default value. I need to test it more but it's my last idea for the battery draining...

jmvaz, 2021-09-21:
Same problem on my side, on september releases. I use zha with conbee 2 adapter, and Symphony remotes last only 2 weeks.... 👎

MattWestb, 2021-09-21:
@zoic21 I using direct children = 0 on EZSP after adding (the first) router (cant joining the first router if its zero from start) and its forcing all end devices using routers and not the coordinator and looks working very well plus ZHA have fixing the long pull interval so its using IKEA standard (not changing it in the device) and also sending fast pull stop then the coordinator is not having any thing more to send to the device then its have doing the check in.

I dont knowing if its possible doing the same config on other radios in Zigpy but bellows its possible doing in the HA config.

@jmvaz My 2 symphonic  is still on original batteries an i was baying the first in the end of the winter (between the lock downs) so somthing is not playing well in some systems like yours. One in my test setup and on in my production system (the last is only 2 months old and was getting it in recycle hub for 9 €).

closed 2022-03-27 by github-actions[bot]