zigpy/zigpy #680: Can Concentration Cluster (0x040d) be added to measurement.py ?

B-Hartley, 2021-03-02:
Hi,

I've recently got a  [Titan Zigbee 3.0 C02 sensor](https://titanproducts.com/zigbee-3-0-co2-sensor/)
It senses temperature, humidity and CO2 level.

I'm trying to use it with Home Assistant.
The temp and humidity work fine, but CO2 doesn't work.

Initially I raised a [Device Support Request Issue](https://github.com/zigpy/zha-device-handlers/issues/792), but I think it's actually the measurement type that needs adding to Zigpy.

Looking at clusters/measurement.py it doesn't seem to be listed.
Is someone able to add 0x040d - Concentration Cluster ?

Or should it just be handled through a Device Handler / quirk ?

The technical manual for the product is at:
[Endpoint Cluster Attribute Manual](https://github.com/zigpy/zha-device-handlers/files/6054963/TitanProducts_TPZ3_0_Endpoint_Cluster_Attribute_Manual_v1.3.pdf)

I don't really know what I'm doing, but is it something like this............

```
class ConcentrationMeasurement(Cluster):
    cluster_id = 0x040d
    name = "Concentration Measurement"
    ep_attribute = "concentration"
    attributes = {
        # Concentration Measurement
        0x0000: ("measured_value", t.float32),    #Float Point, Single-Precision
        0x0001: ("min_measured_value", t.float32),   #Float Point, Single-Precision
        0x0002: ("max_measured_value", t.float32),    #Float Point, Single-Precision
#        0xFFFD: ("cluster_revision", t.uint16_t),    # Unsigned 16-bit Int

    }
    server_commands = {}
    client_commands = {}
```
I'm probably massively oversimplifying what is involved. Sorry !

The Device signature is........
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4467, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x040d"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0307",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0405"
      ],
      "out_clusters": []
    }
  },
[... 79 lines cut ...]
[0xA43A:1:0x0020]: finished channel configuration
[0xA43A:1:0x0019]: initializing channel: from_cache: False
[0xA43A:1:0x0019]: finished channel configuration
[0xA43A:2:0x0000]: initializing channel: from_cache: False
[0xA43A:2:0x0000]: finished channel configuration
[0xA43A:2:0x0405]: initializing channel: from_cache: False
[0xa43a:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=129 command_id=Command.Read_Attributes_rsp>
[0xA43A:1:0x0402]: finished channel configuration
[0xa43a] Delivery error for seq # 0x83, on endpoint id 1 cluster 0x0001: message send failure
[0xA43A:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: [0xa43a:1:0x0001]: Message send failure
[0xa43a] Delivery error for seq # 0x85, on endpoint id 2 cluster 0x0405: message send failure
[0xA43A:2:0x0405]: failed to get attributes '['measured_value']' on 'humidity' cluster: [0xa43a:2:0x0405]: Message send failure
[0xA43A:2:0x0405]: finished channel configuration
[0xA43A:2:0x0000]: 'async_initialize' stage succeeded
[0xA43A:2:0x0405]: 'async_initialize' stage succeeded
[0xa43a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=136 command_id=Command.Read_Attributes_rsp>
[0xA43A:1:0x0001]: finished channel configuration
[0xA43A:1:0x0402]: 'async_initialize' stage succeeded
[0xA43A:1:0x0001]: 'async_initialize' stage succeeded
[0xA43A:1:0x0000]: 'async_initialize' stage succeeded
[0xA43A:1:0x0020]: 'async_initialize' stage succeeded
[0xA43A:1:0x0019]: 'async_initialize' stage succeeded
[0xA43A](TPZRCO2HT-Z3): power source: Battery or Unknown
[0xA43A](TPZRCO2HT-Z3): completed initialization
```

B-Hartley, 2021-03-02:
Sorry, I've been informed that this isn't a zigPy issue as it isn't in 
ZCL revision 7

Sorry !

Adminiuga, 2021-03-04:
This cluster part of the new clusters defined in ZCL revision 7

B-Hartley, 2021-03-05:
does sensor.py need updating to include these new types as well ?
Will the values be shown in the entity as parts per 1 ?  or will they get converted to ppm or % or something else ?

Adminiuga, 2021-03-05:
Zigpy is low-level, it would send whatever the device is sending. ZCL specs are fuzzy in this regards.

Adminiuga, 2021-03-05:
Yes, zhan would need to be updated too.

B-Hartley, 2021-03-05:
ok, I've manually patched zigpy on my installation.
I've added......

```
class _ConcentrationMixin:
    """Mixin for the common attributes of the concentration measurement clusters"""

    attributes = {
        0x0000: ("measured_value", t.Single),  # fraction of 1 (one)
        0x0001: ("min_measured_value", t.Single),
        0x0002: ("max_measured_value", t.Single),
        0x0003: ("tolerance", t.Single),
    }

    server_commands = {}
    client_commands = {}

class CarbonDioxideConcentration(_ConcentrationMixin, Cluster):
    cluster_id = 0x040D
    name = "Carbon Dioxide (CO2) Concentration"
    ep_attribute = "carbon_dioxide_concentration"
```

I'm on core-2021.3.1
Home Assistant OS
supervisor-2021.03.4

I've dropped the quirk for now, as I'm not sure if it is required now zigpy has been updated.
I've got my custom component version of zha and in sensor.py I have added:

```
@STRICT_MATCH(channel_names="carbon_dioxide_concentration")
class CarbonDioxideConcentration(Sensor):
    """Carbon Dioxide Concentration sensor."""

    SENSOR_ATTR = "measured_value"
    _decimals = 0
    _multiplier = 1e6
    _unit = "ppm"
```

in measurement.py I have added:

```
@registries.ZIGBEE_CHANNEL_REGISTRY.register(
    measurement.CarbonDioxideConcentration.cluster_id
)
class CarbonDioxideConcentration(ZigbeeChannel):
    """Gas Concentration measurement channel."""

    REPORT_CONFIG = [
        {
            "attr": "measured_value",
            "config": (REPORT_CONFIG_MIN_INT, REPORT_CONFIG_MAX_INT, 0.000001),
        }
    ]
```

Pairing Log is now.........

```

[0x0000:zdo]   ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
--
[0x0f66:1:0x0000] ZCL request 0x000a:   [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=67>)]]
[0x0f66:1:0x0000] Attribute report   received: app_version=67
Device 0x0dd9   (00:15:8d:00:03:87:7f:49) joined the network
[0x0dd9:zdo] ZDO request   ZDOCmd.Device_annce: [0x0DD9, 00:15:8d:00:03:87:7f:49, 128]
[0x0dd9] Requesting 'Node   Descriptor'
[0x0dd9] Extending timeout for   0xd9 request
[... 9 lines cut ...]
[0x0dd9:1] Manufacturer: Titan   Products Ltd
[0x0dd9:1] Model: TPZRCO2HT-Z3
[0x0dd9:2] Discovering   endpoint information
[0x0dd9] Extending timeout for   0xe1 request
[0x0dd9:2] Discovered endpoint   information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260,   device_type=775, device_version=1, input_clusters=[0, 1, 3, 1029],   output_clusters=[])
Checking quirks for Titan   Products Ltd TPZRCO2HT-Z3 (00:15:8d:00:03:87:7f:49)
[0x0dd9:1:0x0402] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=240   command_id=Command.Configure_Reporting_rsp>
[0x0dd9:1:0x0001] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=242   command_id=Command.Configure_Reporting_rsp>
[0x0dd9:1:0x0001] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=251   command_id=Command.Configure_Reporting_rsp>
[0x0dd9] Extending timeout for   0x01 request
[0x0dd9:1:0x0020] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=253   command_id=Command.Write_Attributes_rsp>
[0x0dd9:2:0x0405] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=3   command_id=Command.Configure_Reporting_rsp>
[0x0dd9:1:0x0003] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=5   command_id=Command.Default_Response>
[0x0dd9:1:0x0402] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=7   command_id=Command.Read_Attributes_rsp>
[0x0dd9:1:0x0001] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=9   command_id=Command.Read_Attributes_rsp>
[0x0dd9:2:0x0405] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=11   command_id=Command.Read_Attributes_rsp>
[0x0dd9:1:0x0001] ZCL   deserialize: <ZCLHeader frame_control=<FrameControl   frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True   disable_default_response=True> manufacturer=None tsn=18   command_id=Command.Read_Attributes_rsp>
```

I've attached debug log.

What should I do next?

Thanks.
[debug-log.txt](https://github.com/zigpy/zigpy/files/6090737/debug-log.txt)

closed 2021-03-05 by Adminiuga

Hedda, 2021-03-08:
Please feel free to continue the related discussion about Concentration Measurements in https://github.com/zigpy/zigpy/discussions/691

Adminiuga, 2021-03-26:
@B-Hartley how long does it last on one full battery charge?

B-Hartley, 2021-03-26:
It takes a lithium aa battery. Not rechargeable. 
I haven’t had it long enough to know battery life. But I’d expect a good while !

Adminiuga, 2021-03-26:
ah, thanks 👍  For some reason I thought it was a 18650 cell.

PR #683 (2021-03-05): Implement ZCL 7 concentration measurement clusters
Defines a bunch of similar concentration measurement clusters (`0x040C` through `0x042B`). The names included chemical formulas with subscripts so I've replaced the ASCII digits with the appropriate characters (`O2` -> `O₂`).

Fixes #680.