zigpy/zigpy #681: Exception SerialTransport._read_ready()

AlexMarent, 2021-03-03:
Hi, 
since a few weeks my zigbee network partly stops working after a few hours or days. Mainly the states are not updating (e.g. motion sensors or lights). After restarting home assistant everything is fine. I see the following two exceptions in the log file: 

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:93 
First occurred: 8:55:22 PM (1 occurrences) 
Last logged: 8:55:22 PM

Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 119, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/uart.py", line 91, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 359, in data_received
    fut.set_result(data)
asyncio.exceptions.InvalidStateError: invalid state
```

```
Logger: zigpy.zdo
Source: /usr/local/lib/python3.8/site-packages/zigpy/util.py:293 
First occurred: 8:55:22 PM (1 occurrences) 
Last logged: 8:55:22 PM

[0x7d44:zdo] Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 305, in _command return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT) File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for raise exceptions.TimeoutError() asyncio.exceptions.TimeoutError During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 192, in request result, msg = await self._application.request( File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 238, in request await self._api.aps_data_request( File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 504, in aps_data_request return await self._command( File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 310, in _command self._awaiting.pop(seq) KeyError: 105
```

I hope someone can help me.

Alex

prigal, 2021-03-12:
Hi, I think I was in the same situation and it totally stop working a week ago.

Details here : https://community.home-assistant.io/t/raspbee-ii-zha-couldnt-start-deconz-error/289008

Error : 

```
2021-03-12 19:02:24 WARNING (MainThread) [zigpy_deconz.api] No response to 'Command.read_parameter' command with seq id '0x02'
2021-03-12 19:02:24 ERROR (MainThread) [zigpy.application] Couldn't start application
2021-03-12 19:02:24 ERROR (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start deCONZ = dresden elektronik deCONZ protocol: ConBee I/II, RaspBee I/II coordinator
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 157, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py", line 66, in startup
    self.version = await self._api.version()
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 444, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 409, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py", line 305, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

MattWestb, 2021-03-12:
Is modem manager installed and is the RaspBee using the hardware or the software  port ?

AlexMarent, 2021-03-12:
Modem manager is not installed. I'm using the Conbee II stick, so it should be a software port?!?

MattWestb, 2021-03-12:
No  but but some was writing RaspBee that is using  the internal comport of the PI.

Check wot comports sou have in your system and make sure you have configured it OK

AlexMarent, 2021-03-12:
Physically only the Conbee II is connected. On the PI only a few DockerContainer are running (HA, DuckDNS, MQTT, Portainer). How I shall check the comports?

prigal, 2021-03-13:
Solved on my side : https://community.home-assistant.io/t/raspbee-ii-zha-couldnt-start-deconz-error/289008

MattWestb, 2021-03-13:
That was the hardware UART problem with PIs !

Great that you was getting it working !!

AlexMarent, 2021-03-13:
But I think my issue is not the caused by this problem? I don't need to restart the PI, just the HA to solve the issue.

MattWestb, 2021-03-13:
@AlexMarent Have you looking with dmesg in one terminal if your USB dongle have reconnected ?

If using supervisor: Supervisor, system, log provider changing to host and you is getting the dmesg from your host system.

I was disconnecting one USB-TTL for some days ago and its looks like  this:
```
[3979505.207659] usb 1-1.3: USB disconnect, device number 13
[3979505.210296] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[3979505.210566] ftdi_sio 1-1.3:1.0: device disconnected
```
And if your Cornbee you is getting it as device type / name.

AlexMarent, 2021-03-13:
Ok, I think thats the problem. Just looked at the dmesg output - my Conbee is reconnecting very often. I will wait for the next time this problem happens, and will check if Conbee has disconnected.

MattWestb, 2021-03-13:
In one native system ZHA / zigpy-deconz have taking care or it and concreted to the Cornbee but Docke cant doing that so ZHA is not getting the contact to the comport until the container is restarted.

Do you have some:
```
[461590.017711] Under-voltage detected! (0x00050005)
[461594.177889] Voltage normalised (0x00000000)
```
If yes trying one  new power supply for your PI.

Also can USB ports having  bad connector or the extraction cable is not OK.

Then you getting the USB stable i think your system is 2.

AlexMarent, 2021-03-13:
I will try a different USB port and a different extension cable, maybe there's the problem. There is no under voltage fault at all. 

```
[Thu Mar 11 17:38:03 2021] usb 1-1.4: USB disconnect, device number 55
[Thu Mar 11 17:38:03 2021] usb 1-1.4: new full-speed USB device number 56 using xhci_hcd
[Thu Mar 11 17:38:03 2021] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 11 17:38:03 2021] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 11 17:38:03 2021] usb 1-1.4: Product: ConBee II
[Thu Mar 11 17:38:03 2021] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 11 17:38:03 2021] usb 1-1.4: SerialNumber: DE2140693
[Thu Mar 11 17:38:03 2021] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[Thu Mar 11 17:38:07 2021] usb 1-1.4: USB disconnect, device number 56
[Thu Mar 11 17:38:07 2021] usb 1-1.4: new full-speed USB device number 57 using xhci_hcd
[Thu Mar 11 17:38:07 2021] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 11 17:38:07 2021] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 11 17:38:07 2021] usb 1-1.4: Product: ConBee II
[Thu Mar 11 17:38:07 2021] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 11 17:38:07 2021] usb 1-1.4: SerialNumber: DE2140693
[Thu Mar 11 17:38:07 2021] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

MattWestb, 2021-03-13:
Very likely is some "mechanical" problem but you never knowing. 
Also updating your CornBee firmware then they have doing some work on the watchdog in the firmware.

I have undervoltage  warnings then i using USB ports for powering ESP32-CAM and other thhings but ZHA is working OK but i using the GPIO UART to my "Billy EZSP".

AlexMarent, 2021-03-20:
Firmware of the ConBee is up to date. Changing the USB Port did not help, today I had a really bad crash - all sensors (motion, temperature, door,...) were unavailable. Maybe the log attached will help. It also looks like docker has a problem. 

```
[Thu Mar 18 18:20:07 2021] usb 1-1.2: USB disconnect, device number 5
[Thu Mar 18 18:20:07 2021] usb 1-1.2: new full-speed USB device number 6 using xhci_hcd
[Thu Mar 18 18:20:07 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 18 18:20:07 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 18 18:20:07 2021] usb 1-1.2: Product: ConBee II
[Thu Mar 18 18:20:07 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 18 18:20:07 2021] usb 1-1.2: SerialNumber: DE2140693
[Thu Mar 18 18:20:07 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Thu Mar 18 18:20:08 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:11 2021] usb 1-1.2: USB disconnect, device number 6
[Thu Mar 18 18:20:11 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:11 2021] usb 1-1.2: new full-speed USB device number 7 using xhci_hcd
[Thu Mar 18 18:20:11 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 18 18:20:11 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 18 18:20:11 2021] usb 1-1.2: Product: ConBee II
[Thu Mar 18 18:20:11 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 18 18:20:11 2021] usb 1-1.2: SerialNumber: DE2140693
[Thu Mar 18 18:20:11 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Thu Mar 18 18:20:12 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:14 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:16 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:19 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:22 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:23 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:27 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:27 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:28 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:29 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:30 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:30 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:32 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:32 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:33 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:34 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:34 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:35 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:36 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:36 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:37 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:37 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:38 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:39 2021] broken atomic modeset userspace detected, disabling atomic
[Thu Mar 18 18:20:46 2021] usb 1-1.2: USB disconnect, device number 7
[Thu Mar 18 18:20:46 2021] usb 1-1.2: new full-speed USB device number 8 using xhci_hcd
[Thu Mar 18 18:20:46 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 18 18:20:46 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 18 18:20:46 2021] usb 1-1.2: Product: ConBee II
[Thu Mar 18 18:20:46 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 18 18:20:46 2021] usb 1-1.2: SerialNumber: DE2140693
[Thu Mar 18 18:20:46 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Thu Mar 18 18:20:49 2021] usb 1-1.2: USB disconnect, device number 8
[Thu Mar 18 18:20:50 2021] usb 1-1.2: new full-speed USB device number 9 using xhci_hcd
[Thu Mar 18 18:20:50 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 18 18:20:50 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 18 18:20:50 2021] usb 1-1.2: Product: ConBee II
[Thu Mar 18 18:20:50 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Thu Mar 18 18:20:50 2021] usb 1-1.2: SerialNumber: DE2140693
[Thu Mar 18 18:20:50 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Thu Mar 18 18:21:26 2021] usb 1-1.2: USB disconnect, device number 9
[Thu Mar 18 18:21:26 2021] usb 1-1.2: new full-speed USB device number 10 using xhci_hcd
[Thu Mar 18 18:21:26 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Thu Mar 18 18:21:26 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Mar 18 18:21:26 2021] usb 1-1.2: Product: ConBee II
[... cut ...]
[Fri Mar 19 12:32:05 2021] usb 1-1.2: new full-speed USB device number 33 using xhci_hcd
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Product: ConBee II
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Mar 19 12:32:06 2021] usb 1-1.2: SerialNumber: DE2140693
[Fri Mar 19 12:32:06 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
```

MattWestb, 2021-03-20:
I agree with you that docker is doing strange things !
It closing your network connections to the host system little now and then.
I cant saying if the 2 things is connected or if its separate issues without any connections.
Docker with USB can normally not reconnecting the the USB com port and need restarting for getting it working a gen so ZHA cant communicating with your CornBee until the container is restarted. 
Latest version of HA is having getting it working better but ii dont knowing if its 100%.

Have you the OS up to date so the docker core is updated ?

One common trouble maker is slow / bad SD-Card that is making all things working too slow and timing out and that can doing may strange problems.

MattWestb, 2021-03-20:
One nearly some kind of problem with CornBee in ZHA.

The devs is trying fixing one timing problem but i dont knowing is relevant in your situation.

The user is also having problem with USB reconnecting but its looks being software triggered.

AlexMarent, 2021-03-23:
Everything is updated, also the firmware of the ConBee. Don't think that here is the problem. Looks like there is a very similar problem here [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4029#issuecomment-804029130](url)

Adminiuga, 2021-04-17:
```
[Fri Mar 19 12:10:13 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Fri Mar 19 12:32:01 2021] usb 1-1.2: USB disconnect, device number 31
[Fri Mar 19 12:32:02 2021] usb 1-1.2: new full-speed USB device number 32 using xhci_hcd
[Fri Mar 19 12:32:02 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Mar 19 12:32:02 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Mar 19 12:32:02 2021] usb 1-1.2: Product: ConBee II
[Fri Mar 19 12:32:02 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Mar 19 12:32:02 2021] usb 1-1.2: SerialNumber: DE2140693
[Fri Mar 19 12:32:02 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[Fri Mar 19 12:32:05 2021] usb 1-1.2: USB disconnect, device number 32
[Fri Mar 19 12:32:05 2021] usb 1-1.2: new full-speed USB device number 33 using xhci_hcd
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Fri Mar 19 12:32:06 2021] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Product: ConBee II
[Fri Mar 19 12:32:06 2021] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[Fri Mar 19 12:32:06 2021] usb 1-1.2: SerialNumber: DE2140693
[Fri Mar 19 12:32:06 2021] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
```

Per log looks like device disconnects/reboots. This is going to cause it missing the commands and errors in the logs. Try with the latest version.

closed 2021-04-17 by Adminiuga