zigpy/zigpy #682: Reach an Error asyncio.excepion.TimeoutError

markicos, 2021-03-04:
Hello there i am having this problem when using the code example in https://github.com/zigpy/zigpy/issues/452#issuecomment-671441387 but i am reaching an exception.Timeout, i don`t know what i am doing wrong or if i miss some module.

[image]

I am using conbee II with zigpy-deconz

Anyone can help?

Adminiuga, 2021-03-04:
Belongs to zigpy-deconz.
Unknown how well serial ports work on windows.
It did not get a reply to the serial command send to the port. Is it correct port?

MattWestb, 2021-03-04:
Shall the  https://github.com/zigpy/zigpy-znp/commit/f85c281b7fd335ff6658d02154999fcfa4832d99 also being done for all other radios so they is not getting the same problem that @puddly  was finding ??

markicos, 2021-03-05:
Thank a lot to both of you, 

It is the correct port,
I downgraded the pyserial-asyncio and it works 👯,

MattWestb, 2021-03-05:
Thanks great to hearing !! 
But your thanks shall going to our @puddly but perhaps @Adminiuga need adding the "patch" to the other radios so they is working OK on windows machine or waiting for the bug being fixed in the lib.

closed 2021-03-05 by markicos