zigpy/zigpy #694: Clean up OTA exception handling in zigpy.ota.provider of Philips Hue images.

systemcrash, 2021-03-13:
I manually updated a few Philips devices with images in the OTA folder. The update was successful. The images in the OTA folder trigger these stack traces (more log noise):
```
2021-03-13 13:21:29 DEBUG (SyncWorker_0) [zigpy.ota.provider] ImageKey(manufacturer_id=4107, image_type=270): ota_update/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee, version: 16783620, hw_ver: (None, None), OTA string:
2021-03-13 13:21:29 DEBUG (SyncWorker_3) [zigpy.ota.provider] ImageKey(manufacturer_id=4107, image_type=268): ota_update/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee, version: 16783874, hw_ver: (None, None), OTA string:
2021-03-13 13:21:29 DEBUG (SyncWorker_5) [zigpy.ota.provider] ImageKey(manufacturer_id=4107, image_type=273): ota_update/100B-0111-01001800-ConfLight-ModuLum-EFR32MG13.zigbee, version: 16783360, hw_ver: (None, None), OTA string:
2021-03-13 13:21:29 DEBUG (SyncWorker_0) [zigpy.ota.provider] File 'ota_update/@eaDir/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee@SynoResource' doesn't appear to be a OTA image
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 259, in parse_ota_image
    return HueSBLOTAImage.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 198, in deserialize
    header, remaining_data = OTAImageHeader.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 109, in deserialize
    raise ValueError(
ValueError: Wrong magic number for OTA Image: 118883584

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/provider.py", line 302, in scan_image
    parsed_image, _ = parse_ota_image(f.read())
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 261, in parse_ota_image
    return OTAImage.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 159, in deserialize
    hdr, data = OTAImageHeader.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 109, in deserialize
    raise ValueError(
ValueError: Wrong magic number for OTA Image: 118883584
2021-03-13 13:21:29 DEBUG (SyncWorker_4) [zigpy.ota.provider] File 'ota_update/@eaDir/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee@SynoEAStream' doesn't appear to be a OTA image
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 259, in parse_ota_image
    return HueSBLOTAImage.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 198, in deserialize
    header, remaining_data = OTAImageHeader.deserialize(data)
  File "/usr/local/lib/python3.8/site-packages/zigpy/ota/image.py", line 109, in deserialize
    raise ValueError(
ValueError: Wrong magic number for OTA Image: 118883584
```

I propose to ignore `manufacturer_id=4107` for these checks. Those images might not have 'standard' magic numbers, but the images are 'correct'. I've read elsewhere that Philips/Signify follow no clear standard in their image structure, so attempting sanity checks on them may well be in vain. 

Keeping the images around is a good idea for when future devices get added - the images are once again useful, so avoiding these tracebacks is an improvement.

systemcrash, 2021-03-13:
Not sure how best to approach this problem - since IKEA and LEDVANCE seem to be the only sanctioned and reliable OTA types, perhaps a whitelist (static for now) of the few `manufacturer_id` which are know to follow the standards, for which the above checks are implemented, is a good way.

systemcrash, 2021-03-13:
A closer  HEX inspection of image `100B-0111-01001800-ConfLight-ModuLum-EFR32MG13.zigbee` and others shows that Philips `manufacturer_id` can be pulled. The `manufacturer_id` `imageType` and `version` are all stamped in network byte order:

[image]

Sorry if this is common knowledge.

Adminiuga, 2021-03-13:
Dunno why it complains about the magic number, looks fine to me. Can you post a link from where you downloaded the update file?

systemcrash, 2021-03-13:
Sure. E.g. [here](https://github.com/Koenkk/zigbee-OTA/blob/master/index.json)

systemcrash, 2021-03-13:
Yeah - Magic looks OK. I guess this specific error is triggered as a result of what should actually be something else.

https://github.com/zigpy/zigpy/blob/4668c8d109d6625822858d1424b7f24a82861092/zigpy/ota/image.py#L57

systemcrash, 2021-03-13:
Hmm - what byte order is the header deserialized to? 
https://github.com/zigpy/zigpy/blob/4668c8d109d6625822858d1424b7f24a82861092/zigpy/ota/image.py#L111

puddly, 2021-03-13:
I don't think the file you linked to is the file zigpy is trying to parse:

```Python
>>> import pathlib
>>> data = pathlib.Path('~/Downloads/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee').expanduser().read_bytes()
>>> from zigpy.ota.image import *
>>> image, rest = parse_ota_image(data)  # parses fine
>>> rest  # no trailing data
b''
>>> image.serialize() == data  # is parsed and serialized perfectly
True
>>> from zigpy.ota.validators import validate_ota_image
>>> validate_ota_image(image)  # is an unknown format but parses fine
<ValidationResult.UNKNOWN: 2>
>>> type(image)
zigpy.ota.image.OTAImage
>>> [s.tag_id for s in image.subelements]
[<ElementTagId.undefined_0xf000: 61440>,
 <ElementTagId.UPGRADE_IMAGE: 0>,
 <ElementTagId.undefined_0xf100: 61696>,
 ...]
>>> (118883584).to_bytes(4, 'little') in data  # !!!
False
>>> (118883584).to_bytes(4, 'big') in data  # !!!
False
```

puddly, 2021-03-13:
Actually, this makes sense. zigpy is parsing these `.zigbee@SynoResource` and `.zigbee@SynoEAStream` files. They look to be metadata for Synology devices, not actual OTA files.

systemcrash, 2021-03-13:
> Actually, this makes sense. zigpy is parsing these `.zigbee@SynoResource` and `.zigbee@SynoEAStream` files. They look to be metadata for Synology devices, not actual OTA files.

Good catch! 🤦 Yes - it's those. Hmm - perhaps we can mask those files out - specifically ignore those, given that Synology (perhaps others) NAS are a popular Docker platform for HA.

puddly, 2021-03-13:
Yeah. Or maybe have a more specialized error message for "definitely not an OTA file" so the traceback is not printed.

systemcrash, 2021-03-13:
> Yeah. Or maybe have a more specialized error message for "definitely not an OTA file" so the traceback is not printed.

Seems like less work, and less device specific. Masks and regexp can be hard...

systemcrash, 2021-03-13:
Although I reckon an ignore mask that can go in the config is a solid way forward, esp if someone rsyncs lots of img and sha1 and md5 files. Just a thought.

closed 2021-09-16 by github-actions[bot]