zigpy/zigpy #699: zha.remove service isn't removing a Zigbee Coordinator

mrneutron42, 2021-03-18:
zha.remove service isn't removing a Zigbee Coordinator, which was swapped/replaced

core-2021.3.4
Home Assistant OS 5.12
supervisor-2021.03.6

I switched from a Sonoff Zigbee Bridge (Silicon Labs EZSP) to a TI CC2531 USB stick and the Silicon Labs EZSP device.
I removed the ZHA integration (which is supposed to remove the Zigbee coordinator from the system files) and then added ZHA integration again, setting one of the TI CC2531 USB sticks as Zigbee coordinator.
The power is now pulled from the former Sonoff Zigbee Bridge, but the Silicon Labs EZSP device will not go away!
With the ZHA.remove command, I used the argument:  ieee: '60:a4:23:ff:fe:xx:xx:xx'
[image]
the result in the log:
`2021-03-17 17:29:55 INFO (MainThread) [homeassistant.components.zha.api] Removing the coordinator (60:a4:23:ff:fe:xx:xx:xx) is not allowed`

Can the ability to remove a Zigbee coordinator be added?
Or, could the ZHA integration be changed to it removes or disables the zigbee.db file when it is removed from HA?

walthowd, 2021-03-18:
You'll probably need to hack it out of `.storage/core.device_registry` and `.storage/zha.storage` manually with HA shut down. Take backups first and make sure you keep valid JSON in the files.

mrneutron42, 2021-03-18:
I tried that.  I followed the advice in this HA forum thread:
https://community.home-assistant.io/t/unable-to-delete-zigbee-device/224041
The zigbee coordinator device is repopulated in the .storage/core.device_registry file upon restart.
I have searched the zigbee.db file and the ieee id for the Sonoff Zigbee Bridge still has 20 instances in the file.  
It appears we need a way to remove a Zigbee coordinator from the zigbee.db file.

walthowd, 2021-03-18:
You can delete out of the .db file with sqlite3, or any SQLLite utility -- but shouldn't be required.  I would imagine it's still in your .storage somewhere?

Or, you can just ignore, it won't cause any issues.

MattWestb, 2021-03-18:
I was having the same problem then updating my Billy EZSP and was putting the wrong NCP back and was getting 2 coordinators.

Look  in all files in the `.storage/` and deleting if you finding the IEEE from the unwanted device and restarting HA

mrneutron42, 2021-03-18:
I also manually deleted the device entries from .storage/zha.storage file.  It returned after every HA restart.
That zha.storage file appears to be system generated, and is a reflection of the devices in the database (like a log file), so removing entries doesn't appear to have any effect.

I'm running HA as a pre-packaged virtual machine inside VMware esxi.  
I believe it's called a container or docker implementation?
Unless they are pre-packaged Integrations, I don't think I have the ability to load software packages, like I would if it were a Ubuntu machine with HA loaded as one of many packages.

MattWestb, 2021-03-18:
The record of your unwanted device is not in the pre packed integrations files, its in your ./config directory.

You can rename or delete the directory and you is getting one clean ZHA installation.

Adminiuga, 2021-03-18:
currently you option is to edit zigbee.db People were removing their active coordinators so a safeguard was introduced. The safeguard is in ZHA. Perhaps better approach for the safeguard is to check if the IEEE address matches ieee address of the ControllerApplication instance

dmulcahey, 2021-03-18:
Yeah we can definitely improve this.

Adminiuga, 2021-03-18:
At the time this was implemented, people replacing their coordinators was inconceivable :)

dmulcahey, 2021-03-18:
Yep. Progress!!

mrneutron42, 2021-03-18:
> At the time this was implemented, people replacing their coordinators was inconceivable :)

Fair enough!  Who would have known that the wifi-to-Zigbee methodology used by the Sonoff Zigbee Bridge would be so unreliable, that people want/need to get rid of it to gain Zigbee network reliability?  That's why I'm changing Zigbee coordinators - Zigbee devices drop off the network at random, and I'm sick of re-pairing Zigbee sensors once a week, etc.

mrneutron42, 2021-03-18:
> The record of your unwanted device is not in the pre packed integrations files, its in your ./config directory.

Yes, I am aware that the dynamically-generated Zigbee device lists/registries are created on the fly and are stored in the ./config directory.
I was mentioning the container/docker implementation to explain why I don't have the ability to load an SQLite database editor, and manually cut out the database entries for the Sonoff Zigbee Bridge device.

Adminiuga, 2021-03-18:
`zigbee.db` is in your `/config` folder. you could copy if offline, modify, stop HA and upload zigbee.db back.

mrneutron42, 2021-03-18:
> zigbee.db is in your /config folder. you could copy if offline, modify, stop HA and upload zigbee.db back.

I just tried several times.  Trying to edit the database broke HA so bad it would fully boot!  The GUI web interface would not start.  I was only able to access the CLI via my VMware esxi console screen.  
When trying to edit the database, I looked in the Devices table for a record specific to the Sonoff Zigbee Bridge, deleted it, and closed the database.  Via SAMBA I moved the altered database back into the /config folder of HA and restarted the HA virtual machine...HA broken.
For some reason, the CLI snapshot restore command did not work.  The CLI claimed that the command completed, but it did not replace the altered (broken) files.  
I had SAMBA access so I used 7zip to unzip a snapshot file in the /backup folder.  After a file replace and a restart, HA came back to working as it was earlier this afternoon.   After I got it working again, I took a VM snapshot in VMware esxi!

So, trying to edit the zigbe.db database didn't work as expected.  It sounded like a valid concept.

puddly, 2021-03-18:
You may not have deleted all the entries in all the tables. This is more or less what zigpy would run:

```
DELETE FROM attributes WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM neighbors WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM node_descriptors WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM clusters WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM output_clusters WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM group_members WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM endpoints WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
DELETE FROM devices WHERE ieee = '60:a4:23:ff:fe:xx:xx:xx';
```

Alternatively, change [`homeassistant/components/zha/api.py:899`](https://github.com/home-assistant/core/blob/314a5518f1b0043c74581970dceac20547b8615b/homeassistant/components/zha/api.py#L899) and remove the coordinator restriction.

Adminiuga, 2021-03-18:
Yeah, edditing the api and removing the safe check is probably easier.

mrneutron42, 2021-03-18:
>Alternatively, change homeassistant/components/zha/api.py:899 and remove the coordinator restriction. 

Arrrgh.  I don't have access to any of those directories and files.  
They are hidden inside the containers/dockers.
All I have easy access to are what I can see from SAMBA.
[image]

Adminiuga, 2021-03-18:
copy https://github.com/home-assistant/core/tree/2021.3.4/homeassistant/components/zha into your `/config/custom_components/zha`, make modifications, then restart hass

mrneutron42, 2021-03-19:
1. since I have ZHA installed as an integration, there is no /custom_configurations/zha folder.
2. if I manually create a /custom_configurations/zha folder, what will make HA look at the files in the /custom_configurations directory, rather than the files in the hidden containers/dockers?

Adminiuga, 2021-03-19:
not `custom_configurations` **but** `/config/custom_components` folder
The url is for web browser access. the repo is https://github.com/home-assistant/core and tag is 2021.3.4

mrneutron42, 2021-03-19:
What will make HA look at the files in the /custom_components directory, rather than the files in the hidden containers/dockers?

mrneutron42, 2021-03-28:
My situation has been resolved by the workaround.  Thanks.

Perhaps this can now be considered a Feature Request?

puddly, 2021-03-28:
Fixed a little over a week ago: https://github.com/home-assistant/core/pull/48147

mrneutron42, 2021-03-28:
Awesome!  Thank you!

closed 2021-03-28 by mrneutron42

mrneutron42, 2021-03-28:
I'm curious how this change will get rolled into installed Home Assistant systems?
Via the next Home Assistant Supervisor update, or via an update to the ZHA integration?

dmulcahey, 2021-03-28:
Next release of HA will have it