zigpy/zigpy #701: How to DIY zigbee device with STM32WB

jmmb13, 2021-03-21:
Hi,

I want to make my own devices compatible with the zigpy library, hence the home assistant. Can someone tell me how I can do it? I'm a little lost in the topic, I understand the zigbee stack, the profiles, but how can I create my board and connect dynamically? What characteristics do I need to follow in my own board? There is some documentation that I can follow?

Thanks for all your attention!!
jmmb

MattWestb, 2021-03-21:
If you is doing one own device you also need making one "application" to running on it.
The application you is doing with the SDK from the manufacture.
If your hardware is Silabs EFR32 and you like doing one Zigbee 3 light you using the Simplicity Studio and building one Zigbee 3 light and flashing it to the device and  hopefully is working well.

For getting access to the SDK for Silabs you need buying one dev kit that for around 100$ and registrating it for getting access to the Zigbee stack function.

TI, NXP, Nordic all have there own SDKs and you must looking how getting access to them and how much it costing and wot hardware you like to using.

I have ordering on Silabs dev kit but is not delivered yet (weekend).
Sirlabs have 2 different versions both for 100§ and both is giving Zigbee stack access.
Some feedback from user and devs https://github.com/zigpy/zigpy/discussions/692#discussioncomment-488667

jmmb13, 2021-03-21:
I bought this kit https://www.st.com/en/evaluation-tools/p-nucleo-wb55.html

And it implements the zigbee stack in one core and the application in the another one. It has a lot of ZigBee examples. I want to buy an zigbee stick to use with my homeassistant and make my custom devices but I couldn't find a guide how to do this connection, I'm not new to embedded systems, but I'm new to this area of wireless zigbee network. So wanted to do a home project with that.

MattWestb, 2021-03-21:
Its looks nice and have support for nearly all Zigbee clusters so shall being possible doing much fun things with !!

I think the easiest doing one on/off switch and trying it out with the dongle in one network without coordinator.

Butt if you like doing little more you need having on coordinator and putting the zigbee alliance link key in your project so they can talking with etch other.

I have not looking wot is possible doing in the IDE but i think is more your part then im more of hardware cooker :-))

jmmb13, 2021-03-21:
Thanks for the answers matt!! It is really nice to talk about it with someone. What ZigBee switch do you suggest?

Adminiuga, 2021-03-21:
In EU you could try getting a ConBee II from dresden electronics or maybe https://elelabs.com/products/elelabs-usb-adapter.html have shipping in EU.

For the device, just implement it  as a regular Zigbee device.

MattWestb, 2021-03-21:
Or building one "[IKEA Billy EZSP](https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module)" for 6€ then you have one SWD flasher on your dev board :-)
But its needs little more hands on . .

Hedda, 2021-03-22:
@jmmb13 if your device follow standard light or remote Zigbee specifications then should just work with zigpy, but if it does not then you might need to write a custom device handler for zha-device-handlers, see https://github.com/zigpy/zha-device-handlers

jmmb13, 2021-03-22:
> In EU you could try getting a ConBee II from dresden electronics or maybe https://elelabs.com/products/elelabs-usb-adapter.html have shipping in EU.
> 
> For the device, just implement it as a regular Zigbee device.

Should I invest in a stick with z-wave and zigbee to have if I need in the future?  The range is good? It looks so small

> @jmmb13 if your device follow standard light or remote Zigbee specifications then should just work with zigpy, but if it does not then you might need to write a custom device handler for zha-device-handlers, see https://github.com/zigpy/zha-device-handlers

Thanks you!! I will buy the stick and then do an on-off switch to test it. Thanks for the answer!

> Or building one "[IKEA Billy EZSP](https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module)" for 6€ then you have one SWD flasher on your dev board :-)
> But its needs little more hands on . .

I will check it. Thanks!!

Adminiuga, 2021-03-22:
>Should I invest in a stick with z-wave and zigbee to have if I need in the future?

Up to you. Zigbee hardware on HUSBZB-1 is old, no new firmwares will be published for it. But it does get work done. 

> The range is good? It looks so small

I'm running it in my production without any issues. Once you get past two zigbee routers in your network, range becomes a non issue.

Hedda, 2021-03-23:
> Should I invest in a stick with z-wave and zigbee to have if I need in the future? The range is good? It looks so small

Get seperater dedicated adapters, suggest checkout one or both of these:

https://www.itead.cc/zigbee-3-0-usb-dongle.html

https://electrolama.com/projects/zig-a-zig-ah/

closed 2021-03-29 by jmmb13

simon3panda, 2024-07-21:
> Hi,
>
> I want to make my own devices compatible with the zigpy library, hence the home assistant. Can someone tell me how I can do it? I'm a little lost in the topic, I understand the zigbee stack, [...]

Hi, recently I'M working on the same stuff, which to trying to make may own stm32wb55 developing board device compatible with zigpy library. It feels really great that knowing someone  had studied on the same topic. So I'm wondering if you finally make it or not? Is customized zigbee device based on stm32wb55 compatible with zigpy library?

Looking forward to your reply~

Hedda, 2024-07-24:
@simon3panda best is if you start a new discussion with specific question instead of posting in this closed issue which (as issues are really for specific bugs and feature requests):

https://github.com/zigpy/zigpy/discussions

Anyway, know that Zigbee gateway implementation that uses zigpy support most standard devices types, see:

https://www.home-assistant.io/integrations/zha#knowing-which-devices-are-supported

However you might need to add a ZHA Device Hander (also known as a quirk) to the ZHA integration if want to expose less common attributes to the UI interface in Home Assistant, see:

https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices

FYI, if you are using the ZHA library with ZHA Device Handlers then you can use quirks to work around attributes that are not mapped by default

https://github.com/zigpy/zha-device-handlers

ZHA/zigpy developers have recently introduced initial support for what they call ”quirks v2” and dmulcahey begun how-to documentation here:

https://github.com/zigpy/zha-device-handlers/pull/3019

Note! That documention only applies to ”quirks v2” (and not to how existing quirks work today, also referred to as ”quirks v1”), so for old quirk format see the readme:

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

There is also some additional linkshere:

https://github.com/zigpy/zha-device-handlers/discussions/2467

PS: If you however want to add support for a new device type then you might need to add that new device type first to the zha library (which will be used in next Home Assistant core release too):

https://github.com/zigpy/zha