zigpy/zigpy #702: The use of "detect_types=sqlite3.PARSE_DECLTYPES" in the sqlite3.connect command can cause SQL crash

Hedda, 2021-03-22:
Any chance that zigpy developers could look at this conflict issue around SQLITE3 usage in zigpy's use of sqlite3.connect commands?

@NjbBzz suggest zigpy should look at the use of sqlite3.PARSE_COLNAMES instead of sqlite3.PARSE_DECLTYPES for "detect_types"?

Another way is to study the use of type converters to convert Python types to SQL and vice-versa? See this page : 

https://docs.python.org/3/library/sqlite3.html#sqlite3-types (search for chapter : Converting SQLite values to custom Python types).

Backstory here is that little over 6-months ago @pipiche38 tried to make [a Zigpy plugin for Domoticz](https://github.com/pipiche38/Domoticz-Zigpy/) (an C++ based open source software that supports Python plugins) but unfortunately he got completly stuck on of [a showstopper conflict around SQLITE3 usage](https://github.com/domoticz/domoticz/issues/4312).

pipiche38 even managed to replicate and duplicate the issue without zigpy and created a small "[domoticz-sqlite3-issue](https://github.com/pipiche38/domoticz-sqlite3-issue)" plugin for Domoticz that just show this same issue but the progress on that got stuck as [Domoticz developers said the root cause was not on their side](https://github.com/domoticz/domoticz/issues/4312), and thus the effort on the mentioned Zigpy plugin for Domoticz was sadly eventually abandoned by pipiche38.

Now however another developer called @NjbBzz has troubleshot that same issue using the "[domoticz-sqlite3-issue](https://github.com/pipiche38/domoticz-sqlite3-issue)" plugin that pipiche38 made, and NjbBzz wrote that he thinks that he has managed to narrow it down to the issue being caused by the use of "detect_types=sqlite3.PARSE_DECLTYPES" in the sqlite3.connect command which crashes SQL in Domoticz due to conflict usage:

https://github.com/pipiche38/domoticz-sqlite3-issue/issues/1

> The cause of the crash is coming from the use of "detect_types=sqlite3.PARSE_DECLTYPES" in the sqlite3.connect command.
> I tested with removing it and there is no more crash.
> Yet, I assume that it's needed as you're using custom types ; I think you should have a look at the use of sqlite3.PARSE_COLNAMES istead for "detect_types" ; another way is to study the use of type converters to convert Python types to SQL and vice-versa : see this page : https://docs.python.org/3/library/sqlite3.html#sqlite3-types (search for chapter : Converting SQLite values to custom Python types)

https://github.com/pipiche38/domoticz-sqlite3-issue

https://github.com/domoticz/domoticz/issues/4312

https://github.com/pipiche38/Domoticz-Zigpy/blob/master/README.md#current-showstoppers

https://github.com/pipiche38/Domoticz-Zigpy/blob/master/README.md

https://github.com/pipiche38/Domoticz-Zigpy

Adminiuga, 2021-04-17:
no plans for implementing this as of now. But this is open source, if it is needed for domoticz plugin then I guess a custom fork would work.

closed 2021-04-17 by Adminiuga

puddly, 2021-04-17:
Yeah, this would require changing just about every single SQL query in `zigpy.appdb` (assuming the broken `sqlite3` module won't crash with `PARSE_COLNAMES` as well).

I believe this was discussed with the JSON persisting listener but nothing prevents an application using zigpy from subclassing `ControllerApplication` and replacing `_load_db` with a new method that loads a different implementation of "`PersistingListener`". You could maintain an in-tree clone of `zigpy.appdb` or even write your own persistence backend. Once I re-visit https://github.com/zigpy/zigpy/pull/521 writing a custom listener will be even more straightforward.