zigpy/zigpy #710: Old zigpy databases without foreign keys can have inconsistent `attributes` table state after restart

puddly, 2021-04-08:
A couple of users have complained that ZHA fails to start up after an update from `2021.3.x` to `2021.4.0` with the following error:

```Python
2021-04-08 20:58:25 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry HubZ for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 249, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 157, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 64, in new
    await app._load_db()
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 56, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 476, in load
    await self._load_attributes()
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 491, in _load_attributes
    dev = self._application.get_device(ieee)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 392, in get_device
    return self.devices[ieee]
KeyError: 90:fd:9f:ff:fe:71:71:70
```

Nothing interesting changed between those two HA releases so I believe this was "caused" by just restarting ZHA.

Here is one such database: [zigbee-attrs1.db.zip](https://github.com/zigpy/zigpy/files/6281140/zigbee-attrs1.db.zip)

Notably, there is in fact no device with that IEEE address, and yet a row for it appears in the `attributes` table:

```Console
$ sqlite3 ~/Downloads/zigbee-attrs1.db .dump | grep -i 90:fd:9f:ff:fe:71:71:70
INSERT INTO attributes VALUES('90:fd:9f:ff:fe:71:71:70',1,6,0,0);
INSERT INTO attributes VALUES('90:fd:9f:ff:fe:71:71:70',1,8,0,75);
```

I believe this is because these zigpy databases are from before the database schema was updated to include foreign key constraints, so the `device` entry being deleted did not also delete the linked `attributes` rows:

```Console
$ sqlite3 ~/Downloads/zigbee-attrs1.db .dump | grep -i 'CREATE TABLE'
CREATE TABLE devices (ieee ieee, nwk, status);
CREATE TABLE endpoints (ieee ieee, endpoint_id, profile_id, device_type device_type, status);
CREATE TABLE clusters (ieee ieee, endpoint_id, cluster);
CREATE TABLE node_descriptors (ieee ieee, value, FOREIGN KEY(ieee) REFERENCES devices(ieee));
CREATE TABLE output_clusters (ieee ieee, endpoint_id, cluster);
CREATE TABLE attributes (ieee ieee, endpoint_id, cluster, attrid, value);
CREATE TABLE groups (group_id, name);
CREATE TABLE group_members (group_id, ieee ieee, endpoint_id,
CREATE TABLE relays (ieee ieee, relays,
CREATE TABLE neighbors (device_ieee ieee NOT NULL, extended_pan_id ieee NOT NULL,ieee ieee NOT NULL, nwk INTEGER NOT NULL, struct INTEGER NOT NULL, permit_joining INTEGER NOT NULL, depth INTEGER NOT NULL, lqi INTEGER NOT NULL, FOREIGN KEY(device_ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
```

Time for more migrations?

puddly, 2021-04-08:
Second user's database: [zigbee-attrs2.db.zip](https://github.com/zigpy/zigpy/files/6281201/zigbee-attrs2.db.zip)

This one has the same problem:

```Console
$ sqlite3 ~/Downloads/zigbee-attrs2.db .dump | grep -i 'cc:cc:cc:ff:fe:29:4d:f4'
INSERT INTO attributes VALUES('cc:cc:cc:ff:fe:29:4d:f4',1,6,0,1);
INSERT INTO attributes VALUES('cc:cc:cc:ff:fe:29:4d:f4',1,8,0,254);
```

But the schema is more recent?

```Console
$ sqlite3 ~/Downloads/zigbee-attrs2.db .dump | grep -i 'CREATE TABLE'
CREATE TABLE devices (ieee ieee, nwk, status);
CREATE TABLE endpoints (ieee ieee, endpoint_id, profile_id, device_type device_type, status, FOREIGN KEY(ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
CREATE TABLE clusters (ieee ieee, endpoint_id, cluster, FOREIGN KEY(ieee, endpoint_id) REFERENCES endpoints(ieee, endpoint_id) ON DELETE CASCADE);
CREATE TABLE neighbors (device_ieee ieee NOT NULL, extended_pan_id ieee NOT NULL,ieee ieee NOT NULL, nwk INTEGER NOT NULL, struct INTEGER NOT NULL, permit_joining INTEGER NOT NULL, depth INTEGER NOT NULL, lqi INTEGER NOT NULL, FOREIGN KEY(device_ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
CREATE TABLE node_descriptors (ieee ieee, value, FOREIGN KEY(ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
CREATE TABLE output_clusters (ieee ieee, endpoint_id, cluster, FOREIGN KEY(ieee, endpoint_id) REFERENCES endpoints(ieee, endpoint_id) ON DELETE CASCADE);
CREATE TABLE attributes (ieee ieee, endpoint_id, cluster, attrid, value, FOREIGN KEY(ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
CREATE TABLE groups (group_id, name);
CREATE TABLE group_members (group_id, ieee ieee, endpoint_id,
CREATE TABLE relays (ieee ieee, relays,
```

Adminiuga, 2021-04-11:
It is strange it happened to the last db. Could have happened on the 1st database in certain circumstances. :shrug:

puddly, 2021-04-11:
It looks like that second SQLite database is itself corrupt, which is a little surprising:

```console
$ sqlite3 ~/Downloads/zigbee-attrs2.db
SQLite version 3.32.3 2020-06-18 14:16:19
Enter ".help" for usage hints.
sqlite> pragma integrity_check;
*** in database main ***
On tree page 32 cell 18: Rowid 977696 out of order
Page 45 is never used
row 141 missing from index attribute_idx
row 142 missing from index attribute_idx
row 143 missing from index attribute_idx
row 144 missing from index attribute_idx
row 145 missing from index attribute_idx
wrong # of entries in index attribute_idx
sqlite> pragma foreign_key_check;
attributes|977653|devices|0
attributes|977654|devices|0
attributes|977656|devices|0
attributes|977657|devices|0
attributes|977668|devices|0
attributes|977669|devices|0
attributes|977670|devices|0
```

closed 2021-05-26 by Adminiuga