zigpy/zigpy #712: Conbee 2 serial device path missing

eye-7, 2021-04-15:
Hi all,

I am trying to install ZHA at home assistant core (comunity package) on a synology ds216j. I have plugged a conbee 2 usb which is recongnized, then I am asked the radio, and then the serial path. At this point I am lost. By default it shows /dev/ttyACM0 which is correct but not accepted.
After google it quite long it seems that a path based on serial-ID should be used instead but I don't how to get that value. All the webs I have checked send to the assistant tab-> hardware of the home assistant to get the value, but unfortunally, as I am running core I don't have that tab. How should I config ZHA?

Thanks in advance

Adminiuga, 2021-04-16:
This is not a zigpy issue, check zigbee channel on ha discord for help. 

Docker doesn't handle device hot plugging. After you plug in the dongle and get the serial device in the host, take a note of what new serial device appeared. 
Make sure that device is passed through to container and then restart the container. 
Enable debug log and configure the integration and check the logs for errors, it would give you more insight, is it a missing device path or dongle just not responding.

eye-7, 2021-04-16:
Thank you, but note that there is not a problem of passing throught the container as there is not container. I am not using docker as it is not supported by my NAS.
Log shows the following:
2021-04-16 15:20:16 WARNING (MainThread) [zigpy_znp.zigbee.application] Failed to probe ZNP radio with config {'path': '/dev/ttyACM0', 'flow_control': None, 'baudrate': 115200}
Traceback (most recent call last):
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/serial/serialposix.py", line 322, in open
self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyACM0'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 167, in probe
await znp.connect()
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/zigpy_znp/api.py", line 199, in connect
self._uart = await uart.connect(self._config[conf.CONF_DEVICE], self)
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/zigpy_znp/uart.py", line 153, in connect
transport, protocol = await serial_asyncio.create_serial_connection(
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 445, in create_serial_connection
serial_instance = serial.serial_for_url(*args, **kwargs)
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/serial/__init__.py", line 90, in serial_for_url
instance.open()
File "/volume1/@appstore/homeassistant/env/lib/python3.8/site-packages/serial/serialposix.py", line 325, in open
raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

I know that the path is not correct, the only problem is that I don't know where to find the correct one.
In any case, as suggested, I will request support on HA.

Adminiuga, 2021-04-16:
/dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0

eye-7, 2021-04-17:
What do you mean with that post? Maybe am I using the correct path but there is a problem with permisions?
If it is the case, which permissions should be given and to whom? And more important, how? Although I have some very basic knowledges I am not a linux user.

Adminiuga, 2021-04-17:
Ha does not have permission to open the serial device. Whether that device is the correct one or not, is unknown. 
"How" is out of the scope. Probably is done the same way as on any other linux distro, but perhaps specific to your nas.
This place is not a support forum, try asking other owners of the nas.

closed 2021-04-17 by Adminiuga