zigpy/zigpy #715: Documentation?

skorokithakis, 2021-04-21:
Is there any documentation on how to use this library to communicate with ZigBee devices anywhere? I haven't managed to find anything.

Adminiuga, 2021-04-22:
There's none. But check through the issues and  discussion, there were some pointers left. 
Other than that, the only options is to dig through home assistant zha integration sources and/or zigpy/bellows cli portion of the bellows python package

skorokithakis, 2021-04-22:
Hmm, I see, thank you.

Adminiuga, 2021-04-22:
And https://github.com/zigpy/zigpy/pull/477

Hedda, 2021-04-30:
> And #477

@skorokithakis also see https://github.com/zigpy/zigpy/issues/452 and https://github.com/zigpy/zigpy/issues/471 as well as the wiki https://github.com/zigpy/zigpy/wiki

You should of course feel free to review and update this zigpy API documentation in PR https://github.com/zigpy/zigpy/pull/477

Hedda, 2021-04-30:
> Other than that, the only options is to dig through home assistant zha integration sources and/or zigpy/bellows cli portion of the bellows python package

@zoic21 will the source code for the new plugin-zigbee "beta" for Jeedom be open source and available in public repository?

https://community.jeedom.com/t/plugin-zigbee-beta/40301

Thinking that a plugin like that might be more modular and as such have source code that could be easier to dig into?

https://github.com/jeedom

@zoic21 Btw, I don't understand French but reading google translation of that thread the beta plugin is now tagged as "stable"?