zigpy/zigpy #739: [Future request] Use ZB3 native Trust Center Link Key instead of hashed

MattWestb, 2021-05-14:
Then MG2X and some high end MG1X chips have enough flash and ram for using real Trust Center Link Keys and storing them in NVM3 and the community have access to the SDK for cooking firmware so its possible putting the max 127 TCL keys in the key table that is doing all the real ZB3 device more secure but if you like paring legacy device you is getting one week link but its up to the user.

The max device is for real ZB3 127 but with old (no ZB3 Xiaomi devices) you can still going over that and other devices that is accepting not getting on new TCLK on joining (i can being wrong but it no big problem for the moment then i still testing the NCP-GP).

One parameter in ZHA config for changing the security policy to / from Hashed to stored link keys in the NCP shall fixing that in one nice way.

Very sad that Silabs not have putting up the TCLK higher then its now possible using  NVM3 instead of SimEE1/2 that can using more flash pages then before for storing them.

Shall we writing tickets to Silabs for putting it up to 256 or 512 ? 
I think its around 10 devs and advanced user that can doing that (have reged WSKT) and perhaps they is doing it.

I think @tube0013 also can being interesting implanting it in his EZSP PRO+ firmware (extended  with GP and real ZB3 security).

Thanks in advance

Mattias

MattWestb, 2021-05-18:
EZSP 6.9.2.0 GP running on IKEA Billy EZSP with max link key table:
```
CONFIG_KEY_TABLE_SIZE=127
```

MattWestb, 2021-11-14:
Still very interesting if like have good security and not  having one not Zigbee 3 devices in one production network.

Adminiuga, 2021-11-14:
Hashed key still give you the zb3 security without limiting devices to 127 zb3 devices.

Adminiuga, 2021-11-18:
This is not related to zigpy.

closed 2021-11-18 by Adminiuga