zigpy/zigpy #741: Error Trust remote zyct-202

amrij, 2021-05-17:
When I try to connect a Trust remote I get the following error:
Logger: zigpy.zcl
Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:363
First occurred: 12:20:45 (5 occurrences)
Last logged: 12:49:52

    [0x0f5d:1:0x0006] invalid literal for int() with base 10: ''
    [0xb12b:1:0x0006] invalid literal for int() with base 10: 'None'

puddly, 2021-05-17:
Can you enable [ZHA debug logging](https://www.home-assistant.io/integrations/zha/#debug-logging) and get a proper debug log?

amrij, 2021-05-17:
This is the log:

2021-05-17 16:41:56 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=set_zigbee_cluster_attribute, service_data=ieee=00:15:8d:00:04:f8:39:06, endpoint_id=1, cluster_id=6, cluster_type=in, attribute=0, value=>
2021-05-17 16:41:56 ERROR (MainThread) [zigpy.zcl] [0xb12b:1:0x0006] invalid literal for int() with base 10: ''
2021-05-17 16:41:56 DEBUG (MainThread) [zigpy.device] [0xb12b] Extending timeout for 0x8d request
2021-05-17 16:41:56 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 141 under 142 request id, data: b'008d02'
2021-05-17 16:41:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 142, 0, <DeconzAddressEndpoint address_mode=2 address=0xB12B endpoint=1>, 260, 6, 1, b'\x00\x8d\x02', 2, 0)
2021-05-17 16:41:56 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 142]

[Error.txt](https://github.com/zigpy/zigpy/files/6494844/Error.txt)

Adminiuga, 2021-05-17:
Do you call set_zigbee_cluster_attribute zha service?

amrij, 2021-05-17:
I am using the Cluster command to link the switch to the led.
[image]

Adminiuga, 2021-05-17:
I couldn't reproduce this problem.
What Version HA? What Environment? What device? Device Signature?

Remotes shouldn't have on_off server (IN) cluster

Adminiuga, 2021-05-17:
It's a remote, but reports an On_Off Server cluster. This is not Zigbee compliant.

amrij, 2021-05-18:
The version is: 2021.5.1 and the device Signature is:
"node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4653, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0, *allocate_address=True, *complex_descriptor_available=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False, *is_valid=True, *logical_type=<LogicalType.EndDevice: 2>, *user_descriptor_available=False)",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": 49246,
      "device_type": "0x03f2",
      "in_clusters": [
        "0x1000"
      ],
      "out_clusters": [
        "0x1000"
      ]
    }
  },
  "manufacturer": "Trust International B.V.",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}

Adminiuga, 2021-05-18:
you can't send on/Off commands to this remote. This would not work. 
Go to developers tab -> Event and start listening to `zha_event`, then press buttons on the remote

closed 2021-11-22 by github-actions[bot]