zigpy/zigpy #758: Tuya CO2 sensor initialization fails

Adminiuga, 2021-06-20:
There's a bug in zigpy new initialization, preventing successful completion of the device initialization.

`zigpy==0.34.0`, EZSP radio, latest HA Core dev tree. Seems like right after `Device._initialize()` gets the model and manufacturer response, it throws this error:
```
2021-06-19 23:41:21 WARNING (MainThread) [zigpy.device] Device failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/device.py", line 153, in initialize
    await self._initialize()
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/device.py", line 220, in _initialize
    self.info(
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/util.py", line 93, in info
    return self._log(logging.INFO, msg, *args, **kwargs)
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/util.py", line 82, in _log
    return self.log(lvl, msg, *args, stacklevel=4, **kwargs)
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/device.py", line 381, in log
    return LOGGER.log(lvl, msg, *args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1508, in log
    self._log(level, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1585, in _log
    self.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
    self.emit(record)
  File "/home/lex/lex/src/ac-hass/homeassistant/components/zha/core/gateway.py", line 730, in emit
    entry = LogEntry(record, stack, _figure_out_source(record, stack, self.hass))
  File "/home/lex/lex/src/ac-hass/homeassistant/components/system_log/__init__.py", line 99, in __init__
    self.message = deque([record.getMessage()], maxlen=5)
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/endpoint.py", line 317, in __repr__
    f" in=[{cluster_repr(self.in_clusters.values())}]"
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/endpoint.py", line 311, in cluster_repr
    [f"{c.ep_attribute}:0x{c.cluster_id:04X}" for c in clusters]
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/endpoint.py", line 311, in <listcomp>
    [f"{c.ep_attribute}:0x{c.cluster_id:04X}" for c in clusters]
  File "/home/lex/.venv/hass/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 536, in __getattr__
    raise AttributeError("No such command name: %s" % (name,))
AttributeError: No such command name: ep_attribute
```

Full log:
```
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command permitJoining: (60,)
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'237321a9762a2941ec7e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3373a5a9762a152a9e7e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4373b1a94d2a8954c27e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 34 (permitJoining) received: b'00'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 25 (stackStatusHandler) received: b'9c'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.zigbee.application] Received stackStatusHandler frame with [<EmberStatus.undefined_0x9c: 156>]
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'357021a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc47a0f17e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5470a1a9fb2a15be2d7e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TRUST_CENTER_POLICY: 0>, <EzspDecisionBitmask.ALLOW_UNSECURED_REJOINS|ALLOW_JOINS: 3>)
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'467121a9012a15b1592e667e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6571a1a9012a153ef27e'
2021-06-19 23:41:19 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-06-19 23:41:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 85 (setPolicy) received: b'00'
2021-06-19 23:41:20 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7571b1a9702af32d88039cdb55726fcd9d4e27aba7f87e'
2021-06-19 23:41:20 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-06-19 23:41:20 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 36 (trustCenterJoinHandler) received: b'e69fd197d6feff27fd8401000000'
2021-06-19 23:41:20 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0x9fe6, 84:fd:27:ff:fe:d6:97:d1, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
[... 222 lines cut ...]
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'737bb1a90d2af32d88039cdb55726fcda8e727db3b7e'
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'e69fd197d6feff27fd8434a900'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x9fe6, 84:fd:27:ff:fe:d6:97:d1, 52, -87, []]
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x9fe6, 84:fd:27:ff:fe:d6:97:d1, 52, -87, [])
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'037bb1a9112a15b65894a524ab5593499cc21f010b519874f0cf5a88fc7e2da5ebc9de6f8edac5626e7e'
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef0101000100008c38aae69fffff0d0939010000120200040000012502'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=140), 56, -86, 0x9fe6, 255, 255, b'\t9\x01\x00\x00\x12\x02\x00\x04\x00\x00\x01%']
2021-06-19 23:41:21 DEBUG (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=57 command_id=1>
2021-06-19 23:41:21 WARNING (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] Unknown cluster-specific command 1
2021-06-19 23:41:21 DEBUG (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] ZCL request 0x0001: b'\x00\x00\x12\x02\x00\x04\x00\x00\x01%'
2021-06-19 23:41:21 DEBUG (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] No handler for cluster command 1
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'137bb1a90d2af32d88039cdb55726fcda4e4275ac77e'
2021-06-19 23:41:21 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'e69fd197d6feff27fd8438aa00'
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x9fe6, 84:fd:27:ff:fe:d6:97:d1, 56, -86, []]
2021-06-19 23:41:21 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x9fe6, 84:fd:27:ff:fe:d6:97:d1, 56, -86, [])
[... cut ...] _ROUTE_DISCOVERY: 256>, groupId=0, sequence=141), 52, -87, 0x9fe6, 255, 255, b'\t9\x01\x00\x00\x12\x02\x00\x04\x00\x00\x01%']
2021-06-19 23:41:21 DEBUG (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=57 command_id=1>
2021-06-19 23:41:21 WARNING (MainThread) [zigpy.zcl] [0x9fe6:1:0xef00] Unknown cluster-specific command 1
```

puddly, 2021-06-20:
It appears that `ep_attribute` isn't actually a class attribute of the base `Cluster` class when it's initialized directly (for non-standard cluster `0xEF00`), causing this attribute error when the endpoint's `__repr__` method tries to format all of its clusters.

Simple fix would be to add it the attribute to the class itself (also in #716):

```Python
class Cluster(util.ListenableMixin, util.CatchingTaskMixin, metaclass=Registry):
    ...
    ep_attribute = None
```

Adminiuga, 2021-06-20:
Doh, cause 0xEF00 is not even in the manufacturer specific range.

puddly, 2021-06-20:
Good job Tuya 😆. It seems like this error only really occurs when debug logging is enabled (otherwise `Endpoint.__repr__` won't be called), so the actual impact of this bug should be quite low, even in `dev`.

Adminiuga, 2021-06-20:
Fixed in #759

closed 2021-06-20 by Adminiuga