zigpy/zigpy #766: Request device Lidl HG06323 (now only listed with battery as

mvroosmalen1970, 2021-07-01:
The remote from Lidl (https://zigbee.blakadder.com/Lidl_HG06323.html) is recognized by ZHA (using zigpy) but only shows up as a device with battery status. No events are logged/connected. Can this be added? Its support by TASMOTA ZIGBEE2MQTT DECONZ, but would be nice if it could be added  in zigpy as well :-)

Adminiuga, 2021-07-01:
Request support in zha-device-handlers repository and provide all requested information by the template.

I'm closing and not moving this issue, as it doesn't have the required information, like device signature, debug log for button presses etc

closed 2021-07-01 by Adminiuga

mvroosmalen1970, 2021-07-02:
Captured this in HA log in ZHA debug modus: 
This sequence was pressed with HG06323:
Long press O, short press O, long press I, short press I, long press *, short press *, long press x, short press x

b'a2020301da023f022d000600120000008e001500000000002b0003000000000000000000000000000000000000000000000000000000000000000000000000000000000000004e0e0000000000000000'
2021-07-02 21:05:28 DEBUG (MainThread) [bellows.zigbee.application] EZSP Counters: COUNTER_MAC_RX_BROADCAST: 674,COUNTER_MAC_TX_BROADCAST: 259,COUNTER_MAC_RX_UNICAST: 730,COUNTER_MAC_TX_UNICAST_SUCCESS: 575,COUNTER_MAC_TX_UNICAST_RETRY: 45,COUNTER_MAC_TX_UNICAST_FAILED: 6,COUNTER_APS_DATA_RX_BROADCAST: 18,COUNTER_APS_DATA_TX_BROADCAST: 0,COUNTER_APS_DATA_RX_UNICAST: 142,COUNTER_APS_DATA_TX_UNICAST_SUCCESS: 21,COUNTER_APS_DATA_TX_UNICAST_RETRY: 0,COUNTER_APS_DATA_TX_UNICAST_FAILED: 0,COUNTER_ROUTE_DISCOVERY_INITIATED: 43,COUNTER_NEIGHBOR_ADDED: 3,COUNTER_NEIGHBOR_REMOVED: 0,COUNTER_NEIGHBOR_STALE: 0,COUNTER_JOIN_INDICATION: 0,COUNTER_CHILD_REMOVED: 0,COUNTER_ASH_OVERFLOW_ERROR: 0,COUNTER_ASH_FRAMING_ERROR: 0,COUNTER_ASH_OVERRUN_ERROR: 0,COUNTER_NWK_FRAME_COUNTER_FAILURE: 0,COUNTER_APS_FRAME_COUNTER_FAILURE: 0,COUNTER_UTILITY: 0,COUNTER_APS_LINK_KEY_NOT_AUTHORIZED: 0,COUNTER_NWK_DECRYPTION_FAILURE: 0,COUNTER_APS_DECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COUNTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 3662,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI_PRI_TX_ABORTED: 0,COUNTER_ADDRESS_CONFLICT_SENT: 0
2021-07-02 21:05:30 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 128 (incomingRouteErrorHandler) received: b'420020'
2021-07-02 21:05:30 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteErrorHandler frame with [<EmberStatus.MAC_INDIRECT_TIMEOUT: 66>, 0x2000]
2021-07-02 21:05:30 DEBUG (MainThread) [bellows.zigbee.application] Processing route error: status=EmberStatus.MAC_INDIRECT_TIMEOUT, nwk=0x2000
2021-07-02 21:05:34 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010100010000a65cb30020ffff040158fd00'
2021-07-02 21:05:34 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=166), 92, -77, 0x2000, 255, 255, b'\x01X\xfd\x00']
2021-07-02 21:05:34 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=88 command_id=253>
2021-07-02 21:05:34 WARNING (MainThread) [zigpy.zcl] [0x2000:1:0x0006] Unknown cluster-specific command 253
2021-07-02 21:05:34 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL request 0x00fd: b'\x00'
2021-07-02 21:05:34 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] No handler for cluster command 253
2021-07-02 21:05:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010100010000a760b40020ffff040158fd00'
2021-07-02 21:05:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=167), 96, -76, 0x2000, 255, 255, b'\x01X\xfd\x00']
2021-07-02 21:05:35 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=88 command_id=253>
2021-07-02 21:05:35 WARNING (MainThread) [zigpy.zcl] [0x2000:1:0x0006] Unknown cluster-specific command 253
2021-07-02 21:05:35 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL request 0x00fd: b'\x00'
2021-07-02 21:05:35 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] No handler for cluster command 253
2021-07-02 21:05:36 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010100010000a85cb30020ffff040158fd00'
2021-07-02 21:05:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=168), 92, -77, 0x2000, 255, 255, b'\x01X\xfd\x00']
2021-07-02 21:05:36 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=88 command_id=253>
2021-07-02 21:05:36 WARNING (MainThread) [zigpy.zcl] [0x2000:1:0x0006] Unknown cluster-specific command 253
[... cut ...]
2021-07-02 21:05:42 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=93 command_id=253>
2021-07-02 21:05:42 WARNING (MainThread) [zigpy.zcl] [0x2000:1:0x0006] Unknown cluster-specific command 253
2021-07-02 21:05:42 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] ZCL request 0x00fd: b'\x01'
2021-07-02 21:05:42 DEBUG (MainThread) [zigpy.zcl] [0x2000:1:0x0006] No handler for cluster command 253
not sure what additional info I could provide, please help me providing it to you

mvroosmalen1970, 2021-07-02:
maybe this is helpful too:
"modelId":"TS1001","manufacturerName":"_TYZB01_bngwdjsr"