zigpy/zigpy #781: Failed to migrate row attributes... NOT NULL constraint failed: attributes_cache_v5.value

deviantintegral, 2021-08-08:
After upgrading Home Assistant from 2021.7.4 to 2021.8.3:

```
Logger: zigpy.appdb
Source: /usr/local/lib/python3.9/site-packages/zigpy/appdb.py:549 
First occurred: 8:07:45 PM (2 occurrences) 
Last logged: 8:07:45 PM

Failed to migrate row attributes(00:15:8d:00:05:d4:15:e6, 2, 18, 0, None): NOT NULL constraint failed: attributes_cache_v5.value
Failed to migrate row attributes(00:15:8d:00:05:cf:32:25, 2, 18, 0, None): NOT NULL constraint failed: attributes_cache_v5.value
```

These are both Aqara Cubes, with quirk `zhaquirks.xiaomi.aqara.cube_aqgl01.CubeAQGL01`. Is there anything to do here? Glad to provide DB excepts if someone can point to what rows / tables would be useful.

puddly, 2021-08-08:
Has device functionality changed at all? This table is just a cache so if everything still works, I think you're fine.

The quirk is using attribute `0x0000` to hold the device state as a string, but also writes `None` as well: https://github.com/zigpy/zha-device-handlers/blob/2f9b0c33e08dfbddb703e44ce017e0a0a9315562/zhaquirks/xiaomi/aqara/cube_aqgl01.py#L195. That is a little odd.

deviantintegral, 2021-08-09:
They look to be working fine. Admittedly, I've yet to find an actual use for them, but I can see events firing in Home Assistant.

closed 2021-09-02 by puddly