zigpy/zigpy #791: Data remains after deserializing ZCL frame (with trådfri symfonisk spin remote)

systemcrash, 2021-08-17:
I spin the remote, and it sends one command at spin start, and one at spin stop like so:

```
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=74 command_id=1>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0001: [0, 195]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 1
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=74 command_id=1>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0001: [0, 195]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 1
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=75 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=75 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
```

I'm using an automation blueprint to marshal these inputs to control stuff. I *think* we can safely ignore these, maybe we can silence them, in such a case. 

HASS core-2021.8.7

Need more info?

device sig:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "SYMFONISK Sound Controller",
  "class": "zhaquirks.ikea.symfonisk.IkeaSYMFONISK"
}
```

```
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=47 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=47 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
```

Adminiuga, 2021-08-17:
Enable debug for your radio and post the payload

systemcrash, 2021-08-18:
Ah, sorry, I was trying to figure out which module needs to have debug to see that. 

Did I get the right one? This is what flies past.

startup

```
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, 1, 260, 8, b'\x11#\x01\x01\xc3\x00\x00', 0, 175, 255, 123, 126, 36, 0, -58]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=35 command_id=1>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0001: [1, 195]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 1
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'11230101c30000'
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, 1, 260, 8, b'\x11#\x01\x01\xc3\x00\x00', 0, 175, 255, 123, 126, 36, 0, -58]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=35 command_id=1>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0001: [1, 195]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 1
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'11230101c30000'
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, 1, 260, 8, b'\x11$\x03\x00\x00', 0, 175, 255, 123, 126, 36, 0, -58]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=36 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'1124030000'
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, 1, 260, 8, b'\x11$\x03\x00\x00', 0, 175, 255, 123, 126, 36, 0, -58]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=36 command_id=3>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0003: []
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 3
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'1124030000'
```

loaded
```
DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...] manufacturer=None tsn=29 command_id=1>
WARNING (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] Data remains after deserializing ZCL frame
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] ZCL request 0x0001: [1, 195]
DEBUG (MainThread) [zigpy.zcl] [0x0f84:1:0x0008] No handler for cluster command 1
DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f84>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'111d0101c30000'
```

elektrinis, 2021-12-06:
I am getting the same issue with this device.
I am also getting multiple events for a single action, 4 events per action every time: start of spin, stop of spin, toggle, double tap, triple tap.

puddly, 2021-12-06:
This will be fixed once https://github.com/zigpy/zigpy/pull/716 is done. It's nothing to worry about.

elupus, 2022-03-30:
I still seem to be getting this warning when spinning symfonisk.

dmulcahey, 2022-03-30:
The referenced PR above still hasn’t hit HA yet.

elupus, 2022-03-30:
Ah, since issue comment was so old, but referenced request was merged, i assumed it was not addressed. Then i will recheck later.

closed 2022-04-30 by puddly