zigpy/zigpy #806: Aqara RTCGQ13LM occupancy sensor (high precision motion sensor)

alexander-vitishchenko, 2021-09-05:
Hi Zigpy Development Team,

Can you please advise if zigpy (and Home Assistant) supports RTCGQ13LM occupancy sensor - https://zigbee.blakadder.com/Aqara_RTCGQ13LM.html?

I'd assume "no" based on https://zigbee.blakadder.com/sensors.html status. Would you be so kind to advise if you have any plans to introduce support for new device type? Apologies if this request is a duplicate, and I didn't see other people asking about it.

Thank you,
Alexander

Hedda, 2022-02-24:
That blakadder Zigbee list is a wiki-style database maintained by end-users of Zigbee and device compatibility there only gets listed if a end-user both tested the device and then made the effort to these submit an edit pull request to update it, meaning that you can not rely on that it is accurate or up-to-date, and especially not because of the way that standard Zigbee interoperability works.

Read -> https://www.home-assistant.io/integrations/zha#knowing-which-devices-are-supported

### _Knowing which devices are supported_

_There is no official compatibility list of supported devices for the simple reason that practically all devices Zigbee Home Automation that are fully compliant with the standards and specifications as set by the [Zigbee Alliance](https://zigbeealliance.org) should technically be compatible with this ZHA integration. The fact remains, however, that some hardware manufacturers do not always fully comply with each set specification, which can cause a few devices to only partially work or not work at all with ZHA, but developers can create workarounds for such issues via a solution for 'ZHA exception and deviation handling' that this implementation features. See that section for more information._

_Tip to new users is that, while there is no official list of supported devices, some ZHA users take comfort that blakadder maintains an unofficial Zigbee Device Compatibility Repository which anyone can submit compatibility reports to, it can be found at [zigbee.blakadder.com](https://zigbee.blakadder.com) and currently contains independent compatibility lists and device pairing tips for several home automation gateway/bridge/hub software, including but not limited to open source Zigbee implementations such as; ZHA, Tasmota, Zigbee2MQTT, and ZiGate._

### _ZHA exception and deviation handling_

_The ZHA implementation in Home Assistant relies on a library called "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)" to resolve issues with Zigbee devices that do not fully conform with the Zigbee standards. The few devices that deviate from the Zigbee specifications set by the [Zigbee Alliance](https://zigbeealliance.org) may therefore require proper bug reports with debug logs from users to assistant the developers in writing custom [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/blob/dev/README.md) for all of a device functions to work properly with the ZHA integration._

_Such a custom "ZHA Device Handler" are Python scripts that internally are also referred to as a "quirk" because they fix "quirks", like deviations from the standard specifications. ZHA Device Handles do this by transparently, acting as a translator, translating and converting non-compliant device messages and instead present them to the application as coming from a virtual compliant device. These ZHA Device Handlers for Home Assistant can thus be used to parse custom messages to and from Zigbee devices. The ZHA Device Handlers that are made can then be reused by all users in future versions of Home Assistant._

_The custom quirks implementations for zigpy implemented as ZHA Device Handlers for Home Assistant are a similar concept to that of [Hub-connected Device Handlers for the SmartThings Classics platform](https://docs.smartthings.com/en/latest/device-type-developers-guide/) as well as that of [Zigbee-Herdsman Converters (formerly Zigbee-Shepherd Converters) as used by Zigbee2mqtt](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html), meaning they are each virtual representations of a physical device that expose additional functionality that is not provided out-of-the-box by the existing integration between these platforms._

Any issues with devices really belong in repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)”  instead of here in zigpy which is for lower-level issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) provides some information about how/why quirks are used and points people to that project:

https://www.home-assistant.io/integrations/zha/#zha-exception-and-deviation-handling

https://www.home-assistant.io/integrations/zha/#reporting-issues

https://www.home-assistant.io/integrations/zha/#debug-logging

###  _ZHA EXCEPTION AND DEVIATION HANDLING_

_The ZHA implementation in Home Assistant relies on a library called “[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” to resolve issues with Zigbee devices that do not fully conform with the Zigbee standards. The few devices that deviate from the Zigbee specifications set by the [Zigbee Alliance](https://zigbeealliance.org/) may therefore require proper bug reports with debug logs from users to assistant the developers in writing custom [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/blob/dev/README.md) for all of a device functions to work properly with the ZHA integration._

_Such a custom “ZHA Device Handler” are Python scripts that internally are also referred to as a “quirk” because they fix “quirks”, like deviations from the standard specifications. ZHA Device Handles do this by transparently, acting as a translator, translating and converting non-compliant device messages and instead present them to the application as coming from a virtual compliant device. These ZHA De [... cut ...]
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

PS: Also, for ZHA general end-user community support, see here -> https://community.home-assistant.io/c/configuration/zha/48

Hedda, 2022-06-17:
@alexander-vitishchenko Will you please consider closing this issue under the zigpy repository yourself?

Again high-level issues with device compatibility problems such as those you mention should be reported as issue to repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” (a.k.a. "[zha-quirks](https://pypi.org/project/zha-quirks)") instead of here in zigpy library repository which is only for low-level Zigbee issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

If test now and still have problem then open an issue there -> https://github.com/zigpy/zha-device-handlers/issues/new/choose

closed 2022-06-17 by alexander-vitishchenko