zigpy/zigpy #809: Device initialization task can be spawned multiple times

puddly, 2021-09-11:
@TheJulianJES reported strange behavior with a device re-joining during initialization and the initialization task somehow being launched *twice*, causing the device to complete zigpy initialization without being initialized at all:

```Python
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [0, <Bool.true: 1>, 0x45ca, 14:b4:57:ff:fe:91:b3:6b, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'22e2b1a9702adff73227dbdb5502265d9d4e27ab21c57e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 36 (trustCenterJoinHandler) received: b'ca456bb391feff57b41401000000'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0x45ca, 14:b4:57:ff:fe:91:b3:6b, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]

# Device joins
2021-08-14 01:34:31 INFO (MainThread) [zigpy.application] New device 0x45ca (14:b4:57:ff:fe:91:b3:6b) joined the network

# First initialization is scheduled (3 attempts)
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.device] [0x45ca] Scheduling initialization
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Send command findKeyTableEntry: (14:b4:57:ff:fe:91:b3:6b, True)
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.util] Tries remaining: 3

# Node descriptor is requested (2 attempts)
2021-08-14 01:34:31 INFO (MainThread) [zigpy.device] [0x45ca] Requesting 'Node Descriptor'
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.device] [0x45ca] Extending timeout for 0x61 request

2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 14:b4:57:ff:fe:91:b3:6b/0x45ca
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (14:b4:57:ff:fe:91:b3:6b, True)
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'23e321a9212a7d5e01c86ab5721e4193e39b7e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'33e3a5a9212aea21947e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 117 (findKeyTableEntry) received: b'ff'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'34e021a92a2a7d5e01c86ab5721e419382517e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'43e2b1a9112a15b259954a25aa1592499c5dc77f278b9874f8476389fc7e3d83937e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000000010000004000000013e0d4ca45ffff05810000000002'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=1, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=19), 224, -44, 0x45ca, 255, 255, b'\x81\x00\x00\x00\x00']
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x45CA ieee=14:b4:57:ff:fe:91:b3:6b is_initialized=False> from ep 0 to ep 0, cluster 1: b'\x81\x00\x00\x00\x00'
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.zdo] [0x45ca:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0x0000, 0, 0]
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'53e3b5a9362a7e01c86ab5721e410f827e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'64e0a5a92a2a759e7e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 98 (incomingSenderEui64Handler) received: b'6bb391feff57b414'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Received incomingSenderEui64Handler frame with [14:b4:57:ff:fe:91:b3:6b]
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 126 (setExtendedTimeout) received: b''
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'74e3b1a9112a11b259874a25aa5596499c5ac77f278b9874f144a9cc97cdae59149a6a7b0ffd9d357e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'040000130000000004000014e0d4ca45ffff0c82ca456bb391feff57b4148002'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=20), 224, -44, 0x45ca, 255, 255, b'\x82\xcaEk\xb3\x91\xfe\xffW\xb4\x14\x80']
2021-08-14 01:34:31 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0x45ca, 14:b4:57:ff:fe:91:b3:6b

# Device announces itself
2021-08-14 01:34:31 INFO (MainThread) [zigpy.application] Device 0x45ca (14:b4:57:ff:fe:91:b3:6b) joined the network
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.device] [0x45ca] Scheduling initialization
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.device] [0x45ca] Canceling old initialize call
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x45CA ieee=14:b4:57:ff:fe:91:b3:6b is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x82\xcaEk\xb3\x91\xfe\xffW\xb4\x14\x80'
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.zdo] [0x45ca:zdo] ZDO request ZDOCmd.Device_annce: [0x45CA, 14:b4:57:ff:fe:91:b3:6b, 128]
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x45CA, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=97), 98, b'a\xcaE')

# Initialize is re-started, 3 tries left
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.util] Tries remaining: 3

# And 2 for the node descriptor
2021-08-14 01:34:31 INFO (MainThread) [zigpy.device] [0x45ca] Requesting 'Node Descriptor'
2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.util] Tries remaining: 2

2021-08-14 01:34:31 DEBUG (MainThread) [zigpy.device] [0x45ca] Extending timeout for 0x64 request
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'40e121a9602a15781c944a27aa5592099d4e27ca8fcd0641b8bc467e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'05e1a1a9602a15364d547e'
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0084'
2021-08-14 01:34:31 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x45CA, EmberApsFrame(profileId=0, clusterId=32769, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=129), 99, b'\x81\x00d\xf3\xa5\xfe\xff\xcc\xcc\xcc\x00\x00\x00\x00')
2021-08-14 01:34:31 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'51e621a9602a15781c944a242a5592099d4e272a8ec0e68b9935c67703b2f36bebcdde6f102f7e'
[... 128 lines cut ...]
2021-08-14 01:34:33 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x45CA, EmberApsFrame(profileId=0, clusterId=4, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=108), 109, b'l\xcaE\x01')
2021-08-14 01:34:33 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'75ec21a9602a15781c944a21aa5592099d4e27c780ca0b41b8c7f4227e'
2021-08-14 01:34:33 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'50eca1a9602a1538c54a7e'
[... cut ...]
2021-08-14 01:34:34 INFO (MainThread) [zigpy.device] [0x45ca] Discovered basic device information for <Device model=None manuf=None nwk=0x45CA ieee=14:b4:57:ff:fe:91:b3:6b is_initialized=False>

# ???
2021-08-14 01:34:34 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model=None manuf=None nwk=0x45CA ieee=14:b4:57:ff:fe:91:b3:6b is_initialized=False>
```

I haven't written a unit test to reproduce the problem yet and the existing test suite never spawns two concurrent initialization tasks. Possibly related to [https://bugs.python.org/issue42130](https://bugs.python.org/issue42130)?

puddly, 2021-12-29:
Possibly unrelated but I found this new addition to the [Python `asyncio` documentation for `asyncio.create_task()`](https://docs.python.org/3/library/asyncio-task.html#asyncio.create_task) interesting (https://github.com/python/cpython/pull/29163):

> **Important**
> Save a reference to the result of this function, to avoid a task disappearing mid execution. 

We also use `asyncio.ensure_future()` instead of `asyncio.create_task()` exactly once in zigpy:

https://github.com/zigpy/zigpy/blob/fc8e923e2f2482810b9643c0b2bde149a56a1930/zigpy/util.py#L265-L274

Could this have any unintentional side effects?

Adminiuga, 2021-12-30:
Ensure_future is from earlier python compatibility. I think it safe to drop it in favor of create_task.
The bug with task dropping is interesting. This mixin is primarily used to spawn task from sync incoming command handler context to send replies, like default response, time server cluster response etc. If task is dropped, it shouldn't affect the rest of the system.

puddly, 2022-06-28:
Not stale, unfortunately.

closed 2023-01-01 by github-actions[bot]