zigpy/zigpy #840: Popp valve parameters missing

Aposei, 2021-10-31:
Hello. I'm a user of Jeedom's zigpy-based zigbee plugin.
Recently the Jeedom team have integrated the Popp radiator valve into the plugin. The parameters obtained from the valve are the local temperature, the setpoint, the head opening and the mode. I have tried to integrate the valve in zigbee2mqtt and I get many more parameters, all listed here:
https://www.zigbee2mqtt.io/devices/701721.html
How to solve this to have all the parameters in zigpy (plugin zigbee / jeedom)?
My apologies if I do not explain well, I do not speak English or I am a computer scientist.
Thanks.

MattWestb, 2021-10-31:
Then looking in the [danfoss quirk](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/danfoss/thermostat.py) its looks having more or less all attributes and commands implanted.
The question is if Jeedom have patched the quirk for loading the Popp TRV.
If not is you only getting the "standard" functions and not  the extra danfoss is using.

@zoic21 shall we adding the Popp `eT093WRO` and Hive `UK7004240` to the Danfoss Ally quirk so they is getting all the functions that the Ally is having ?

I have asking before in the Danfoss issue but it was no interest doing that but is easy done if user need it.

MattWestb, 2021-10-31:
@Aposei Can you posting the manufacturer and model reported from the device in the zigbee network ?

Aposei, 2021-10-31:
> @Aposei Can you posting the manufacturer and model reported from the device in the zigbee network ?

Is that what you ask of me? I still have it on z2m to test.
[image]

Aposei, 2021-11-02:
@MattWestb, this is what I get in the Jeedom plugin

Manufacturer: eT093WRO
Model: eT093WRO

zoic21, 2021-11-02:
Hello,
I'am not sur that Popp had same paramaters as Danfoss it's not the same manufacturer.

MattWestb, 2021-11-02:
Thanks for replay @zoic21  !!

One user have testing with ZHA and lit looks being the same and software but can being running on different releases but the function shall working OK and one Danfoss with the same software version. https://github.com/zigpy/zha-device-handlers/issues/1005#issuecomment-940805715

I giving it one try but having problem finding the right `MODELS_INFO` but we can testing is it loading.

Is it possible to see if the device have loading the right quirk in Jeedom and is the local quirk also working or must patching the docker container or local code ?

MattWestb, 2021-11-02:
@Aposei I have adding the Popp and Hive TRVs to the quirk in my test branch https://github.com/MattWestb/zha-device-handlers/blob/Popp-Hive-TRV/zhaquirks/danfoss/thermostat.py so you can downloading it replacing the original file in you quirk folder or editing the original with the changes.

I dont knowing how to do it best in Jeedom but there devs is knowing that.

Pleas reporting back if its working OK or not !!

zoic21, 2021-11-02:
Very weird, Popp and Danfoss are competitor... 

Yes you can see in Jeedom, in "node configuration" class used by device.

User can add file in quirks folder but it's not easy...

MattWestb, 2021-11-02:
And Hive looks using all Danfoss commands 2:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/af1b1bf7fc10cb9e239c3b4a0c0077bf4ff399d2/devices/hive.js#L153-L155

Can you helping the user testing the updated quirk ?

 If its working OK i doing one pull request for it.

Hedda, 2021-11-02:
Does the official Jeedom Zigbee Plugin depend on zha-device-handlers / zha-quirks as an upstream dependency for quirk(s)?

https://github.com/zigpy/zha-device-handlers

https://pypi.org/project/zha-quirks/

If so then this issue perhaps being in https://github.com/zigpy/zha-device-handlers/issues instead of here in zigpy issues?

MattWestb, 2021-11-02:
Moving or closing issues is one thing for the maintainers of the Zigpy organization and as long i can helping the user is not problem is is not 110% right posted if working well fixing the problem.

Its the Jeedom's devs that have implanting the Danfoss Ally and many more devices in our quirks so i cant complaining and its good four all users of Zigpy system like ZHA and Jeedom is getting better device support so is one win win situation for all.

zoic21, 2021-11-02:
Yes Jeedom use zha-device-handlers / zha-quirks (same as HA). When we do quirks if quirks is fully working we make a PR to zha-device-handlers. I have 4 quirks in testing mode but they don't works at 100% so for now I don't make a PR.

Hedda, 2021-11-02:
> Moving or closing issues is one thing for the maintainers of the Zigpy organization and as long i can helping the user is not problem is is not 110% right posted if working well fixing the problem. [...]

I fully understand that. I'm only suggesting that it would just be good if Jeedom SAS could get the feedback from their users that they should really be referring to [zha-device-handlers / zha-quirks](https://github.com/zigpy/zha-device-handlers) for device compatibility issues/requests instead of zigpy.

MattWestb, 2021-11-02:
@Hedda If not having one instruction how to testing new / updated quirks in Jeedom then is not very likely the Zigpy organisation is merger them because they is completely untested = dont belonging in one production system.

The best is if Jeedom can having one guide for testing quirks in there system then its being very easy or all sides and also the user is knowing to do all things right from the beginning.

MattWestb, 2021-11-02:
@zoic21  I have made one updated guide for editing quirks in HAs docker container that is working but is not for very low skill users https://github.com/zigpy/zha-device-handlers/discussions/693.

If you like you can doing one post in the same discussion or one new one so we can helping the user testing there new device is there systems.

Aposei, 2021-11-02:
@zoic21 I have downloaded the thermostat.py file, do I have to paste it as is in a directory? or is it more complicated than that?

zoic21, 2021-11-02:
You juste have to past it in this folder : /var/www/html/plugins/zigbee/resources/zigbeed/quirks than restart deamon.

I note to add this in documentation.

MattWestb, 2021-11-02:
Is it possible pasting nested quirks like one` __init__.py `  and one `device.py` and the `device.py` is getting data form the ` __init__.py ` or is it  like ZHA that the `device.py` is always getting its data from the installation (container) ?

Some quirks need changes in both for working and good to knowing what is possible and not in Jeedom.

I have writing one note for testing quirks in Jeedom and is updating it for future needs ;-))

zoic21, 2021-11-03:
@Aposei could you please post a snapshot of "node configuration" (first tab),it's display the quirks used by device.
For command you need to change device to danfoss valve for testing.

MattWestb, 2021-11-03:
_Hola buenos dias_ @albalaing !
You is using the Popp TRV with ZHA,

Can you looking if i have doing the device ID OK i the https://github.com/zigpy/zigpy/issues/840#issuecomment-957631371 and if ZHA is loading the quirks for the Popp TRV if installing it as local quirk (or patched in the HA container) ?

Jeedom is also using Zigpy and quirks but  is not  so easy for there user to verified that it working OK.

I like getting it pushed in ZHA so all user can using them with all function implanted in there system :-)))

Thanks in advance !!

albalaing, 2021-11-03:
It recognizes me with these IDs
``  MODELS_INFO: [(DANFOSS, "eTRV0100"),("D5X84YU", "eT093WRO")],  ``
I have copied the quirk of danfoss/thermostat.py in as a local quirk and I have changed the previous line to it. And it works for me the same as danfoss, I also have you.

Aposei, 2021-11-03:
@zoic21 That?

[image]

Switching to Danfoss, I get the same, more Fênetre

MattWestb, 2021-11-03:
Thanks for feed back !!

Then i have making the manufacture ID wrong and i shall fixing it i my git.
Then the problem is finding the right for the Hive :-((

I pinging then have finding the right parameters and have doing the update so Jeedom  can testing it 2 :-))

MattWestb, 2021-11-03:
Have updating the git with:
```
        MODELS_INFO: [
            (DANFOSS, "eTRV0100"),
            ("D5X84YU", "eT093WRO"),
            ("DANFOSS", "RV001"),
        ],
```
Then have finding Popp: 
[image]
And Hive: [image]
So very strange Hive is using Danfoss aas manufacture !!

Hope getting it confirmed of some user that is working.

Aposei, 2021-11-05:
Hi, @MattWestb @zoic21 So is there a solution?  Can I contribute something else?

MattWestb, 2021-11-05:
I like having one OK for one user that the Pop is working OK with the changes i have made (the quirk is being loaded by the system for Popp TRVs) then i posting one PR for adding it, but no one have making that, so its on hold from my side.

I think then its being added in ZHA all shall working OK in Jeedom or perhaps need little attention form its devs for getting it working OK.

zoic21, 2021-11-05:
Yes I have some work todo I will add configuration for the valve in next beta.

Aposei, 2021-11-08:
Hi @MattWestb @zoic21. I have re-integrated the Popp thermostat after the latest updates to the Zigbee plugin. I still miss it, compared to when I integrate it into z2m, mainly; battery, keypad_lockout, setpoint_change_source, window_open_external and algorithm_scale_factor.

MattWestb, 2021-11-08:
Can you see if the modified Danfoss quirk is loaded of the system and adding the extra attribute for the device ?

I not sure its working OK until someone can confirming it.

And i think **Zoic21** is planing adding more function for the device in Jeedom but i dont knowing that.

Aposei, 2021-11-08:
> Can you see if the modified Danfoss quirk is loaded of the system and adding the extra attribute for the device ?
> 
> I not sure its working OK until someone can confirming it.
> 
> And i think **Zoic21** is planing adding more function for the device in Jeedom but i dont knowing that.

Yes, I look at it, but I need @zoic21 to tell me where to look at it.

zoic21, 2021-11-08:
Hello,
You need to go "node configuration" (blue button) when you are on the device in Jeedom and you have "classe" label : 
[image]

Aposei, 2021-11-08:
Thanks @zoic21.  This I get.
[image]

MattWestb, 2021-11-08:
The device IDs looks OK but the quirk is not loaded ("D5X84YU", "eT093WRO").
Did you using the second updated version then the first  was wrong ?

Aposei, 2021-11-08:
> The device IDs looks OK but the quirk is not loaded ("D5X84YU", "eT093WRO"). Did you using the second updated version then the first was wrong ?

If you mean the plugin version, I used the latest beta version that came out today

MattWestb, 2021-11-08:
I think it have not being added then the no PR is being merged for the TRVs.
You need adding it by hand or better copy my latest danfoss quirk in your system and restarting it for taking effect.

Aposei, 2021-11-08:
I have downloaded the thermostat.py file, I have copied it to the folder: /var/www/html/plugins/zigbee/resources/zigbeed/quirks
I have integrated the thermostat again and have the same commands
[image]

zoic21, 2021-11-09:
Hello,
Can you test with danfoss valve configuration ?

Aposei, 2021-11-09:
> Hola, ¿Puedes probar con la configuración de válvulas danfoss?

Yes, I try. But do I have to delete the thermostat.py file or do I keep it?

zoic21, 2021-11-09:
No not on zigpy on jeedom only : 
[image]

Aposei, 2021-11-09:
> No, no solo en zigpy en jeedom: [image]

Doing that, I have these commands...

[image]
[image]
[image]

Aposei, 2021-11-23:
Hello @zoic21 @MattWestb. Some things missing compared to zigbee2mqtt are keypad_lockout, external_measured_room_sensor and algorithm_scale_factor which in the case of z2m it is possible to select from 1 to 10
I'm on the latest beta version of the zigbee plugin

zoic21, 2021-11-23:
Hello,
Can you please open an jeedom ticket, for that it's an issue with jeedom not zigpy.

MattWestb, 2021-11-23:
Thanks @zoic21  !!
The Popp is added in the quirk by one user so it shall working in some weeks without patching then the new device handle is released (for HA 2021.12).

Hedda, 2022-01-26:
@Aposei can this issue be closed now that https://github.com/zigpy/zha-device-handlers/pull/1148 by @hertleinj for Popp D5X84YU  (POPZ701721) has been merged into [zha-device-handlers](https://github.com/zigpy/zha-device-handlers) as part of the 0.0.64 release? Maybe continue in https://github.com/zigpy/zha-device-handlers/issues/1005 if needed?

Hedda, 2022-06-17:
@Aposei Will you please consider closing this issue under the zigpy repository yourself?

Again high-level issues with device compatibility problems such as those you mention should be reported as issue to repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” (a.k.a. "[zha-quirks](https://pypi.org/project/zha-quirks)") instead of here in zigpy library repository which is only for low-level Zigbee issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

If test now and still have problem then open an issue there -> https://github.com/zigpy/zha-device-handlers/issues/new/choose

closed 2022-06-17 by Aposei