zigpy/zigpy #854: Blinking IKEA lights

keesj, 2021-11-26:
Hi,

I have been using  HA  with the ZHA plugin and CONBEE II  and have (over the last year) been experiencing problems with my IKEA lights.

The lights quite annoyingly start blinking on and off, not like when pairing but .. really annoying 

Video on the subject (and pcapng attached )

https://youtu.be/WChPa4tbrAo

[capture_0x9edf_blinks.pcapng.zip](https://github.com/zigpy/zigpy/files/7609802/capture_0x9edf_blinks.pcapng.zip)

Adminiuga, 2021-11-27:
debug log? Capture without the network key is useless

keesj, 2021-11-27:
Hello, 
There is little visible in the debug log. perhaps that I can enable an option to increase the verbosity.
The network key is the default key as found in the python source code.

I did not know where to find the transport key and therfore followed a tutorial on the topic.
https://www.digiblur.com/2021/08/zigbee-network-key-sniffing.html
With That I was able to capture the transport key but after pairing the problems did not reapeer yet
Are there more logs I can enable to capture the next time?

MattWestb, 2021-11-27:
The default "well known key" is being used for encrypting your network key that is called transport key in wireshark.

Then you have getting the network key it shall being possible looking in your "old" sniff if you not have destroying the network and building on new one.

Only back draw is if you is posting your network key its possible doing bad things with your network if knowing where its is and being near and have the knowledge.

You can positing it if you like or must sending in some way in private to the devs.

keesj, 2021-11-27:
Hi,

Thank you for response. The key is 6f3ce99ffb36752ae7df21bcf3b82d25 this in combination with the [default network key](https://github.com/zigpy/zigpy/blob/dev/zigpy/config/defaults.py#L10) allows to decrypt messages.

keesj, 2021-11-27:
Additional information is that i have been running the system for over a year and that I am running a pip installed home assistant and recently upgraded to python 3.9 (on ubuntu 21.10 running on odroid).

MattWestb, 2021-11-27:
I have comparing with one IKEA CWS3 being repowered and its broadcasting device announcement 1 time and the coordinator is requesting groups and the light is responding with its groups after have testing the rout to the coordinator.

Your system is making the same but your light is sending one new (with one different sequence number) device announcement after some seconds and starting working for some more seconds and sending device announcement . . .

Device announcement is being sent of Zigbee 3 routers then the firmware is booting up if they is in one network = OK.

But its look like your light is having problems and is rebooting all the time (and then doing the blinking).
I cant see any wrong in the communication between the coordinator and the light.

You can trying deleting the light in ZHA and waiting one minute and then adding the device **after doing one hardware rest off it** (6 X re power the light and its start flashing) so its getting all network information new if its somthing wrong in the saved in the device.

I think you light is not OK so if its not working you shall returning it to IKEA and getting on new one.

PS: if you like you can deleting the post with the network key so not all other user can using it.

Adminiuga, 2021-11-27:
> There is little visible in the debug log. perhaps that I can enable an option to increase the verbosity.
The network key is the default key as found in the python source code.

If HA/zigpy_deconz does not command the light blinking, then very little you could do then.

keesj, 2021-11-27:
I think that they key is understanding the last we messages sent (from the bulb to 0x0000)

Adminiuga, 2021-11-27:
you get a bunch of "Device Announcement" from the bulb, every 2s and those only happen on physical power on or internal resets. Check your wiring or maybe the bulb is bad, cause there's no need under normal circumstances for the bulb to send the device announcement broadcasts. 
[image]

keesj, 2021-11-27:
I have seen this issues with several bulbs over the last year. I added more logging now.

The bulb was initially somewhere else when it started showing this behavior and I moved it closer to my desk to keep an eye on it. After the new pairing the bulb does not show the problem any more. The lamp resting is an interesting hypotheses.

MattWestb, 2021-11-27:
If its working after repairing  (making one hard reset) it can being that it was having problem writing tokens in the NWM and was restarting then try writing or reading data and its fixed after making one new file for token data then being resettled.

keesj, 2021-11-27:
The thing is that I would then expect that I would not be able to control the light (turn on and off) but that did work. 

in the [video at before 72](https://youtu.be/WChPa4tbrAo?t=72) is is "clear" that is takes a few seconds before the problems appears. I .. might want to to some dangerous soldering to get to the bottom of this but first I want to understand the log a bit better.

MattWestb, 2021-11-27:
You was doing it in the sniffing 2 but some seconds after the light was restarting it self.
Also IKEA current firmware is have power outage memory and need saving on/off and light level in the NWM and its one background task and if its getting panic the watchdog is resetting the device.

Lock in the log if the device is doing Device Announcement and if not then is the problem solved and if returning  it to IKEA.

Adminiuga, 2021-11-27:
> in the video at before 72 is is "clear" that is takes a few seconds before the problems appears

You can see in the packet capture the same, there's a period when it does not send any "Device Announcement" messages after being controlled by HA/ZHA, but then starts flooding those messages again.

I can't say why the bulb is behaving this way, but I can say for sure that it is not commanded by HA/ZHA/Zigpy_deconz or the stick itself.

keesj, 2021-11-27:
It looks like indeed the problem might not be in the mentioned components. Perhaps a wrong OTA happened or I am just very unlucky. I will close the issue as there is not much more to add at this point. Thank you for your time.

puddly, 2021-11-27:
What OTA configuration did you have? Are you able to read the current firmware version out of the bulb? 

If it's a bad OTA image (unlikely, since you're the first to report an issue like this), you could force a downgrade.

keesj, 2021-11-27:
Hi,

The version of the firmware is 0x20029623

[image]

my OTA configuration is the following

```
  zha:
    zigpy_config:
      ota:
       otau_directory: /home/home/ota
       ikea_provider: true
       ledvance_provider: true
```
The only suspicious log item I see is

```
2021-11-27 17:25:15 WARNING (MainThread) [zigpy.ota.image] Fixing IKEA OTA image with trailing data (1100 bytes)
```

The OTA directory is empty and the logs show the following

```
2021-11-27 18:19:41 DEBUG (MainThread) [zigpy.zcl] [0xbfa7:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': field_control=1, manufacture_id=4476, image_type=16900, current_file_version=537040419, hardware_version=1
2021-11-27 18:19:41 DEBUG (MainThread) [zigpy.zcl] [0xbfa7:1:0x0019] OTA image version: 537040419, size: 209342. Update needed: False
2021-11-27 20:36:50 DEBUG (MainThread) [zigpy.zcl] [0xbfa7:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': field_control=1, manufacture_id=4476, image_type=16900, current_file_version=537040419, hardware_version=1
2021-11-27 20:36:50 DEBUG (MainThread) [zigpy.zcl] [0xbfa7:1:0x0019] OTA image version: 537040419, size: 209342. Update needed: False
```

perhaps I can force an update to be sure?

MattWestb, 2021-11-27:
Its running the latest production firmware.
The warning of extra bytes is normal for the moment on many IKEA firmware (one small Puddly thing).
If you like trying  downgrading the firmware its the instruction here https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#downgrading-your-device-firmware but i think its better letting if it its working OK.
I can looking if i can finding one older OTA file for the devices type.

MattWestb, 2021-11-27:
The version before the latest i have founding in my old deCONZ instalation.
[10038562-TRADFRI-sy5882-bulb-ws-2.0.023.ota.ota.zip](https://github.com/zigpy/zigpy/files/7612369/10038562-TRADFRI-sy5882-bulb-ws-2.0.023.ota.ota.zip)
Unzip it and putting in your OTA folder you have configured.

keesj, 2021-11-27:
Just for the for the record and future generations:p

I fixed a second light that was originally not attached to home assistant (just an IKEA button) that started blinking by adding it to home assistant and changing the light level (it this the NVM problem that @MattWestb  was talking about?

The sniffing log (does something with routing)

[mancave_crazy_blink_recover.pcapng.zip](https://github.com/zigpy/zigpy/files/7612389/mancave_crazy_blink_recover.pcapng.zip)

keesj, 2021-11-30:
I have some additional questions. I only captured a single transport key but it appears that the same key is used for all devices (with that key i can decode messages going to different lights) I do not understand. Should Unicast communication not use different keys?

Both the article here state something similar e.g.
https://www.blackhillsinfosec.com/understanding-zigbee-and-wireless-mesh-networking/
`
Security within the Zigbee network is based on a network key and link keys. Unicast communication between any two application layer devices is secured using a 128-bit link key shared by the two devices, while broadcast communications are secured by a 128-bit network key shared among ALL devices in the network. The intended recipient(s) of the communication is always aware of the precise security arrangement; by that, the recipient knows whether the frame is protected with a link key or a network key.
`

https://resources.infosecinstitute.com/topic/hacking-zigbee-networks/
`
   1) Network key is a 128-bit key distributed and shared among every device in the network to secure broadcast communication.
   2)  Link key, also 128-bits, serves as a means to secure unicast communication on Application layer, and it is shared only between two devices.
`

MattWestb, 2021-11-30:
The 1 is used for all communication in the network.
Its 2 types of link key, If Trust Center (TC) is one or the parts is one trust center link key.
If  2 device (with TC is not one of them) its one application link key.

I think wireshark is labeling all key commands as link key but you have flags that saying if its one network or link key if looking in the data is sent.

Zignee 3 devices is trying updating the "default well known link key" then paring and is then using the updated link key. If the coordinator is running HA1.X its not possible updating the link key.

If 2 devices is using app link keys they is encrypting the payload in the frame as its one private for the 2 device so other can understanding it but it using the normal encryption with network key.

Adminiuga, 2021-11-30:
Network key -- same across entire network encrypts both network and data payload unconditionally.
Aps Link keys encrypts payload if enabled. Currently it is not enabled.

keesj, 2022-02-11:
Hi, I had even more Ikea lights die on me over the last months. I can not recommend using them but by now I mostly stopped using HA for controlling my lights (Stil use it for the heating). Closing the issue and I wish you thank you for your time.

closed 2022-02-11 by keesj

MattWestb, 2022-02-11:
I think all this is problems with the firmware in the Cornbee II then deCONZ is trying hot patching the firmware for network conflicts that is not true.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5626