zigpy/zigpy #860: Enforcing minimum SQLite version or using another DB

puddly, 2021-12-03:
The issue fixed by #515 seems to be present in other portions of the appdb code (https://github.com/home-assistant/core/issues/57200).

Instead of being held back by older SQLite versions not supporting `UPSERT` and other useful features, perhaps we should enforce a minimum version. Some possibilities:

 1. [`pysqlite3-binary`](https://github.com/coleifer/pysqlite3) is *mostly* `sqlite3`-compatible and wheels exist for Linux.  Windows Python builds are already fine, as is macOS Homebrew Python, so it should fix most problems immediately.
    - The major problem with it is that `aiosqlite` doesn't work with alternative drivers yet (https://github.com/omnilib/aiosqlite/pull/106) and pysqlite3 isn't 100% compatible with sqlite3's API.
 2. Switch to another single file database, like [DuckDB](https://duckdb.org/).  It's more like a [typical SQL databases](https://duckdb.org/docs/sql/introduction) in terms of types and has a first-party Python module that's simply `pip install`-able.
    - It hasn't had a 1.0.0 release yet.
    - Has no async wrapper. On the other hand, we already have a separate synchronous loop to process database events so it's not like we're really using `aiosqlite` to begin with.
 3. Just throw a runtime error and leave it to the user to figure out how to upgrade SQLite.
    - Realistically, who would this affect?

Adminiuga, 2021-12-04:
At this point I think it is the time.

dmulcahey, 2021-12-04:
couple things here… async needs to be maintained for HA compliance (that’s how aiosqlite came in to play) and we can’t conflict version wise with anything required / depended on by the core of HA.

Adminiuga, 2021-12-04:
Agreed. That's why keeping the system SQLite keeps aiosqlite in place, but we just have to have a minimum version?

Adminiuga, 2021-12-04:
As long as people are not using Ubuntu 18?

dmulcahey, 2021-12-04:
I’m all for a minimum version so long as it isn’t higher than what HA depends on OOTB. That would cause a conflict no?

puddly, 2021-12-04:
`UPSERT` was introduced in 3.24.0:

 - HA OS ships with 3.34.1 ✅
 - Python installer for macOS and Windows ships with 3.31.1 ✅
   - macOS Homebrew Python is always the latest released version: 3.37.0 ✅
 - Ubuntu 18.04 ships 3.22.0 😦

My main concern with doing this:

```Python
import sqlite3

if sqlite3.sqlite_version_info < (3, 24, 0):
    raise RuntimeError("zigpy requires SQLite 3.24.0 or newer")
```

Is that updating SQLite system-wide *just for zigpy* is not really supported.  You can manually install a `.deb` file from a newer distribution and hope no other system programs break, but this is not a good idea.  You can also patch it at runtime with `LD_LIBRARY_PATH` after compiling sqlite3 yourself and keeping the `.so` file somewhere.  Neither seems like a particularly good idea.

`pysqlite3` is mostly an out-of-tree copy of the built-in `sqlite3` module and does not depend on the system SQLite installation. Surprisingly, the entire zigpy test suite passes with the following runtime patch:

```Python
import sys
import pysqlite3

sys.modules['sqlite3'] = pysqlite3
```

I'm definitely not advocating for including this code in zigpy, but I think fixing aiosqlite to allow it to work with pysqlite3 is the best approach. That way, we can at least tell the user to run `pip install pysqlite3-binary` when their system SQLite is too old and zigpy will automatically use it instead.

Adminiuga, 2021-12-04:
Does this mean we have to depend on pysqlite additionally? 
I personally would prefer the system SQLite, and have dist maintainers to keep it up-to-date.

The minimum version would not break HA OOB, and only those running core on the older systems in venv might be affected. And considering  HA core imposes min Python version requirements anyway, would it be problematic to ask for a newer SQLite?

Adminiuga, 2021-12-04:
Otherwise, if could get pysqlite support upstreamed into aiosqlite then could use pysqlite, but imo not too much gain va the effort?

puddly, 2021-12-04:
We can selectively import it if it's available and only if the system version is too old.  See my linked draft PR (with one failing unit test...) for one possible implementation.

dmulcahey, 2021-12-14:
So we may be good to go even with the Ubuntu issue. There is an ADR in the HA architecture repo that states the minimum supported databases:

Limit DB support to the following:

MariaDB ≥ 10.3 (2017)
MySQL ≥ 8.0 (2018)
PostgreSQL ≥ 12 (2019)
SQLite ≥ 3.31.0 (2020)

it’s approved and merged as of last month.

PR #861 (2022-01-04): Enforce minimum SQLite version and use `UPSERT` syntax
Fixes #860
closed 2022-01-04 by puddly