zigpy/zigpy #866: Attribute Polling / Issue with Tuya MeteringCluster

dumpfheimer, 2021-12-15:
Hi!

I just received some new Tuya Smart Zigbee Plugs with Energy Metering capability: _TZ3000_w0qqde0g
https://www.aliexpress.com/item/1005003444957152.html?spm=a2g0s.9042311.0.0.2f844c4dHm9l76
[image]

It seems to be working nicely except for the Energe Metering.
The values are readable manually:
[image]

And it seems like they are updated once when HA starts, but they do not change regularly by itself.

I would appreciate it if someone could point me in the right direction on how to implement attribute polling because I currently am not even sure whether to start looking into zigpy, zha or zhaquirks.

Thank you!

3v1n0, 2021-12-20:
The very same happens for `_TZ3000_cphmq0q7`.

For few days I actually got the metering value updated, but then it stopped for good... In any case I feel that the spot where to address this is zhaquirks.

dumpfheimer, 2021-12-20:
I think that there will be changes needed in zhaquirks - probably the definition of which attributes to poll and in which interval.
But if, for example, a timer is needed or a thread or some kind of event manager I believe the best place would probably be zigpy, or alternatively zha component.

So my best guess was: the "heavy lifting", designing and most of the implementation would probably be done here, at zigpy, and some configuration of that (new?) component /part would be in zhaquirks for this specific device.

dumpfheimer, 2021-12-20:
@3v1n0 I built a quick and dirty workaround that seems to work fine for me, if you want to check it out: https://github.com/dumpfheimer/hass_zha_attr_poll_workaround

3v1n0, 2021-12-20:
@dumpfheimer I was actually playing with it too last night, and eventually it seems that it's not that hard to fix zha-device-handlers, see https://github.com/zigpy/zha-device-handlers/pull/1226

The initial commit is actually more than enough to have this working, if you just want to try that.

dumpfheimer, 2021-12-21:
Possibly related:
https://github.com/home-assistant/core/issues/59081
https://github.com/zigpy/zha-device-handlers/pull/1211

dumpfheimer, 2021-12-21:
@3v1n0 Nice, thank you, this looks great!

How do you feel about making the async part reusable, so moving it out of the Tuya Cluster and somewhere more central?
I think there might be other devices that could benefit from this. Maybe frequency and attribute could be combined to a tuple, so every cluster could have multiple attributes being polled at different intervals.

After a quick search I really did not find any other issues with attributes not being updated other than these energy meters, so maybe this is unnecessary.

Thanks again!

dmulcahey, 2021-12-21:
Let’s leave this in Tuya.

dumpfheimer, 2021-12-21:
Okay, thanks. Then this is no longer a zigpy issue.

closed 2021-12-21 by dumpfheimer

3v1n0, 2021-12-21:
@dmulcahey I definitely made it generic, not sure if that i something the maintainers want though, so let's see how it will go the sh PR.