zigpy/zigpy #870: After upgrade: "UNIQUE constraint failed: devices_v7.ieee"

mdeweerd, 2021-12-20:
I upgraded a setup from "Dec 21  2020" (one year ago) using "pip install --upgrade zigby-znp" and I now have "zigpy 0.6.4".

The database ".dump" contains:
```sql
CREATE TABLE devices_v7 (
    ieee ieee NOT NULL,
    nwk INTEGER NOT NULL,
    status INTEGER NOT NULL
);
INSERT INTO devices_v7 VALUES('00:12:4b:00:22:08:ed:1a',47588,2);
INSERT INTO devices_v7 VALUES('00:12:4b:00:0e:0d:7d:6d',0,2);
```

So initally it's all unique.

I am running this using 'cygwin' - which was working just before the upgrade.

I now get the following error:
```
Traceback (most recent call last):
  File "./zig.py", line 85, in <module>
Exception in thread Thread-1:
    asyncio.run(main())
Traceback (most recent call last):
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
  File "/cygdrive/d/mdeweerd/workspace/pyznp/venv/lib/python3.8/site-packages/aiosqlite/core.py", line 98, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    result = function()
sqlite3.IntegrityError: UNIQUE constraint failed: devices_v7.ieee

During handling of the above exception, another exception occurred:
```

Below is the script - the code to report network information is from zigpy-cli.  
As is the code does not thow an exception, it throws exceptions when:

- `dump_app_info(app)` is uncommented, or,
- `await app.permit(60)` is uncommented.

I already added a device before the upgrade which I can't no longer do using the same code.
For completeness, I am using a CC2531 USB stick with ZNP for the coordinator.

```python
#!/bin/env python
from __future__ import annotations
import asyncio
import logging

from zigpy_znp.zigbee.application import ControllerApplication

from pprint import pprint

LOGGER = logging.getLogger(__name__)

class MainListener:
    """
    Contains callbacks that zigpy will call whenever something happens.
    Look for `listener_event` in the Zigpy source or just look at the logged warnings.
    """

    def __init__(self, application):
        self.application = application

    def device_joined(self, device):
        print(f"Device joined: {device}")

    def attribute_updated(self, device, cluster, attribute_id, value):
        print(f"Received an attribute update {attribute_id}={value}"
              f" on cluster {cluster} from device {device}")

def dump_app_info(app):
    if app.pan_id is not None:
        print(f"PAN ID:                0x{app.pan_id:04X}")

    print(f"Extended PAN ID:       {app.extended_pan_id}")
    print(f"Channel:               {app.channel}")

    if app.channels is not None:
        print(f"Channel mask:          {list(app.channels)}")

    print(f"NWK update ID:         {app.nwk_update_id}")
    print(f"Device IEEE:           {app.ieee}")
    print(f"Device NWK:            0x{app.nwk:04X}")

    if getattr(app, "network_key", None) is not None:
        print(f"Network key:           {format_bytes(app.network_key)}")
        print(f"Network key sequence:  {app.network_key_seq}")

def format_bytes(data: bytes) -> str:
    return ":".join(f"{b:02x}" for b in data.serialize())

async def main(enable_join=False):
    app = await ControllerApplication.new ({
        "database_path": "./zigbee.db",
        "device": {
            "path": "/dev/ttyS2",
        }
    },auto_form=True)
    # Fails when next line is not commented
    #dump_app_info(app)

    listener = MainListener(app)
    app.add_listener(listener)

    # Fails when next line is not commented
    # await app.startup(auto_form=True)
    # await app.startup(auto_form=False)
    # dump_app_info(app)

    # Permit joins for a minute
    #await app.permit(60)
    #await asyncio.sleep(60)

    #device=app.get_device('00:12:4b:00:22:08:ed:1a')
    #pprint(vars(device))

    # Just run forever
    await asyncio.get_running_loop().create_future()

if __name__ == "__main__":
    asyncio.run(main())
```

puddly, 2021-12-20:
Database migration is done within a transaction so if `devices_v7` is in your database, your database has a consistent state on zigpy startup, as expected. The error must be happening when  you're trying to join a device.

Can you enable full debug logging, post the output (including the complete traceback), and post the complete output of `sqlite3 zigbee.db .dump` so that I can try replicating your problem?

```python
if __name__ == "__main__":
    import coloredlogs  # pip install coloredlogs
    coloredlogs.install(level="DEBUG")

    asyncio.run(main())
```

mdeweerd, 2021-12-20:
Hi

I've done quite a few things in the mean time to work around the startup issues, including an upgrade to python3.9, cleaning up python3.8 and updating it, reinstall zigpy on python3.8 .  So my setup is not exactly the same as before.

I tried to reproduce it using the previous virtual environment with debugging on, using the posted script and recreating the database from the dump I made earlier.
I just got this:
```
Device is partially initialized: <Device model='NEW-VTH20' manuf='NEWPilote' nwk=0xB9E4 ieee=00:12:4b:00:22:08:ed:1a is_initialized=False>
```
[issue870.zip](https://github.com/zigpy/zigpy/files/7746977/issue870.zip)

Maybe the partial initialisation had something to do with it.

mdeweerd, 2021-12-20:
I ran into the issue again, albeit with slightly different code and - IMHO - different condition.

After upgrading the FW to the most recent "Koenk" version, the network channel change to 15, and as I want to have it on 11, I did a  python -m zigpy_znp.tools.nvram_reset`, added the network parameters to the script provided in the zip file, and launched the zig.py script.  
The database remained in place (copy also in the zip) and it ran up to the error (I copied the transcript to `notunique.log`).

[notunique.zip](https://github.com/zigpy/zigpy/files/7747392/notunique.zip)

puddly, 2021-12-20:
You seem to be calling running `app.startup(auto_form=True)` while the app is running. This is done for you with the `new` classmethod.

mdeweerd, 2021-12-20:
I am still discovering zigpy, the new alone did not seem to force the channel, so I added the app.startup.

Here is another way to trigger the "UNIQUE constrainte failed".  I added "    device=app.get_device('00:12:4b:00:22:08:ed:1a')" to the script which triggers it.  Commenting it makes the issue go away.  
I have to determine what the right way to use `get_device` is, but I'ld suggest that the   method would check if the ieee address exists in the table and return `None` if it doesn't - currently it exits with an exception.

[getdevice.zip](https://github.com/zigpy/zigpy/files/7747596/getdevice.zip)

puddly, 2021-12-20:
It's standard Python practice to throw exceptions instead of returning `None`:

```Python
try:
    dev = app.get_device(ieee=t.EUI64.convert("00:12:4b:00:22:08:ed:1a"))
except KeyError:
    # Device is not found
```

I believe your problem is a combination of your platform (Cygwin) and the fact that you're generating a traceback immediately when starting, so Python shows both. The `UNIQUE` error is normal (since the device already exists) but it should not be seen as a traceback.

In theory, if you add `await asyncio.sleep(0.1)` before your call to `app.get_device`, you won't see the `UNIQUE` error, just the `KeyError`.

mdeweerd, 2021-12-20:
Ok, I added a small delay `await asyncio.sleep(.1)`.  
I imported the types `import zigpy.types as t`.  
And corrected the get_device call: `    device=app.get_device(t.EUI64.convert('00:12:4b:00:22:08:ed:1a'))` .

That removed the errors.

With regards to the exception: standard practice is a point of view.

I'ld throw an error if it's something that almost certainly a failure, and None if it corresponds to a "Not found".  Design Patterns even have the Null Object which avoids checking for Null.

Exceptions have some overhead in execution as well as in coding.  In this case we can assume that if the caller uses the None value as a valid return, its misuse will almost certainly throw an exception itself.

Also, there is no check if both ieee and nwk are provided, so if `ieee` matches while `nwk` does not, an invalid result is returned as if it is a valid one.  In that case an exception would be appropriate.  

IMHO `getDevices` should have thrown an exception or made an automatic conversion from a string to EII64 because the input format of 'ieee' was not an EUI64 list, not because the value was "Not found".  

I was surprised to see a list used as a dict key, which look like a [mutable type that must not be used as a dict key](https://docs.python.org/3.9/library/stdtypes.html?highlight=dict#mapping-types-dict), and I am surprised that it works and could see anything ensuring that at first glance.

puddly, 2021-12-20:
I don't advocate keeping the delay, I just suggested it in case both exceptions showed up again and Python for some reason merged their tracebacks. Your code should work without it.

> With regards to the exception: standard practice is a point of view.
> I'ld throw an error if it's something that almost certainly a failure, and None if it corresponds to a "Not found". Design Patterns even have the Null Object which avoids checking for Null.

I would consider calling `get_device` with a NWK or IEEE that doesn't exist an unusual situation where an error is acceptable. It happens very rarely at runtime, and when it does it's something that needs to be handled explicitly.

> In this case we can assume that if the caller uses the None value as a valid return, its misuse will almost certainly throw an exception itself.

I would prefer to see `KeyError: aa:bb:cc:dd:aa:bb:cc:dd` with a traceback pointing to the exact line of code calling `get_device` instead of `AttributeError: 'NoneType' object has no attribute 'asd'` somewhere far away.

Worse yet, if the `None` value somehow makes its way into other data structures instead of a `Device` object, you will get cryptic errors later on, long after this call to `get_device`.

> Exceptions have some overhead in execution as well as in coding.

Python doesn't pretend to be a high-performance language so exceptions do not meaningfully slow it down, especially in async code. We're talking about *nanoseconds* of overhead `get_device` call. Unless you're calling it hundreds of thousands of times in a tight loop for some reason, it really doesn't matter.

> Also, there is no check if both ieee and nwk are provided, so if ieee matches while nwk does not, an invalid result is returned as if it is a valid one. In that case an exception would be appropriate.

Agreed, the two arguments should be mutually exclusive so throwing a `ValueError` if they are both provided is useful. If you'd like to make the change, submit a pull request.

> IMHO getDevices should have thrown an exception or made an automatic conversion from a string to EII64 because the input format of 'ieee' was not an EUI64 list, not because the value was "Not found".

Python doesn't have runtime type checking so we can't guard against every possible misuse but the parameters and return type should probably be documented in `application.py` to at least make it more clear. The `devices` dictionary is annotated:

```Python
        self.devices: dict[t.EUI64, zigpy.device.Device] = {}
```

Also, it's pretty rare to call `get_device` with string IEEE addresses in normal code, since you should almost always be working directly with `EUI64` objects. Perhaps a better change would be to throw an exception if the type of `ieee` is not `EUI64`, since `str(t.EUI64.convert("...")) == "..."` in the error message.

> I was surprised to see a list used as a dict key, which look like a mutable type that must not be used as a dict key, and I am surprised that it works and could see anything ensuring that at first glance.

This is usually true. The underlying type is mutable but since it's never mutated and defines a `__hash__` method, it works. You can probably change the base class of `FixedList` to `tuple` here: https://github.com/zigpy/zigpy/blob/0e901b013da074d24de18aca1e579287e756cd84/zigpy/types/basic.py#L627

Many unit tests will break in both zigpy and other libraries so it would require significant effort to fix. If you can get them passing in zigpy and all of its associated libraries, a pull request would be appreciated.

closed 2021-12-20 by puddly

mdeweerd, 2021-12-20:
FYI: I managed to achieve my first goal - read a manufacturer attribute ;-).

Regarding code changes: I prefer contributing through "Issues" - there's been too many occasions where for several reasons the code changes were refused - or worse - incompatible because the main author(s) made other changes in the mean time making the changed unmergeable and incompatible.