zigpy/zigpy #872: Make distinction between manufacturer attributes and foundation attributes

mdeweerd, 2021-12-21:
As far as I can see in the [zcl code](https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/__init__.py#L307) manufacturer attributes share the same storage inside a cluster (self.attributes[attr_id]).

As far as I understand, manufacturer attribute ids do not need to be exclusive to the foundation attribute ids.
The ZCL specification says:

> 2.3.3 Manufacturer Specific Extensions
> Manufacturers are free to extend the standard in the following ways:
> Add manufacturer specific attributes to a standard cluster.
>
> 2.6.1.4 Attribute Identifier
> Manufacturer specific attributes within a standard ZigBee cluster can be defined over the full 16-bit range.

Is it correct that Zigpy ignores attr_id specificity (while it accepts specifying the manufacturer code for commands)?  That values will clash when a manufacturer attr_id also exists as a standard attr_id?

If yes, will this evolve in Zigpy?

puddly, 2021-12-21:
Proper attribute definition schemas are in my WIP [ZCL7/8 PR](https://github.com/zigpy/zigpy/pull/716) ([`zigpy/zcl/foundation.py`](https://github.com/puddly/zigpy/blob/b6f4a63a0efd71774ec3cd00f3ae441c62caa0a6/zigpy/zcl/foundation.py#L628)).

> That values will clash when a manufacturer attr_id also exists as a standard attr_id?

Has this been observed?

mdeweerd, 2021-12-21:
a. Ok, so it's WIP.
b. Yes, I observe it.

Adminiuga, 2021-12-22:
Manufacturer attribute defined in zigpy take precedence over ZCL attributes.

What is the exact problem?

mdeweerd, 2021-12-22:
The problem is when the manufacturer attribute exists at the same attr_id at the same time as the Foundation Attribute.

dmulcahey, 2021-12-22:
> a. Ok, so it's WIP.
> b. Yes, I observe it.

What device and attribute causes different things to be returned when using the mfg code vs not using it for the same attrid. Or am I misunderstanding what you mean when you say you have observed it?

mdeweerd, 2021-12-22:
A device still under development, can't tell too much about it.

Here you have two relevant frames from the tshark log:
```raw

abbc@abbc.ynamics.com:~/zigbeelogs/sniffer2$ cat abc.log
Frame 3465: 70 bytes on wire (560 bits), 70 bytes captured (560 bits)
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Dec 22, 2021 09:56:16.468965000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1640163376.468965000 seconds
    [Time delta from previous captured frame: 0.011759000 seconds]
    [Time delta from previous displayed frame: 0.012240000 seconds]
    [Time since reference or first frame: 784.436542000 seconds]
    Frame Number: 3465
    Frame Length: 70 bytes (560 bits)
    Capture Length: 70 bytes (560 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:zbee_aps:zbee_zcl]
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x46ab
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..1. .... = Acknowledge Request: True
        .... .... .1.. .... = PAN ID Compression: True
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        10.. .... .... .... = Source Addressing Mode: Short/16-bit (0x2)
    Sequence Number: 206
    Destination PAN: 0x218a
    Destination: 0x0000
    Source: 0x46ab
    Frame Check Sequence (TI CC24xx format): FCS OK
        RSSI: 26dB
        FCS Valid: True
        LQI Correlation Value: 108
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x46ab
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 01.. .... = Discover Route: Enable (0x1)
        .... ...0 .... .... = Multicast: False
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
        .... 0... .... .... = Destination: False
        ...0 .... .... .... = Extended Source: False
        ..0. .... .... .... = End Device Initiator: False
    Destination: 0x0000
    Source: 0x46ab
    Radius: 30
    Sequence Number: 242
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 43170
        Extended Source: TexasIns_00:22:08:ed:1a (00:12:4b:00:22:08:ed:1a)
        Key Sequence Number: 0
        Message Integrity Code: d3f456f0
        [Key: 2999bee04ed9ec997282d44f35ac7a4c]
        [Key Label: 4]
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
[... 82 lines cut ...]
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: Simple Metering (0x0702)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 189
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 12
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 12
    Command: Read Attributes Response (0x01)
    Status Record, Uint: 14
        Attribute: Client: Functional Notification Flags / Server: Current Summation Delivered (0x0000)
        Status: Success (0x00)
        Data Type: 48-Bit Unsigned Integer (0x25)
        Uint48: 14 (0x000000000000000e)
```

Adminiuga, 2021-12-23:
> The problem is when the manufacturer attribute exists at the same attr_id at the same time as the Foundation Attribute

What is the problem exactly? Zigpy allows redefinition of the attributes as manufacturers attributes and those take precedence, no?

BTW, what is a foundation attribute? Do you mean ZCL?

IMO defining new attributes (extending attributes of the existing zcl8 clusters) is absolutely acceptable. Redefining ZCL attributes is somewhat ok (but why?) And changing the type of the well defined ZCL attributes is absolutely a no go.

mdeweerd, 2021-12-23:
IHMO manufacturer attributes exist alongside the non-manufacturer attributes (that I called Foundation Attributes, but that's not a good name).  
The ZCL indicates that "Manufacturer specific attributes within a standard ZigBee cluster can be defined over the full 16-bit range."  
But why would they replace the base attributes ?  As far as I understand it the manufacturer attribute creates some kind of manufacturer specific "namespace" - they complement the base attributes and each manufacturer "namespace" is distinct.

I do not see anything in the ZCL that says that:
- Manufacturer attributes "replace" the base attributes that have the same attr_id;
- Manufacturer attributes must have the same type as the standard attributes with the same attribute id;
- I only see a restriction on "extending clusters", but this seems to rather explain how future extensions to clusters in the ZCL work - i.e. evolutions of a standard cluster specification are backward compatible with previous versions, they do not add any new mandatory attributes, etc.  

# Tentatively answering the initial questions myself:

## Is it correct that Zigpy ignores attr_id specificity (while it accepts specifying the manufacturer code for commands)?

In Zigpy, the manufacturer attr_id takes precedence over the standard attr_id.  
Zigpy considers that a manufacturer attribute replaces the standard attribute with the same attribute id.

# That values will clash when a manufacturer attr_id also exists as a standard attr_id?

Zigpy does not consider that this is a clash as the manufacturer attribute is considered to replace the standard attribute with the same id.

# If yes, will this evolve in Zigpy?

No.

Adminiuga, 2021-12-23:
> IHMO manufacturer attributes exist alongside the non-manufacturer attributes (that I called Foundation Attributes, but that's not a good name)

No. There's no separate name space for ZCL attributes and manufacturer defined attributes. Manufacturers are free to extend the existing ZCL attributes and add their own attributes if the one in ZCL specs are insufficient. However both attributes use 65536 attribue IDs. You redefine the data type of the existing ZCL attributes only if you are interested in making a closed ecosystem, a-la Tuya, incompatible with anything else.

Adminiuga, 2021-12-23:
> If yes, will this evolve in Zigpy?

**No**

I don't see a compelling reason to do so.

closed 2021-12-23 by Adminiuga

Adminiuga, 2021-12-24:
in other words, the abstract application above the zigbee/zigpy layer expects the Attribute ID 0x0000 of the 0x0702 cluster to represent current summation delivered and be a signed 48 bit integer.  Why there's a need to send another data type, over the very same attribute ID, even if it preceded with manufacture code frame control, while there are plethora of other unused attributes id is beyond me. 

Maybe I'm missing anything, but there are already enough closed eco systems on the market already. But the code is available, and you are free to modify it as you'd like.

mdeweerd, 2021-12-24:
> No. There's no separate name space for ZCL attributes and manufacturer defined attributes. Manufacturers are free to extend the existing ZCL attributes and add their own attributes if the one in ZCL [...]

I do not see how the ZCL specifies that manufacturer attributes override standard attributes.  
There's really no use of a manufacturer code for standard attributes IMHO, and a bad idea to redefine standard attributes.  
Extensibility can not be guaranteed for manufacturer attributes if those replace standard attributes.   The ZCL public profiles could be extended with new attributes that could have the same attribute ids as manufacturer attributes that were defined prior to the ZCL update.  
If the alliance wanted to ensure extensibility, they should have reserved an "address space" of the attributes for manufacturer attributes, same way they did for clusters and profiles.  But the ZCL says instead that the entire 16-bit range is available for manufacturer attributes.

It's hard to find a concrete public example, but SiLabs provices one in item 17.1>4. on page 50 of https://www.silabs.com/documents/public/user-guides/ug391-zigbee-app-framework-dev-guide.pdf :
"Example: In the sample-extensions.xml file included with the application framework, Silicon Labs has defined a single attribute Transition Time which shares the same attribute ID with the on/off state in the on/off cluster 0x0000. However, the transition time attribute also contains the manufacturer code 0x1002, indicating that it is Silicon Labs’ manufacturer-specific attribute" .
The contents of the 'sample-extensions.xml' file can be found in https://github.com/Jim-tech/IoT-Developer-Boot-Camp/wiki/Zigbee-Custom-Clusters .

That example show a uint16 for the SiLab manufacturer attribute id 0x0000 for cluster id 0x0006 where the standard/public attribute for that attribute id in that cluster is a bool.

That example is in line with my understanding of the specification, but of course it could be violating the specification because of a misinterpretation the author of that example and myself are making.

dmulcahey, 2021-12-24:
I think part of the point is in regards to interoperability. A manufacturer can do what you are proposing but if a manufacturer does what you are proposing then they need to control the entire ecosystem to ensure things “just work”. For the public example you shared there is little chance that this device would work with an Alexa, SmartThings, Hue hub or any other Zigbee hub because they wouldn’t have the knowledge of the manufacturer specific attributes. 

I guess the question I have is why would you do this? And IMO “because the spec doesn’t 100% explicitly state that I can’t” isn’t a valid enough reason to support it. What advantage is there to using a well defined attribute with a potentially different type for a different purpose as a manufacturer? Just pick an attribute id that isn’t in use already or use a manufacturer specific cluster. The question I have is are there real world use cases with real advantages to do things like this? We have dealt with 100’s of devices at this point and I have yet to come across this situation.

mdeweerd, 2021-12-24:
It's fine by me that Zigpy does not support everything that is in the specification.  My point was to confirm that it doesn't and that it will or will not, to help improve or clarify the Zigpy functionnality.  

I am used to working to a specification - in the microelectronics world the real world solutions often do not exist yet.  So the [Robustness principle](https://en.wikipedia.org/wiki/Robustness_principle) is more applicable there.  
So yes that means that on the device it's better to define manufacturer attributes at some high attribute id far from the potential public attribute ids.  
But the principle could also apply to Zigpy that should then accept this kind of attributes.  Zigpy already has permissivity built in through quirks to accept also implementations that are not compliant.

There are several reasons for manufacturer attributes, the specification even allows to "hide" them.   In this case the feature is not supported by any public cluster attribute.  Whether or not I have an influence on the attribute id is another matter.

Now, I understand that Zigpy is contrained by its current implementation, and that the choice is to not evolve it in that regard because of the effort involved and because the team has not observed any device compromising this behavior yet.

Adminiuga, 2021-12-24:
Thanks for the Silabs link. I did not realize they actually keep the attributes in different name spaces.

> But the principle could also apply to Zigpy that should then accept this kind of attributes.
 
Zigpy does accept this kind of attributes, but they have to be defined in the quirk for the device. Zigpy just doesn't separate them into different namespaces, so the manufacturer specific attributes with the conflicting attribute IDs are going to override the standard attributes. While pretty much everything is there to completely separate those, ATM there are no plans to implement it. In fact, with some device manufacturers we see the opposite, when the manufacturer specific attributes are only accessible if you access them without the manufacture id in frame control. 

> There are several reasons for manufacturer attributes, the specification even allows to "hide" them.

What do you mean "to hide them"? hide manufacturer specific attributes or hide the standard attributes? So far, devices in the wild employed a very specific method of "hiding" the attributes or commands, they just don't support them. Looks pretty effective to me.

Adminiuga, 2021-12-24:
> Is it correct that Zigpy ignores attr_id specificity (while it accepts specifying the manufacturer code for commands)? That values will clash when a manufacturer attr_id also exists as a standard attr_id?

Yes. And whichever manufacturer overrides the standard attributes to different types would go into my personal blacklist ;)

mdeweerd, 2021-12-25:
# Hide

I was referring to "2.5.19.1.1 ZCL Header Fields" of the ZCL:
> ... or the manufacturer wishes to hide the fact that it supports extensions.

For instance, the manufacturer could hide commands that provide:
- Testing functionality;
- Enabling/disabling features;
- Serial number programming;
- Security related functions;
- Functionnality reserved to their closed system or partners.

# Manf Attributes only accessible without the manufacturer attribute in some cases.

Sure, I'ld expect that is not compatible with the ZCL specification.

But on the other hand some silicon manufacturers do not help designers to add manufacturer attributes correctly - I even read for one of them that it was not required to implement manufacturer attributes for zstack certification, so it was not implemented in the stack (i.e., nothing to help adding manufacturer specific stuff).

# Silabs

Apparently [they go even further than that](https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1023255/cc2652p-some-information-about-zigbee-standard-such-as-manufacturer-code-attributes-commands-got-from-silicon-labs?tisearch=e2e-sitesearch&keymatch=zstack%2525252520manufacturer%2525252520attribute%2525252520certification%2525252520required).

# Blacklist

That will be implicit as you'll stick with Zigpy.  And then - even if you desire - you may not be able to use them.

While not a guarantee, manufacturers can only truly avoid "clashes" if the manf attributes are placed in the upper addresses but not too high as public attributes have been added from the top downwards as well because there is no constraint on the future public attribute addresses.

Adminiuga, 2021-12-25:
>> There are several reasons for manufacturer attributes, the specification even allows to "hide" them. In this case the feature is not supported by any public cluster attribute. 

>  was referring to "2.5.19.1.1 ZCL Header Fields" of the ZCL:

So that paragraph pertains only to "Discover Commands Received Response" command. If you check the same spec for the discover attributes response command, then verbatim:
```
2.5.23.1.1 ZCL Header Fields
The frame control field SHALL be specified as follows. The frame type sub-field SHALL be set to indicate a
global command (0b00). The manufacturer-specific sub-field SHALL be set to the same value included in the
original Discover Attributes Extended command, with the exception that if the manufacture ID is 0xffff
(wildcard), then the response will contain the manufacture ID of the manufacturer-specific attributes, or will
not be present if the cluster supports no manufacturer-specific extensions.
The command identifier field SHALL be set to indicate the Discover Attributes Extended Response command.
```

In other words it doesn't say anything about hiding the attributes, so please stick to the specs and don't hide your manufacturer specific attributes, but feel free to hide the commands. Although, none of the manufacturers (but Danfoss, kudos to them) divulge their commands/attr implementation, so whether the command is hidden or not, it doesn't matter much cause you still don't know the number of parameters and their type. 

So the spec does allow manufacturer specific attributes for the full range, I still don't read it as: go ahead and redefine the standard attributes to your liking. The manufacturer specific attributes and commands are there to "extend" functionality, not to cripple the standard attributes and commands, just because you can.

mdeweerd, 2021-12-25:
Ok, I just saw that paragraph about hiding commands, and "extended" that to attributes.

Regarding Danfoss: they do specify a lot on attributes not in the public clusters.  Not so much kudos for them for reinventing the wheel on some attributes that are already in the public attribute list (Such as the LocalTemperatureOffset), and for some functionnality not respecting the Zigbee specification (they were informed).

I haven't checked on the device, but the Danfoss seems to define the "cluster_revision" as a manufacturer attribute according to [the source code](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/danfoss/thermostat.py#L102) .
So in Zigpy that overrides that Public attribute.  What if Danfoss updates its manufacturer cluster revision number to reflect updates to the manufacturer attributes, suppose it updates it to 2.   And at some point one of the public cluster definitions is updated and its revision code is publicly changed to 2 to reflect those public changes.

Now Danfoss is not on your blacklist because they did not change the type (uint16) of the 0xFFFD attribute, but we can suppose that they did change the semantics because it reflects the changes to manufacturer attributes (because otherwise there is no point in making it a manufacturer attribute).  

But maybe, it's just mistakingly listed as a manufacturer attribute in the Zigpy code.

Anyway, to me that is perfectly acceptable to me because I consider the attributes are in separate "name spaces", but I guess it is not acceptable for Zigpy.