zigpy/zigpy #873: Add write_attributes_raw to cluster methods

mdeweerd, 2021-12-21:
There is a "read_attributes_raw" method that is helpful as it allows reading attributes by using their numerical values.

There is no "write_attributes_raw" method.

It guess this could essentially be:

```python
    async def write_attributes(
        self, attributes: Dict[Union[str, int], Any], manufacturer: Optional[int] = None
    ) -> List:
        args = self._write_attr_records(attributes)
        return self.write_attributes_raw(args,manufacturer)       

    async def write_attributes_raw(
        self,  List[foundation.Attribute],, Any], manufacturer: Optional[int] = None
    ) -> List:
        result = await self._write_attributes(args, manufacturer=manufacturer)
        if not isinstance(result[0], list):
```

I was able to write to the manufacturer attribute like this:
```python
      # cluster is a Cluster instance.
      try:
         attr_value=foundation.TypeValue(0x41,t.LVBytes([0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]))
         attr=foundation.Attribute(0x0000, value=attr_value)
         result=awaitc luster._write_attributes(attr_list,manufacturer=0xF000)
      except asyncio.exceptions.TimeoutError:
         print("Exception on write manf attribute")
```

puddly, 2021-12-21:
Can you turn this change into a pull request?

mdeweerd, 2021-12-21:
Ok, I will fork and make & PR.  Leave it open, it will be on my issues & notificatons lists in Github.

Adminiuga, 2021-12-22:
Perhaps better is to expose the foundation commands to be allowed to called directly

mdeweerd, 2022-01-22:
Closing this because of the integration of the PR.

closed 2022-01-22 by mdeweerd