zigpy/zigpy #886: [REQUEST] zigpy to convert LQI into a universal standard with a unified scale and format for radio libraries?

Hedda, 2022-01-17:
Request for zigpy to convert LQI (Link Quality Indication ) into a universal standard with a unified scale and format for radio libraries?

Alternatively, could all radio libraries for zigpy to convert/translate LQI into a universal standard for zigpy so that in the end the user is always presented with a numerical sequence scale used for LQI measurement that maintains proportion between different radio types?

Would be great if all zigpy implementations and different radio libraries could calculate LQI the same way or at least similarly and always present values in a unified LQI scale for easier comparisons.

Currently all Zigbee stacks/radio manufacturers seem to calculate and handle their LQI measurements and scale differently. 

This is especially when users migrate from a Silicon Labs Zigbee Coordinator ([bellows radio library](https://github.com/zigpy/bellows)) to a Texas Instruments Zigbee Coordinator ([zigpy-znp radio library](https://github.com/zigpy/zigpy-znp)) as TI present a different scale with much lower range and thus many users will as a result commonly believe that they are getting much worse LQI. 

For example: 

Silicon Labs EmberZNet uses positive display numbers for LQI where higher is better: they display a zero error rate as a 255 LQI value, and a three error rate (worse) as a 63 LQI value. 

Texas Instruments Z-Stack _computes an 8-bit “link quality index” (LQI) for each received packet from the 2.4 GHz radio. The LQI is computed from the raw “received signal strength index” (RSSI) by linearly scaling it between the minimum and maximum defined RF power levels for the radio. This provides an LQI value that is based entirely on the strength of the received signal. This can be misleading in the case of a narrowband interferer that is within the channel bandwidth – the RSSI may be increased even though the true link quality decreases._

_The TI radios also provide a “correlation value” that is a measure of the received frame quality. Although not considered by the TI MAC in LQI calculation, the frame correlation is passed to the ZMAC layer (along with LQI and RSSI) in MCPS data confirm and data indication callbacks. The ZMacLqiAdjust() function in zmac_cb.c provides capability to adjust the default TI MAC value of LQI by taking the correlation into account._

https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/477818/query-regarding-lqi-given-by-ti-z-stack

http://software-dl.ti.com/simplelink/esd/plugins/simplelink_zigbee_sdk_plugin/1.60.00.14/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/z_stack_developers_guide/z-stack-overview.html#zmac-lqi-adjustment

The reason for requesting this is the question of why there is a huge difference in LQI values when migrating from a Silicon Labs Zigbee Coordinator ([bellows radio library](https://github.com/zigpy/bellows)) to a Texas Instruments Zigbee Coordinator ([zigpy-znp radio library](https://github.com/zigpy/zigpy-znp)), or vice versa, using the [backup and restore method for migration](https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration), is becoming more common in the Home Assistant community forum.

Back-story; 

LQI and RSSI are two of the main indicators that users will use to assess the quality and strength of connections respectively within their Zigbee network mesh. 

The problem is that while IEEE 802.11 already has a universal standard of signal strength with a unified scale and format for RSSI (Received Signal Strength Indication) that applies to  all Zigbee stacks / radio libraries, there is no IEEE 802.11 standard scale for LQI and instead the radio manufacturers of different Zigbee stacks have made up their own range for how LQI is presented.

The IEEE 802.15.4 specification only provides some general statements on the subject of LQI. From section 6.7.8: “The minimum and maximum LQI values (0x00 and 0xFF) should be associated with the lowest and highest IEEE 802.15.4 signals detectable by the receiver, and LQI values should be uniformly distributed between these two limits.” From section E.2.3: “The LQI (see 6.7.8) measures the received energy and/or SNR for each received packet. When energy level and SNR information are combined, they can indicate whether a corrupt packet resulted from low signal strength or from high signal strength plus interference.”

For reference; RSSI is always represented in -dBm format (0 to -100) which always is the power ratio in decibels (dB) of the measured power referenced to one milliwatt, a rule that already applies to all Zigbee stacks and radio libraries.

https://en.wikipedia.org/wiki/Received_signal_strength_indication#In_802.11_implementations

closed 2022-07-24 by github-actions[bot]

Hedda, 2023-03-29:
@puddly Will this issue of not using a standardized uniform scale for the LQI values between different radio libraries be indirectly resolved with your PR https://github.com/zigpy/bellows/pull/538 for bellows or does that only apply to the https://github.com/zigpy/zigpy/pull/1183 PR?

FYI, this issue was originally opened because many users were/are reporting that they keep seeing only low or medium-quality connections in ZHA's network virtualization even though their Zigbee devices are located within line-of-sight and close to the Zigbee Coordinator (even after they have tried to perform all the recommended actions to avoid electromagnetic interference).

pipiche38, 2023-03-29:
My experience on Zigate, is that there are 2 versions of the Zigate V1 and V2. They are based on different NXP chipset and also a different NXP Zigbee stack.
The end result is that the LQI values which are giving by the stack to the zigpy-zigate or even in native mode are different. So I'm afraid that depending of the chipset+stack you would be using their calculation is different.

Hedda, 2023-03-29:
> I'm afraid that depending of the chipset+stack you would be using their calculation is different.

So my original suggested idea in this request is still valid then?

MattWestb, 2023-03-29:
I think the best / only way getting realistic is using one reference device and testing how its sending and receiving and calculating in the radio lib how it shall being corrected.
If having one Silabs WSDK its possible flashing one firmware with manufacturer lib and / oe test harness and putting the device under test and getting report how is see its neighbors and looking what looks realistic and what need being corrected. 
As @pipiche38 was saying i think its the same with TI is depends of the chipset and how its implements in hard and software.
Asd we also knowing that deCONZ is very optimistic in the calculating so its not one small work to do if like digging deeper in it.

Hedda, 2023-03-31:
FYI, Hubitat developers recommend users to look at "lastHopLqi" in their Zigbee logs when troubleshooting unstable connections:

https://docs2.hubitat.com/en/user-interface/settings/zigbee-logs

* **lastHopLqi** - is an unsigned 8-bit integer ranging from 0 to 255, with the maximum value representing the best possible link quality index (LQI)( for delivery of the current packet across the last hop of the incoming route. LQI data is also used by the Zigbee stack to determine connectivity between neighboring nodes, and to select parent devices when joining the network as a ZigBee end device. An LQI below 200 represent links with high error rates, while and LQI of 200 represents approximately an 80% reliability of receiving the data packet intact.

* **RSSI** or Receiver Signal Strength (**B**) - is a signed 8-bit integer ranging from approximately -100 to 127, with each value representing the energy level in dBm or decibel-milliwatts at the radio receiver. Indication is the signal strength. Closer to 0 the better as that means less power was used to transmit and is an indication of stronger signal strength. This may be helpful to locate a Zigbee Endpoint device that is dropping off due to interference or weak signal. Since RSSI information does not pertain to any specific link with another device, the measured signal energy at a given frequency can come from any transmitter or device causing interference on that frequency, whether it is another nearby node on the same network, a device from a different PAN (Personal Area Network) on the same channel, or a non-ZigBee, non-802.15.4 source of interference such as a WiFi transmitter or microwave. However, it is important to remember that RSSI information is only between the Endpoint device and the first hop from that device (nearest router / repeater), so this may not necessarily be the hub.

[image]