zigpy/zigpy #895: Possible Database error

dumpfheimer, 2022-01-28:
I was looking through the zigpy debug logs for another reason when I stumbled over these errors:
cat /home/homeassistant/.homeassistant/home-assistant.log | grep FOREIGN

```
2022-01-28 21:44:45 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:8d:f0:52:aa:f9, 1, 0, 4, '_TZ3000_gjnozsaz') params: FOREIGN KEY constraint failed
2022-01-28 21:44:45 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:8d:f0:52:aa:f9, 1, 0, 5, 'TS011F') params: FOREIGN KEY constraint failed
2022-01-28 21:44:53 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (08:6b:d7:ff:fe:5d:0b:9b, 1, 0, 4, '3A Smart Home DE') params: FOREIGN KEY constraint failed
2022-01-28 21:44:53 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (08:6b:d7:ff:fe:5d:0b:9b, 1, 0, 5, 'LXN59-2S7LX1.0') params: FOREIGN KEY constraint failed
2022-01-28 21:46:07 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:f4:06:b2:91:18, 1, 0, 4, '_TZ3000_gjnozsaz') params: FOREIGN KEY constraint failed
2022-01-28 21:46:07 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:f4:06:b2:91:18, 1, 0, 5, 'TS011F') params: FOREIGN KEY constraint failed
2022-01-28 21:47:00 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (00:17:88:01:0b:45:15:34, 11, 0, 4, 'Signify Netherlands B.V.') params: FOREIGN KEY constraint failed
2022-01-28 21:47:00 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (00:17:88:01:0b:45:15:34, 11, 0, 5, 'LCL001') params: FOREIGN KEY constraint failed

```

Is this expected and can be ignored?

dumpfheimer, 2022-01-28:
Possibly relevant, I seem to get these messages every time HA starts:

```
022-01-28 21:42:39 DEBUG (MainThread) [zigpy.appdb] SQLite version for <module 'sqlite3' from '/usr/lib/python3.9/sqlite3/__init__.py'>: 3.35.5
2022-01-28 21:42:40 DEBUG (MainThread) [zigpy.appdb] Current database version is v3
2022-01-28 21:42:40 INFO (MainThread) [zigpy.appdb] Migrating database from v3 to v4
2022-01-28 21:42:40 INFO (MainThread) [zigpy.appdb] Migrating database from v4 to v5
2022-01-28 21:42:41 INFO (MainThread) [zigpy.appdb] Migrating database from v5 to v6
2022-01-28 21:42:41 INFO (MainThread) [zigpy.appdb] Migrating database from v6 to v7
```

sqlite tables:

```
# sqlite3 -readonly /home/homeassistant/.homeassistant/zigbee.db 
SQLite version 3.35.5 2021-04-19 18:32:05
Enter ".help" for usage hints.
sqlite> .tables
attributes                 in_clusters_v6           
attributes_cache_v5        in_clusters_v7           
attributes_cache_v6        neighbors                
attributes_cache_v7        neighbors_v4             
clusters                   neighbors_v5             
devices                    neighbors_v6             
devices_v5                 neighbors_v7             
devices_v6                 node_descriptors         
devices_v7                 node_descriptors_v4      
endpoints                  node_descriptors_v5      
endpoints_v5               node_descriptors_v6      
endpoints_v6               node_descriptors_v7      
endpoints_v7               out_clusters_v5          
group_members              out_clusters_v6          
group_members_v5           out_clusters_v7          
group_members_v6           output_clusters          
group_members_v7           relays                   
groups                     relays_v5                
groups_v5                  relays_v6                
groups_v6                  relays_v7                
groups_v7                  unsupported_attributes_v7
in_clusters_v5           
sqlite> 
```

Also possibly related, I have noticed that one of my tuya smart plugs was not "removable". I pressed "remove" on the device page, but when i power cycled the plug it came back.

puddly, 2022-01-28:
The foreign key errors are normal when a device is joining, they're just saying that the attribute cache can't be saved because there is no corresponding cluster/endpoint/device row, since the device won't be saved until it's fully initialized.

`Current database version is v3` is not normal.  Are you sure you see this every time you restart HA?

dumpfheimer, 2022-01-28:
You are right, I must have made that up. I restarted again and can now only find 2 log entries in my logs:
```
Jan 28 21:42:40 server run.sh[1028181]: 2022-01-28 21:42:40 DEBUG (MainThread) [zigpy.appdb] Current database version is v3
Jan 28 21:56:23 server run.sh[1030471]: 2022-01-28 21:56:23 DEBUG (MainThread) [zigpy.appdb] Current database version is v7
```

Is it reasonable that my DB was on V3? I normally use the latest git development state of zigpy and zigpy-znp

dumpfheimer, 2022-01-28:
Strange, if I check the backup logs I do have another V7 entry:
`Jan 17 08:31:19 server run.sh[793987]: 2022-01-17 08:31:19 DEBUG (MainThread) [zigpy.appdb] Current database version is v7
`

puddly, 2022-01-28:
No, v3 is from last July. Are you restoring `zigbee.db` from an old backup?

dumpfheimer, 2022-01-28:
No, I am positive I did not touch the zigbee.db file.

puddly, 2022-01-28:
The `user_version` setting is set by an application so *something* had to have run `PRAGMA user_version = 3` on your database.  Zigpy would have done this but only if you ran a build from last July or earlier.  Newer versions would not.

puddly, 2022-01-28:
Yes, they're only run when `PRAGMA user_version` is less than the zigpy database version (currenly 7).  This definitely should not be happening.  Could you zip up all of the `zigbee.db` files (including any `-shm` and `-wal` files, if they exist) and email them to me?

You could also post the files here in a `.zip` archive. The database contains no network information aside from IEEE addresses.

dumpfheimer, 2022-01-28:
I sent you an email, thanks!

puddly, 2022-01-28:
Got it. Nothing out of the ordinary as far as I can tell:

```
zigpy.appdb DEBUG Current database version is v7
```

A few IKEA devices are partially initialized, which can happen if you downgrade the database and re-upgrade it. This would cause all of the "new" tables to be erased when the migration happens, basically resetting your database to the way it was last year.

MattWestb, 2022-01-28:
@puddly if the upgrad is finding not OK devices is it not good rasing one warning or logging with devices its is so the user can repowering it / them or if its not working repairing it for not getting problem in the future ?

dumpfheimer, 2022-01-28:
It happens as soon as I use your hass core zha component:

```
sqlite3 -readonly zigbee.db 
SQLite version 3.35.5 2021-04-19 18:32:05
Enter ".help" for usage hints.
sqlite> PRAGMA user_version;
3

```

puddly, 2022-01-28:
Partially initialized devices are logged:

```Python
2022-01-28 16:30:25.126 ubuntu zigpy.application WARNING Device is partially initialized: <Device model='TRADFRI on/off switch' manuf='IKEA of Sweden' nwk=0x5743 ieee=... is_initialized=False>
2022-01-28 16:30:25.127 ubuntu zigpy.application WARNING Device is partially initialized: <Device model='TRADFRI control outlet' manuf='IKEA of Sweden' nwk=0xAFAE ieee=... is_initialized=False>
2022-01-28 16:30:25.132 ubuntu zigpy.application WARNING Device is partially initialized: <Device model='TRADFRI control outlet' manuf='IKEA of Sweden' nwk=0xD683 ieee=... is_initialized=False>
```

Since they're mains-powered, they will be automatically initialized whenever they next check in or send a response.

The only way that I was expecting this type of database problem to occur  would be if someone downgrades to zigpy from earlier than July 2021.  All newer releases will log a warning but will not change `user_version`, so no data will be changed:

```
This zigpy release uses database schema v3 but the database is v7.
Downgrading zigpy is *not* recommended and may result in data loss.
Use at your own risk.
```

@dumpfheimer are you restoring from any sort of HA backup that could possibly be replacing `zigbee.db`?

dumpfheimer, 2022-01-28:
okay It seems like I am getting somewhere
https://github.com/zigpy/zigpy/blob/a7ddb4dc2c4a6ba69ffde16bd8179a6d12fc8d86/zigpy/appdb.py#L14
$ pip freeze | grep zig
zigpy==0.26.0
zigpy-cc==0.5.2
zigpy-deconz==0.10.0
zigpy-xbee==0.13.0
zigpy-zigate==0.6.2
zigpy-znp==0.2.1

dumpfheimer, 2022-01-28:
Okay, I found the problem and it is sitting in my chair.

I cloned your repository and missed checking out the right branch. I was on the defautl branch which - tadaaaa - has zigpy 0.26.0
https://github.com/puddly/core/blob/c7bbba9d48f067e6a11532046191c5236ea56ac6/homeassistant/components/zha/manifest.json#L12

Sorry for wasting your time - back to testing zcl7!

closed 2022-01-28 by puddly