zigpy/zigpy #898: Higher decimal precision for electrical measurements

Scrumplex, 2022-02-03:
I have a Xiaomi Aqara Smart Plug (EU, `plug_mmeu01`). When I was using Zigbee2MQTT before I was getting the power measurement with a few decimal places of precision. With zigpy, I only get integer values.

My main problem is, that Home Assistant isn't updating my data in InfluxDB if the value didn't change (in this case it doesn't change as there is not enough precision).

In the screenshot below you can see an excerpt from my InfluxDB, where you can clearly see when I switched from Z2M to ZHA (2021-06-04):

[image]

I was looking in these places already:

https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/xiaomi/aqara/plug_mmeu01.py#L107

https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/xiaomi/__init__.py#L515-L546

and eventually landed here:
https://github.com/zigpy/zigpy/blob/8d7886d04c2c5715934725c67a9afdc625e9fa8b/zigpy/zcl/clusters/homeautomation.py#L79-L98

So I am unsure on how to continue, as this seems to be a limitation of `ElectricalMeasurement`

glcos, 2022-02-03:
This is indeed a great limitation.
For my sensor I could get around this by monkey patching the class from inside a quirk

```
from homeassistant.components.zha.sensor import ElectricalMeasurement as EM

def myFormatter(self, value: int) -> int | float:
    """Return 'normalized' value."""
    multiplier = getattr(self._channel, f"{self._div_mul_prefix}_multiplier")
    divisor = getattr(self._channel, f"{self._div_mul_prefix}_divisor")
    value = float(value * multiplier) / divisor
    return round(value, self._decimals)

EM.formatter = myFormatter
```

Not sure whether this can be done for your Aqara sensor.

Scrumplex, 2022-02-03:
I thought about adding random noise to it. the multiplier and divisor are both `1` for my sensor.

closed 2022-02-03 by Adminiuga