zigpy/zigpy #903: Add asyncio.exceptions.CancelledError as accepted error for retryable_retryable

mdeweerd, 2022-02-05:
I got a "CanceledError" on an outgoing command, and that reminded me that this error is not catched in the retryable_request :

https://github.com/zigpy/zigpy/blob/fc8e923e2f2482810b9643c0b2bde149a56a1930/zigpy/util.py#L145

There are quite a few methods that are not defined as retryable - I do not know if you want to update thos.  I am implementing the retryable functionnality in zha-toolkit when I need it and when it's missing in zigpy.

puddly, 2022-02-05:
If an outgoing command is cancelled, it shouldn't be retried. Preventing cancellation doesn't seem like a good idea.

mdeweerd, 2022-02-05:
It happened during a script run where I do 101 iterations and this exception stopped the loop.

I do not have an apparent reason why this occurred.

After adding Cancelled to the accepted exceptions list, it passed.

For the record, this is an exception trace (HA2021.12.10):

```raw
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1368, in _send_request_raw
    response = await asyncio.shield(
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1511, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/config/custom_components/zha_toolkit/__init__.py", line 415, in toolkit_service
    raise handler_exception
  File "/config/custom_components/zha_toolkit/__init__.py", line 379, in toolkit_service
    await handler(
  File "/config/custom_components/zha_toolkit/zcl_attr.py", line 255, in attr_write
    result_read = await cluster.read_attributes(
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 297, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request
    result, msg = await self._application.request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 453, in request
    return await self._send_request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1483, in _send_request
    response = await self._send_request_raw(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1386, in _send_request_raw
    raise InvalidCommandResponse(
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 128, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 203, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

You can close this for me - I am handling it locally anyway.

puddly, 2022-02-05:
What function call is generating `CancelledError`? If you're not explicitly cancelling it then this may be a bug in zigpy-znp.

mdeweerd, 2022-02-05:
```

puddly, 2022-02-07:
Can you provide debug logs of `zigpy-znp==0.7.0` sending these requests?

mdeweerd, 2022-02-07:
Ok, my setups are HAOS based - [2022.2.3 has this version](https://github.com/home-assistant/core/blob/2022.2.3/requirements_all.txt) so I'll upgrade my setup.

What do you want me to acdtivate for debugging?
An HA logger setup like this is ok (you know, but just to make sure you notice it: the # lines are not activated).
````
default: warning
logs:
  #homeassistant.core: debug
  #homeassistant.components.zha: debug
  #bellows.zigbee.application: debug
  #bellows.ezsp: debug
  custom_components.zha: debug
  zigpy: debug
  zigpy_znp: debug
  #zigpy_deconz.zigbee.application: debug
  #zigpy_deconz.api: debug
  ##zigpy_xbee.zigbee.application: debug
  ##zigpy_xbee.api: debug
  #zigpy_zigate: debug
  #zhaquirks: debug
  ## Extra, not sure this works:
  #zigpy.zdo: debug
  #zigpy.device: debug
  #zigpy_znp.api: debug
  #zigpy_znp.zigbee.zdo_converters: debug
  #zigpy_znp.zigbee.application: debug
  #zigpy_znp.commands.application: debug
  #zigpy_znp.commands.zdo: debug
  #zigpy_znp.commands.znp: debug
  #zigpy_znp.logger: debug
  #zigpy.util: debug
```

mdeweerd, 2022-02-15:
A TimeoutError following an attribute read seems to be converted in a Cancelled Error zigpy 0.43.0 zigpy-znp 0.7.0:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1286, in _send_request_raw
    response = await asyncio.shield(
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 372, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 575, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1630, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1667, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/config/custom_components/zha_toolkit/__init__.py", line 561, in toolkit_service
    raise handler_exception
  File "/config/custom_components/zha_toolkit/__init__.py", line 525, in toolkit_service
    await handler(
  File "/config/custom_components/zha_toolkit/__init__.py", line 608, in command_handler_default
    await default.default(
  File "/config/custom_components/zha_toolkit/default.py", line 33, in default
    await handler(app, listener, ieee, cmd, data, service, params, event_data)
  File "/config/custom_components/zha_toolkit/zcl_attr.py", line 392, in attr_write
    result_read = await cluster_read_attributes(
  File "/config/custom_components/zha_toolkit/zcl_attr.py", line 252, in cluster_read_attributes
    return await cluster.read_attributes(attrs, manufacturer=manufacturer)
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 297, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request
    result, msg = await self._application.request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 439, in request
    return await self._send_request(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1401, in _send_request
    response = await self._send_request_raw(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1304, in _send_request_raw
    raise InvalidCommandResponse(
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```