zigpy/zigpy #906: “python_requires” should be set with “>=3.6”, as zigpy 0.43.0 is not compatible with all Python versions.

PyVCEchecker, 2022-02-09:
Currently, the keyword argument **python_requires** of **setup()** is not set, and thus it is assumed that this distribution is compatible with all Python versions. 
However, I found it is not compatible with **Python<3.6**. My local Python version is 3.5, and I encounter the following error when executing _“pip install zigpy”_ 

```
Collecting zigpy
  Downloading zigpy-0.43.0-py3-none-any.whl (132 kB)
     |████████████████████████████████| 132 kB 495 kB/s 
……
Collecting aiohttp
  Downloading aiohttp-3.6.3-cp35-cp35m-manylinux1_x86_64.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 755 kB/s 
ERROR: Could not find a version that satisfies the requirement aiosqlite>=0.16.0 (from zigpy) (from versions: 0.1.0, 0.2.0, 0.2.1, 0.2.2, 0.3.0, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.8.0, 0.8.1, 0.9.0, 0.10.0, 0.11.0, 0.12.0)
ERROR: No matching distribution found for aiosqlite>=0.16.0 (from zigpy)
```
Dependencies of this distribution are listed as follows:
```bash
"aiohttp"
"aiosqlite (>=0.16.0)"
"crccheck"
"pycryptodome"
"voluptuous"
```
I found that **aiosqlite>=0.16.0** requires **Python>=3.6**, which results in installation failure of zigpy in Python3.5.

Way  to fix:
modify **setup()** in **setup.py,** add **python_requires** keyword argument:
```python
setup(…
     python_requires='>=3.6'
     …)
```
Thanks for your attention.
Best regrads,
PyVCEchecker

puddly, 2022-02-09:
It is strange that other than [sklearn-pandas](https://github.com/scikit-learn-contrib/sklearn-pandas/issues/251), Zigpy is the only other repository this bot has looked at: https://github.com/PyVCEchecker?tab=overview&from=2022-01-01&to=2022-01-31

PyVCEchecker, 2022-02-10:
We are not bot, and Zigpy is not the only other repository we look at. 
We are studying on Python version constraint error recently, and have found hundreds of such errors. We plan to submit a small number of issues first, and will submit more issues after gathering some feedback. I would be glad to know your opinions on this issue.
Thank you.