zigpy/zigpy #912: Database inconsistency with Device Clusters

dumpfheimer, 2022-02-10:
I am having an issue with my IKEA E1810 Remote Control. The symptom is the "dim up long press" is not working with HA device triggers. Reconfiguring / Re-pairing the device did not change anything.

It looks like the root cause is, that the wrong quirk is being chosen by zigpy as has been reported here: https://github.com/zigpy/zha-device-handlers/issues/1261

After taking a look into it a bit closer it seems to me like the clusters are not being stored in the database correctly:
This is the zigbee device signature from the UI:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1"
}
```

And this is from the zigbee.db:
```
select * from out_clusters_v7 where ieee = 'd0:cf:5e:ff:fe:d9:4e:aa';
d0:cf:5e:ff:fe:d9:4e:aa|1|3
d0:cf:5e:ff:fe:d9:4e:aa|1|4
d0:cf:5e:ff:fe:d9:4e:aa|1|6
d0:cf:5e:ff:fe:d9:4e:aa|1|8
d0:cf:5e:ff:fe:d9:4e:aa|1|25
d0:cf:5e:ff:fe:d9:4e:aa|1|4096
```

The output cluster with the ID 5 seems to be missing which causes this error at startup:
```
... DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote3'>
... DEBUG (MainThread) [zigpy.quirks.registry] Fail because output cluster mismatch on at least one endpoint

```

After inserting the cluster manually with `insert into out_clusters_v7 values ('d0:cf:5e:ff:fe:d9:4e:aa', 1, 5);` it uses the correct quirk
```
... DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote3'>
... DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for d0:cf:5e:ff:fe:d9:4e:aa: <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote3'>
```

Is there a reason the clusters are not updated in the database or could this be a bug?

Versions:
zha-quirks: https://github.com/puddly/zha-device-handlers@55d173b54558b94c405f6976fc31c14efaa72a56
zigpy: https://github.com/zigpy/zigpy@e91ac25d968403fd8e26c828e2177cfa2797973a
zigpy-znp: https://github.com/zigpy/zigpy-znp.git@f069b4580e9b39eb0c2bda09a04ae9c236447a31
home assistant: 2022.2.5

MattWestb, 2022-02-10:
If updating you loosing group binding but you can doing device binding instead and then its using the version 3 of the quirk.

MattWestb, 2022-02-10:
If you is searing lights direct without the system being online you can binding the remote to groups or devices (depends f the firmware) or using for automation in HA you dont need binding at all.

dumpfheimer, 2022-02-10:
Not quite sure how this is related to the issue to be honest, sorry.

MattWestb, 2022-02-10:
The device signature is reed from the device then pairing and is normally not being changed.
Firmware update can changing the hardware config like cluster and can needing repairing the device.

I only saying what is changed and can braking function in our system  and its up to you how  doing  things in yours.

puddly, 2022-02-10:
I think @MattWestb is referring to https://github.com/Koenkk/zigbee2mqtt/issues/7716 and is suggesting fixes for what will happen when the new firmware will likely break your existing automations.

> Should zigpy be detecting signature changes sometime after pairing?

It's not easy to tell why a device re-joined the network. Battery change? Power outage? Old device from the database that was reset and is joining again? OTA upgrade that changed the endpoints and clusters?

I think we should read the firmware version when a device joins so we can quickly check if a device needs to be re-initialized on rejoin.  This will probably come after we make OTA configurable per-device and maybe implement the end-device request queue (since they're really difficult to contact, you may be waiting for minutes for it to wake up).

dumpfheimer, 2022-02-10:
Yes that makes sence.

How about the "reconfigure device" button? Could the signature be updated during that process? It needs to be reachable for that anyway to work, right?

dumpfheimer, 2022-02-10:
Ah, nice! I will try to take a look at your branch, thanks!

I was afraid the automations (using device triggers) would need to be reconfigured because I expected it to be a new Device (in terms of home assistant). But I could be wrong about that (will test that shortly).

dumpfheimer, 2022-02-10:
The device ID seems to stay the same, which is great!
I did have to restart Home Assistant for the automations to start working again but still this is good news.

Since there seem to be efforts on this matter anyway I will close this issue.

Thanks for your support!

closed 2022-02-10 by dumpfheimer

TheZoker, 2022-04-05:
I have a similar issue up/down/left/right do not work, but the main button works without issues.
Repairing the device did not solve the issue for me. 
Any suggestions?

dumpfheimer, 2022-04-06:
What quirk is being used? It should say on the Home Assistant device page.

TheZoker, 2022-04-06:
Quirk:
```
zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1
```

Device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote1"
}
```

dumpfheimer, 2022-04-06:
It's using the wrong quirk. It should be IkeaTradfriRemote3, I believe. You can try removing the device on the device page, restart HA and pair the remote again.

MattWestb, 2022-04-06:
@TheZoker  Is your remote updated to the last version or is it on one old version ??
The current with not working group binding is 2.3.080 and shall using the last version on the quirk as @dumpfheimer  was writing if IKEA is not having making some wrong in the device setting in your device (The shortcut buttum and Open/Close is having some "strange" versions).

TheZoker, 2022-04-06:
I guess it is an old one (I don't own a tradfri bridge and bought the button like 2 years ago, no update since).
Firmware is 0x23014631

Interesting thing is, that it worked until the latest 2022.04 update of HA. After that it seems to stop working (maybe coincidence)

MattWestb, 2022-04-06:
Its much changes made in the ZHA beta that can making things with devices that is not 100% Zigbee Cluster Labery compatible but most is very well tested. 
Is you needing Zigbee group bringing for searing lights ?
If no update the firmware to latest if yes keep the old working with group binding.

I have the device in latest firmware version but i cant running the HA beta then im running one RCP firmware and is having other not related problems  but if i getting time in the week end i can moving little around and testing the remote with the beta.

puddly, 2022-04-06:
If it worked with 2022.03 and broke with 2022.04, there's some regression. Updating the firmware might change things but there seems to be some underlying problem that wasn't caught with the betas.

MattWestb, 2022-04-06:
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

MattWestb, 2022-04-06:
Ops 2022.4.0 is released !!!

MattWestb, 2022-11-24:
Our beloved @puddly !!

Is it some plans implanting https://github.com/zigpy/zigpy/issues/912#issuecomment-1035498550 in ZHA ??

Some of our France friends need it then doing new versions of there device firmware with OTA and changing cluster configs on the endpoints.

Current all user must deleting and re-adding the device for getting the right quirk device class (new device class) loading and not the old one.

I also see use for it then testing devices quirks and getting all clean without deleting, waiting and rejoining the deice all the time.

Also the "converting" NCP to NCP controllers and having problems with not complete initiated controllers.

Can being nice contaminated with sending one live with rejoin flag for getting tuya device getting the tuya magic triggered then the device is in init mode.