zigpy/zigpy #919: Aqara 2 channel relay not working as expected

leonardpitzu, 2022-02-15:
I own quite a few [Aqara](https://www.aliexpress.com/item/1005003586535373.html?spm=a2g0o.search0304.0.0.721867b9InO1DJ&algo_pvid=9cbb73a1-78ec-4b26-9efc-59c8c62eb011&aem_p4p_detail=202202150745191451588026228210149812636&algo_exp_id=9cbb73a1-78ec-4b26-9efc-59c8c62eb011-5) relays that i have built into the walls of my home.
I used deconz and it worked quite OK but i had to ditch it for some reasons and switched to native HA (ZHA).
I can switch the relay and it works as expected but power and energy are missing.
[image]

device signature:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x000c",
        "0x0010",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0010"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.relay.c2acn01",
  "class": "zhaquirks.xiaomi.aqara.relay_c2acn01.Relay"
}`

Hedda, 2022-02-24:
Such issues like these really belong in the repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)”  instead of here in zigpy which is lower-level issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) provides some information about how/why quirks are used and points people to that project:

https://www.home-assistant.io/integrations/zha/#zha-exception-and-deviation-handling

###  _ZHA EXCEPTION AND DEVIATION HANDLING_

_The ZHA implementation in Home Assistant relies on a library called “[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” to resolve issues with Zigbee devices that do not fully conform with the Zigbee standards. The few devices that deviate from the Zigbee specifications set by the [Zigbee Alliance](https://zigbeealliance.org/) may therefore require proper bug reports with debug logs from users to assistant the developers in writing custom [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/blob/dev/README.md) for all of a device functions to work properly with the ZHA integration._

_Such a custom “ZHA Device Handler” are Python scripts that internally are also referred to as a “quirk” because they fix “quirks”, like deviations from the standard specifications. ZHA Device Handles do this by transparently, acting as a translator, translating and converting non-compliant device messages and instead present them to the application as coming from a virtual compliant device. These ZHA Device Handlers for Home Assistant can thus be used to parse custom messages to and from Zigbee devices. The ZHA Device Handlers that are made can then be reused by all users in future versions of Home Assistant._

_The custom quirks implementations for zigpy implemented as ZHA Device Handlers for Home Assistant are a similar concept to that of [Hub-connected Device Handlers for the SmartThings Classics platform](https://docs.smartthings.com/en/latest/device-type-developers-guide/) as well as that of [Zigbee-Herdsman Converters (formerly Zigbee-Shepherd Converters) as used by Zigbee2mqtt](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html), meaning they are each virtual representations of a physical device that expose additional functionality that is not provided out-of-the-box by the existing integration between these platforms._

PS: Also, for ZHA general end-user community support, see here -> https://community.home-assistant.io/c/configuration/zha/48

Hedda, 2022-06-17:
@leonardpitzu Will you please consider closing this issue under the zigpy repository yourself?

Again high-level issues with device compatibility problems such as those you mention should be reported as issue to repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” (a.k.a. "[zha-quirks](https://pypi.org/project/zha-quirks)") instead of here in zigpy library repository which is only for low-level Zigbee issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

If test now and still have problem then open an issue there -> https://github.com/zigpy/zha-device-handlers/issues/new/choose

closed 2022-12-21 by github-actions[bot]