zigpy/zigpy #922: How to see or wipe the binding table

HarvsG, 2022-02-16:
When pairing Tradfri on/off switches I am able to pair a variable number of bulbs. With one switch I was able to  pair 4 or 5 previously. Now with another I am able to pair only 3 and with a third I am able to pair only 1.5.

When doing the pair I watch my homeassistant logs with zigpy.log set to debug.

A successful bind for one bulb looks like this in the logs

```log
2022-02-16 21:55:47 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] processing cluster: 6 Bind_req --> [0c:43:14:ff:fe:bf:6a:4d]
2022-02-16 21:55:47 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] processing cluster: 8 Bind_req --> [0c:43:14:ff:fe:bf:6a:4d]
2022-02-16 21:55:49 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] cluster: 6 Bind_req --> [0c:43:14:ff:fe:bf:6a:4d] completed: [<Status.SUCCESS: 0>]
2022-02-16 21:55:49 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] cluster: 8 Bind_req --> [0c:43:14:ff:fe:bf:6a:4d] completed: [<Status.SUCCESS: 0>]
```

As I pair each bulb after a while is see something like this

```log
2022-02-16 21:55:53 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] processing cluster: 6 Bind_req --> [0c:43:14:ff:fe:cb:c5:61]
2022-02-16 21:55:53 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] processing cluster: 8 Bind_req --> [0c:43:14:ff:fe:cb:c5:61]
2022-02-16 21:55:54 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] ZDO request ZDOCmd.NWK_addr_req: [0c:43:14:ff:fe:bf:6a:4d, 0, 0]
2022-02-16 21:55:56 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] cluster: 6 Bind_req --> [0c:43:14:ff:fe:cb:c5:61] completed: [<Status.TABLE_FULL: 140>]
2022-02-16 21:55:56 DEBUG (MainThread) [zigpy.zdo] [0x82c7:zdo] cluster: 8 Bind_req --> [0c:43:14:ff:fe:cb:c5:61] completed: [<Status.TABLE_FULL: 140>]
```

I can create 'space' by unbinding the co-ordinator and bind an extra bulb with no loss in function. However eventually I hit the table full error again.

The variable results makes me think something is clogging up the table - perhaps orphaned devices that have since left and re-joined the network and so can't be unbound - it seems unlikely that the different switches would actually have different table sizes.

Is there a way to print or wipe the binding table? I have tried unbinding each device and doing 'reconfigure device' - all that seems to do is re-bind the co-ordinator.

Unfortunately factory resetting the on off buttons is not an easy option for me as they are installed in the wall :/

This issue has been posted to the HA forums as well: https://community.home-assistant.io/t/tradfri-on-off-only-able-to-pair-three-bulbs/393616

MattWestb, 2022-02-16:
IKEA wall switches in UK :-)))
For my is one large **PITA** then i have one very nice tuya double dimmer module in my kitchen that i can dim both sides of under cabin lights and also OnOff local and with groups, But it was broken then IKEA was removing the group binding and i must doing device binding => OnOff commands is doing both gangs and no dim function from the remotes.

I think the best is to bombing tradfri forum on reddit so IKEA is putting back the group binding in there remotes so they is working OK with light groups.

If you is binding only On/Off devices (outlets) you can doing more and its TC or color lesser then ZHA is trying binding all cluster its thinking its good to binding and the color lights have more clusters that is binding = taking more space of the bind table in the remote.

HarvsG, 2022-02-16:
Yes, clusters is definitely part of it, however even pairing a power outlet used 2 clusters (6,8) And even taking clusters into account I can pair different amounts with 3 different switches. (10, 6 and 3). Something is definitely clogging up the binding table.

HarvsG, 2022-02-16:
> IKEA wall switches in UK :-)))

I'd rather not go into it here as it is off topic, but they are DIY wall switches. https://gist.github.com/HarvsG/106603c4023f6278b52563e4c64a3670

Adminiuga, 2022-02-17:
That's a creative way to make a wall switch :)

Currently it is not possible to read binding tables in ZHA, although devices should be smart enough and not to create duplicate binding entries, if you are trying to pair the same device. 

But should be possible to add a custom service to zha custom service integration

mdeweerd, 2022-02-18:
- There is a Zigbee command to reset a device, command 0 of cluster 0.  So you can reset your buttons without having to access them.  
  You can do that using the standard service zha.issue_zigbee_cluster_command.

There is a way to get the Binding Table of a device by makeing an "APSME-GET.request Parameters".
The binding table is provided through AIB attribute 0xC1 "apsBindingTable".

I'll try to add it to [zha_toolkit](/mdeweerd/zha-toolkit).

mdeweerd, 2022-02-18:
I learned more about APSME which is not an airborn interface.
Apparently "Mgmt_Bind_req (ClusterID=0x0033)" should do it.

mdeweerd, 2022-02-19:
> IMO that's depends on device implementation,

In principle it should reset the table because it's a factory reset.  If there is no other way, that can always be tested.

Section 2.3.5 of the ZCL specification:

> A factory reset is a deliberate behavior to reset the above described persistent data back to its original state when the product left the factory.

And in the "above described persistant data", bindings are listed.

HarvsG, 2022-02-19:
Many thanks for your reply, help and developing zha_toolkit @mdeweerd 

I have run the service you suggested and captured the logs.

```json
    "result":[
       "<Status.SUCCESS":0>,
       16,
       0,
       [
          "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"b9":"70":40,
          SrcEndpoint=1,
          ClusterId=1,
          DstAddress=MultiAddress(addrmode=3,
          "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
          endpoint=1)),
          "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"b9":"70":40,
          SrcEndpoint=1,
          ClusterId=32,
          DstAddress=MultiAddress(addrmode=3,
          "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
          endpoint=1)),
          "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"b9":"70":40,
          SrcEndpoint=1,
          ClusterId=258,
          DstAddress=MultiAddress(addrmode=3,
          "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
          endpoint=1))
       ]
    ],
```

Interestingly is see each binding being to the co-ordinator with different cluster ID targets. But I do not see any unexpected entries. Is this really the number max number of binding my device can support 🤔

HarvsG, 2022-02-19:
The conclusion this had brought me to is that there is indeed a limited number of space in the binding table - 16, a nice binary number - and that the binding table in many of my switches is 'clogged'.

HarvsG, 2022-02-19:
Got one of my newer switches, unbound it from everything and lo and behold:
```
[
   "<Status.SUCCESS":0>,
   3,
   0,
   [
      "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"bc":"63":10,
      SrcEndpoint=1,
      ClusterId=1,
      DstAddress=MultiAddress(addrmode=3,
      "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
      endpoint=1)),
      "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"bc":"63":10,
      SrcEndpoint=1,
      ClusterId=32,
      DstAddress=MultiAddress(addrmode=3,
      "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
      endpoint=1)),
      "Binding(SrcAddress=b4":"e3":"f9":"ff":"fe":"bc":"63":10,
      SrcEndpoint=1,
      ClusterId=258,
      DstAddress=MultiAddress(addrmode=3,
      "ieee=00":"21":"2e":"ff":"ff":"07":"4a":"ad",
      endpoint=1))
   ]
],
```

mdeweerd, 2022-02-19:
Yes, the 16 is the number of entries currently in the table.  I did not implement looping, but as the records are quite long I understand that only 4 are provided in a call.

HarvsG, 2022-02-19:
OK, I follow you - I seem to be getting 3 pre call - they always show the co-ordinator as the DstAddress ieee. Is there any way for me to see all the entries and the actual target ieees then I could send custom unbind requests to de-clog the table

mdeweerd, 2022-02-19:
I just released an new version of zha-toolkit where all bindings should be retrieved.  I do no have a device that has enough bindings to test it, so hopefully the loop works.

The output is formatted better in the 'result' fields:

```json
        "result": {
            "0": {
                "src": "00:12:4b:00:24:c2:25:3f",
                "src_ep": 1,
                "cluster_id": "0x0006",
                "dst": {
                    "addrmode": 3,
                    "dst_ieee": "00:12:4b:00:01:6a:41:0c",
                    "dst_ep": 1
                }
            }
        }
```

mdeweerd, 2022-02-19:
Regarding what you are trying to do: in principle it's better to add your lights and control in the same group (I believe that according to the specification that's recommended when there are more than 4)

HarvsG, 2022-02-19:
Agreed, unfortunately the ikea switches I am using do not support group bindings since a software update.

In order to restore my config to its former glory, I need two things
1. To know the ieees of all the devices my switch is bound to
2. To be able to issue unbind requests

Or alternatively - to issue a "wipe binding table" command. 
I have unbound from each of the ieees I own and still entries remain in the binding tables. I suspect these entries are old lightlink groups I used before the software update.

HarvsG, 2022-02-19:
Problem solved! In ZHA I issued unbind requests for each of the lightlink groups I had ('Default Light Link Group', 'No name Group 0x0003' and 'No Name Group 0x0004') - all clusters. It was one of those that was clogging the binding table.

Thank you for you help @mdeweerd, although your software was not needed for the solution - it was a great help with the debug.

closed 2022-02-19 by HarvsG

mdeweerd, 2022-02-19:
Great!  It's a toolkit - not intended to be part of the end solution or production quality, but a means to get there.

Now that the binding table can be read, it's possible to build on that to loop over all found bindings to unbind them.

mdeweerd, 2022-02-20:
Just released zha_toolkit.binds_remove_all .
First call: removed 3 bindings.
Second call: nothing to remove.
Reconfigure device
Third call: nothing to remove (ZHA does not redo the bindings?)
 
Strange, but at least the bindings are removed.

Adminiuga, 2022-02-20:
Zha only the redoes binding on device joining or re-configuration

mdeweerd, 2022-02-20:
@Adminiuga I did a "Reconfigure device" and no bindings were found afterwards for that test.

I just tried with another light and then with the original one - both got the bindings added again upon reconfiguration.

HarvsG, 2022-03-08:
@mdeweerd strange, when I do reconfigure I always get a re-bind to my conbee stick if it is unbound - doesn't affect other bindings. I haven't tested with your (amazing) toolkit

mdeweerd, 2022-03-08:
@HarvsG   When you reconfigure, HA will send the binding requests - you can check the detail of the reconfiguration and you'll see that the binding operations are listed.