zigpy/zigpy #924: Ensure that manufacturer id can be force to "None" from high level call

mdeweerd, 2022-02-19:
It appeared in https://github.com/zigpy/zha-device-handlers/discussions/1321#discussion-3849987 that even if the "manufacturer" was set to None in the high level read_attributes call, the Zigbee command actually sent did include the manufacturer code.

In this specific cas that was because a quirk with a cluster id automatically designated as a manufacturer cluster was active.

I think that setting "manufacturer" to `b""` avoids that it is replaced in the current code and not sent in the request.

There may however be caveats in the current implementation, and a method should continue to exist in future implementations, the typing hints need to be updated as well (manufacturer can only be None or uint16 .

Another (additional) method would be to recognize 0 as No Manufacturer.

Wireshark indicates that Manufacturer Code "0" stands for No Manufacturerer.

In the Metering Cluster and for Tunneling it is said that "A value of 0xFFFF indicates that the Manufacturer Code is not used"  but in OTA it is said to be a "wild card".  So 0xFFFF does not seem suitable.

So I propose:
- To continue to accept None as a non-enforced manufacturer code that can be set in lower levels;
- To use b"", "" and 0 as codes to indicate "No Manufacturer" and ensure that this does not result in the manf bit being set;
- Alternatively define a constant (string) allowing to detect the "No manufacturer code" parameter.

puddly, 2022-02-26:
I'm not able to find where the manufacturer ID is ever overriden within zigpy. All the request methods just pass it through so `None` should be respected.

mdeweerd, 2022-02-26:
Here is one example:

https://github.com/zigpy/zigpy/blob/dev/zigpy/quirks/__init__.py#L175-L178

mdeweerd, 2022-02-26:
The code I refered to is inside zigpy - defining the default behavior of quirks.

This can break setting manf to `b""` (and there is another location) :
https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/foundation.py#L542-L543

(When manf=b"", it is empty, but the is_manufacturer_specific flag is set).

There is some use in setting the manufacturer code and is_manufacturer_specific flag when the manufacturer is None, and I'ld propose that a manufacturer of 0 or b"" forces is_manufacturer_specific  to "False" and sets fields accordingly.  

Some results from a grep:
```
zigpy/quirks/__init__.py:        if manufacturer is None and self._has_manuf_attr(
zigpy/quirks/__init__.py:        if manufacturer is None and self._has_manuf_attr(attribute_ids):
zigpy/quirks/__init__.py:        if manufacturer is None and self._has_manuf_attr(
zigpy/quirks/__init__.py:        if manufacturer is None and self._has_manuf_attr(
zigpy/zcl/foundation.py:        if frame_control is not None and manufacturer is not None:
zigpy/zcl/foundation.py:        if name == "manufacturer" and self.frame_control is not None:
zigpy/zcl/foundation.py:            self.frame_control.is_manufacturer_specific = value is not None
zigpy/zcl/foundation.py:                is_reply=is_reply, is_manufacturer_specific=(manufacturer is not None)
zigpy/zcl/foundation.py:        manufacturer: int | t.uint16_t | None = None,
zigpy/zcl/foundation.py:                is_reply=is_reply, is_manufacturer_specific=(manufacturer is not None)
```

puddly, 2022-02-27:
I don't think it is valid to set a manufacturer code without also setting the `is_manufacturer_specific` frame control bit.  It dictates the  size of the header.  A compliant device would parse the first byte of the manufacturer code as the TSN and the second as the command ID, shifting the real TSN and command ID into the command payload and breaking all further parsing.

puddly, 2022-02-27:
My mistake, I misread your comment.  I think https://github.com/zigpy/zigpy/issues/924 will address the problem.

PR #928 (2022-03-03): Allow disabling the automatic manufacturer ID override in quirks
Fixes #924
closed 2022-03-03 by puddly