zigpy/zigpy #938: Integration of Legrand OTA files

tmartinez69009, 2022-03-30:
Hello,
I'm trying to update Legrand Devices with the OTA files. Theses files are provided by Legrand from their [website](https://developer.legrand.com/documentation/operating-manual/) and some other files are available [here](https://github.com/Koenkk/zigbee-OTA/pull/85/files).
Theses files seems working with other zigbee solutions (Z2M,Deconz,Zigate...)
When i add them to my OTA folder, i got this error : 
`ValueError: Wrong magic number for OTA Image: 536872960`

[image]

With another Legrand OTA file :

[image]

The Magic Number should be the same on all the files no ?

Any idea on how to solve that ?
Thanks you

puddly, 2022-03-31:
Can you upload `NLP.bin`? Even though the images are encrypted, Legrand doesn't seem like they're going to just provide it:

> After investigation of your request, we will send you all needed information to access firmware download page (We reserve the right not to accept every request)

The three other images you linked to have a non-standard structure and contain a four byte file size prefix and an unknown 17 byte suffix. You can get zigpy to recognize them right now by just cutting off the four byte prefix:

```console
$ dd if=input.bin bs=1 skip=4 of=output.bin
```

The suffix may be a hash and the images can possibly be validated.

tmartinez69009, 2022-03-31:
Thanks you very much for you work.

Your code works well with the firmware files available on github. The firmwares from Legrand website doesn't work.
Here is 2 samples firmwares:
[legrand_firm.zip](https://github.com/zigpy/zigpy/files/8392367/legrand_firm.zip)

Thanks you for your help !

puddly, 2022-03-31:
Hmm, those images don't seem to contain any data compatible with the normal Zigbee OTA data format. Perhaps they are meant to be wrapped and given a header?

Without a traffic capture of a compatible hub sending one of these as an OTA update to a Legrand device, I don't think there is much I can do with them, unfortunately.

tmartinez69009, 2022-03-31:
Thanks you, i'm working with Legrand's team to make the firmwares compatible with Zigpy. So, i will contact the dev team. I hope they will deliver compatible firmwares.

Hedda, 2022-04-06:
FYI, IIRC believe read that guys working on [Zigbee for Domoticz Plugin](https://github.com/zigbeefordomoticz/Domoticz-Zigbee) project may have a lot of experience with Legrand OTA?

@pipiche38 worked on a "Capture-OTA-from-Wireshark" project to build Zigbee firmware OTA from Legrand wireshark capture?

https://github.com/pipiche38/Capture-OTA-from-Wireshark

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Corner_Retreiving-Legrand-Firmware.md

pipiche38, 2022-04-06:
@hedda, Legrand provides official firmware update on their web site https://developer.legrand.com/documentation/operating-manual/ You have to be registered in order to get access to the firmware

Unfortunately there is a very  short list of firmware available for now

Now as @tmartinez69009 is reporting some non-official firmware from @Koenkk repo, you might check with him where they are coming from and how they have been retrieved. May be the retrieval was wrong.

On my end I have NLD-8 and NLP-42

puddly, 2022-04-06:
The "non-official" firmware works but the official firmware is not in any format that I've seen, nor does it contain any type of Zigbee OTA header or image data. Are you able to utilize it, @pipiche38?

pipiche38, 2022-04-06:
Yes I do

```
Available Firmware - ManufCode: 1021 ImageType: 0x0011 FileVersion:   3e4203 Size:   250647 Bytes Filename: NLL-31.ota
Available Firmware - ManufCode: 1021 ImageType: 0x0010 FileVersion:   2c4203 Size:   242439 Bytes Filename: NLM-44.ota
```

```
 md5sum *.ota
96417a02e252e0b6d8e22175e24b123d  NLL-31.ota
73f1299c259158c86b8344119a38cf0c  NLM-44.ota
```

PR #944 (2022-04-08): Parse Legrand OTA images
Fixes #938
closed 2022-04-08 by puddly