zigpy/zigpy #948: How to wrap a `t.Bool` attribute?

javicalle, 2022-04-02:
During a device quirk (https://github.com/zigpy/zha-device-handlers/issues/1442) we thought it would be a good idea to wrap one of the attributes of type `t.Bool` with a class that gives information about the True/False values meaning in a similar way to attributes that accept values. `t.enum`.
This way, the value shown in HA is more informative to the user as it has a functional meaning of the value.
Some examples would be:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L831-L846

But for values of type `t.Bool` this can not be done. This:
```python
class ChildLock(t.Bool):
    """Tuya child lock states."""

    LOCK = True  # maybe 1 or 0x01
    UNLOCK = False  # maybe 0 or 0x00
```

will not work because of:
```
TypeError: ChildLock: cannot extend enumeration 'Bool'
```

A `t.enum8` can be used like this:
```python
class ChildLock(t.enum8):
    """Tuya child lock states."""

    LOCK = t.Bool.true
    UNLOCK = t.Bool.false
```

but then values can not be setted via HA UI.

Other combinations with 0/1 or True/False values will raise errors `Status.INVALID_DATA_TYPE`:
```
2022-03-24 11:38:32 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [6] cluster_type: [in] endpoint_id: [1] attribute: [32768] manufacturer: [None] response: [0] failure: [{}],
2022-03-24 11:38:38 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=set_zigbee_cluster_attribute, service_data=ieee=a4:c1:38:e0:b5:a5:bb:c6, endpoint_id=1, cluster_id=6, cluster_type=in, attribute=32768, value=True>
2022-03-24 11:38:38 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x574F](TS011F): set: True for attr: 32768 to cluster: 6 for ept: 1 - res: [[WriteAttributesStatusRecord(status=<Status.INVALID_DATA_TYPE: 141>, attrid=32768)]]
2022-03-24 11:38:38 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [6] cluster_type: [in] endpoint_id: [1] attribute: [32768] value: [True] manufacturer: [None] response: [[[WriteAttributesStatusRecord(status=<Status.INVALID_DATA_TYPE: 141>, attrid=32768)]]]
```

So the question would be: is there a way to wrap a `t.Bool` attribute in a helper class?

I think it's a pretty common situation, but I haven't found any implementation or example of how to achieve it.

The goal would be to give more meaningful values to true/false attributes. 
Some other examples:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L1044-L1050

puddly, 2022-04-02:
Python enums unfortunately can't be subclassed:

```Python
import enum

class Enum1(enum.Enum):
    FOO = 1

# TypeError: Enum2: cannot extend enumeration 'Enum1'
class Enum2(Enum1):
    pass
```

https://github.com/ethanfurman/aenum may be an alternative stdlib `enum`-compatible module to check out.

Alternatively, I was also thinking of replacing the [`DATA_TYPES` dictionary](https://github.com/zigpy/zigpy/blob/76b97fd0fbeb45d818e1dd2eb89a11868016e178/zigpy/zcl/foundation.py#L165-L224) with per-type data, maybe something like this:

```Python
class ChildLock(t.enum8, type_id=zigpy.zcl.foundation.Types.Boolean):
    """Tuya child lock states."""

    LOCK = 0x01
    UNLOCK = 0x00
```

dmulcahey, 2022-04-02:
I like the alternative option. I was going to suggest being able to set the type of after watching all of the testing they were doing

closed 2024-04-20 by github-actions[bot]

puddly, 2024-08-30:
I think this has been addressed by https://github.com/zigpy/zigpy/pull/1463.