zigpy/zigpy #959: How to detect rejoin information

agatah2333, 2022-04-12:
Hello developer, Zigpy is a perfect project using python to build connections with Zigbee devices. I am trying to send some commands to my own devices. But sometimes, I find the device will restart. The device restart is like the device will be offline and rejoin with the dongle. Cause my ThirdReality smart switch needs to repair for no reason. Could anyone help me detect when the device is offline?

puddly, 2022-04-12:
Zigpy uses the following logic to detect when a device is a new join as opposed to a re-join:

https://github.com/zigpy/zigpy/blob/3857bfe3e7bdafffb0850c8171e06598471cdaba/zigpy/application.py#L281-L310

If you're building your own application on top of zigpy, you can check `device.last_seen` or handle the `device_joined` event.

agatah2333, 2022-04-12:
Thank you so much for your reply! I will try as above. And I am pretty curious about how to value attribute _device.last_seen_. 
For example, if we successfully send the 'on' command at 14:18,  _last_seen_ will value as a time value of 14:18? But what if the device is idle for a while without sending any command, will the _last_seen_ attribute update?

puddly, 2022-04-12:
It's updated when the radio library successfully sends a request: https://github.com/zigpy/zigpy/blob/2f8f841a2444bac97e0aa38fa6c4859072096c44/zigpy/device.py#L303

Or one is received from the device: https://github.com/zigpy/zigpy/blob/2f8f841a2444bac97e0aa38fa6c4859072096c44/zigpy/device.py#L323

agatah2333, 2022-04-12:
Thank you so much for your helpful explanation. And I wonder, is there any standard time interval for the device to report to the coordinator? The device Vendor?

puddly, 2022-04-12:
With the exception of some non-compliant devices like Aqara that do not support it, attribute reporting is something you configure on the device. For example, you can set up a battery powered sensor to report every 0.1 degree change once a second (and drain its battery in a few hours), or do it at most once per hour:

https://github.com/zigpy/zigpy/blob/3857bfe3e7bdafffb0850c8171e06598471cdaba/zigpy/zcl/__init__.py#L594-L622

Zigpy is low-level so if you are looking to just interact with Zigbee devices, you may find https://github.com/zigpy/zha-websocket-server/ slightly easier to use, since it sets up binding and attribute reporting for you.

agatah2333, 2022-04-12:
Thank you for your guidence!

closed 2022-04-12 by agatah2333