zigpy/zigpy #960: (Hue) OTA updates seemingly broken

TheJulianJES, 2022-04-13:
All Hue device OTA updates are currently failing or not even starting for me with Home Assistant 2022.4.3 (zigpy 0.44.2)

The following image is used: http://fds.dc1.philips.com/firmware/ZGB_100B_011D/16785152/100B-011D-01001F00-ConfLight-ModuLumV2-EFR32MG13.zigbee

This is what happens after sending the "notify update" command to any Hue device that has an update available:
<details>
<summary>Debug Logs</summary>

```
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=88, command_id=0, *is_reply=True)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Sending reply: image_notify(payload_type=<ImageNotifyPayloadType.QueryJitter: 0>, query_jitter=100, manufacturer_code=None, image_type=None, new_file_version=None)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x2684), DstEndpoint=11, DstPanId=0x0000, SrcEndpoint=1, ClusterId=25, TSN=88, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x19\x58\x00\x00\x64')
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=88)
2022-04-13 06:25:45 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x2684](1741530P7): Issued cluster command: cluster_id: 25 command: 0 command_type: client args: (0, 100) cluster_id: out manufacturer: None endpoint_id: 11
2022-04-13 06:25:45 DEBUG (MainThread) [homeassistant.components.zha.api] Issued command for: cluster_id: [25] cluster_type: [out] endpoint_id: [11] command: [0] command_type: [client] args: [0, 100] manufacturer: [None] response: Status.SUCCESS
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=25, SrcAddr=0x2684, SrcEndpoint=11, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=12975865, TSN=0, Data=b'\x01\xD3\x01\x00\x0B\x10\x1D\x01\x00\x1D\x00\x01', MacSrcAddr=0x6458, MsgResultRadius=27)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Received ZCL frame: b'\x01\xD3\x01\x00\x0B\x10\x1D\x01\x00\x1D\x00\x01'
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=211, command_id=1, *is_reply=False)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Decoded ZCL frame: Ota:query_next_image(field_control=0, manufacturer_code=4107, image_type=285, current_file_version=16784640, hardware_version=None)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Received command 0x01 (TSN 211): query_next_image(field_control=0, manufacturer_code=4107, image_type=285, current_file_version=16784640, hardware_version=None)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] OTA query_next_image handler for 'Philips 1741530P7': field_control=0, manufacture_id=4107, image_type=285, current_file_version=16784640, hardware_version=None, model=1741530P7
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] OTA image version: 16785152, size: 458138. Update needed: True
2022-04-13 06:25:45 INFO (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Updating: Philips 1741530P7
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=211, command_id=2, *is_reply=True)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Sending reply: query_next_image_response(status=<Status.SUCCESS: 0>, manufacturer_code=4107, image_type=285, file_version=16785152, image_size=458138)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x2684), DstEndpoint=11, DstPanId=0x0000, SrcEndpoint=1, ClusterId=25, TSN=211, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=30, Data=b'\x19\xD3\x02\x00\x0B\x10\x1D\x01\x00\x1F\x00\x01\x9A\xFD\x06\x00')
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-13 06:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=211)
2022-04-13 06:25:46 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=25, SrcAddr=0x2684, SrcEndpoint=11, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=13058851, TSN=0, Data=b'\x01\xD4\x03\x02\x0B\x10\x1D\x01\x00\x1F\x00\x01\x00\x00\x00\x00\x40\x00\x00', MacSrcAddr=0x6458, MsgResultRadius=27)
2022-04-13 06:25:46 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Received ZCL frame: b'\x01\xD4\x03\x02\x0B\x10\x1D\x01\x00\x1F\x00\x01\x00\x00\x00\x00\x40\x00\x00'
2022-04-13 06:25:46 DEBUG (MainThread) [zigpy.zcl] [0x2684:11:0x0019] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=212, command_id=3, *is_reply=False)
2022-04-13 06:25:46 ERROR (MainThread) [zigpy.device] Failed to parse message (b'01d403020b101d01001f000100000000400000') on cluster 25, because Data is too short to contain 1 bytes
```

</details>

This message seems to be the most interesting:
``Failed to parse message (b'01d403020b101d01001f000100000000400000') on cluster 25, because Data is too short to contain 1 bytes``
As far as I can see, there aren't any more messages after that in regards to the OTA upgrade.

It looks like the device isn't sending data correctly. However, I can't imagine Philips/Signify broke OTAs this badly in their previous release on different devices?
Since it doesn't even really start the OTA update, I thought if it's perhaps related to the recent ZCL v7 update? (Couldn't find anything easily for now.)

puddly, 2022-04-13:
Looks like "old" zigpy wasn't parsing the OTA `field_control` byte at all and ignoring the two unparsed bytes at the end of the packet. "New" ZCL parsing is stricter and fails due to the extra bytes being interpreted as the start of the `request_node_addr` field.

The `field_control` byte will need to be implemented and a complex structure substituted for every OTA command schema that requires it.

joe248, 2022-04-13:
I'm getting a similar message while trying to do an OTA update on a Securifi Peanut plug on Home Assistant 2022.4.2.

joe248, 2022-04-13:
@puddly your fix resolved my issue.  Thanks!

PR #961 (2022-04-13): Implement OTA frame control fields
Fixes #960
closed 2022-04-13 by puddly