zigpy/zigpy #965: Why do not need to add attributes when sending command?

agatah2333, 2022-04-15:
I use the below _request function_ to send messages to the device (Aqara smart switch), data variable sets b'\x01\x02\x01'. After sending this message, my Switch device changed its state from off -> on.

https://github.com/zigpy/zigpy/blob/dev/zigpy/device.py#L258-L307

 I wondered how to assemble the data variable, and I found I could check the profile python file. And my device belongs to _OnOff_ cluster, so I check the code below.
https://github.com/zigpy/zigpy/blob/56fae33898757a63a708f23c72b6611967e0ddbf/zigpy/zcl/clusters/general.py#L602-L654

But I am quite confused about the attribute part. I did not find any attributes information in the request function. Does this request function only focus on the _server_commands_ and _client_commands_ part? And may I know more about the data variable, like why we use 0x01 as the first Byte, from Zigbee specification?

Thanks for your valuable time!

Adminiuga, 2022-04-17:
Device.request() is a low level network request. The zdo and zcl/cluster classes help you to create the `data` for device.request()

E.g.
```
dev = application.get_device(nwk=0x1234)
endpoint = dev.endpoints[1]

# bellow is equivalent to cluster = endpoint.on_off
cluster = endpoint[6]

# send on command
res = await cluster.on()

# write attribute(s) to on off cluster
res = await cluster.write_attributes({0x0000: 1, "on_transition_time": 20 })
```

agatah2333, 2022-04-18:
Thank you so much for your reply!
I have a sengled bulb device, and I want to write 0x0000 "on_off" attributes to it since it has an OnOff Cluster (cluster id 6). But it could not work as I send a cluster.on() command. In my point of view, if I set a **Bool.True** value to the attribute "on_off", it should work as a cluster.on() command.
When I try to read the Zigbee specifications, I find there are two types of commands, global command and cluster command. I wonder if I could know more difference between cluster command and write attribute.  I am so sorry if I ask sth basic or stupid.

dmulcahey, 2022-04-18:
That’s not how this works. Look at the spec you will see the attribute is read only and reportable. You can’t expect this to work like this.

Adminiuga, 2022-04-18:
Yes, that would work. As Dave suggested, find the ZigBee Cluster Library specification, check the zcl header, frames and aps frames format.

agatah2333, 2022-04-19:
Thank you so much

closed 2022-04-19 by agatah2333